FreeCAD/FreeCAD #16699: Measurement: Center of Mass mode is broken

FEA-eng, 2024-09-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Center of Mass mode of the Measure tool is broken for me. It doesn't show any correct results and causes the following errors:

```
15:10:17  pyException: Traceback (most recent call last):
  File "E:\FreeCAD_weekly-builds-38794-conda-Windows-x86_64-py311\Mod\Measure\MeasureCOM.py", line 114, in parseSelection
    obj.Element = (o, item["subName"])
    ^^^^^^^^^^^
<class 'ValueError'>: PropertyLinkSub does not support external object
```

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

semhustej, 2024-09-21:
Hi,

I seem to have similar behavior, but I was not sure, whether I am using the tool in the correct way. The error message is the same. See the behavior in the video:

https://github.com/user-attachments/assets/ce66ebb7-2ac0-41d5-92dd-741a51ff0ee7

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git)
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

FEA-eng, 2024-09-22:
@maxwxyz or @luzpaz IMO this should be a Blocker.

maxwxyz, 2024-10-14:
@hlorus what is left here?

hlorus, 2024-10-14:
> > @hlorus what is left here?
> 
> #17109 should close it.

Exactly

closed 2024-10-14 by yorikvanhavre