FreeCAD/FreeCAD #16700: PartDesign: Simple Pad Up to shape fails with "Could not create face from sketch" error

FEA-eng, 2024-09-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I can't get a simple Pad with Up to shape mode to work (with all faces or with their manual selection):

[image]

Same if the plane is on the other side. Previously, there were no problems with cases like this so it seems to be a regression after #16030.

```
15:19:00  Pad001: Could not create face from sketch.
Intersecting sketch entities or multiple faces in a sketch are not allowed.
```

[Pad up to shape fails.FCStd.zip](https://github.com/user-attachments/files/17084183/Pad.up.to.shape.fails.FCStd.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-22:
cc @FlachyJoe

FEA-eng, 2024-09-22:
@maxwxyz or @luzpaz IMO this should be a Blocker.

FlachyJoe, 2024-09-22:
Fix found, the PR come soon.

PR #16734 (2024-09-22): [PD] Fix Pad uptoshape and add unit test
Fix #16700
closed 2024-09-22 by chennes