FreeCAD/FreeCAD #16701: CAM: FreeCAD 1.0RC1: V-Carve of ShapeString "cuts" above surface

chaotix-, 2024-09-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

V-Carve generates tool moves that are too high and end up above the surface on narrow areas.

As a testcase I just created a flat surface with a ShapeString on top and used a 60° V-Bit tool to create a V-Carve-Operation of this ShapeString:
[image]

On the view from the front it is clearly visible that some of the generated moves are above the surface:
[image]

Testfile:
[V-Carve-Height-Test.FCStd.zip](https://github.com/user-attachments/files/17084429/V-Carve-Height-Test.FCStd.zip)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-22:
@phaseloop any chance you can take a look at this ?

spanner888, 2024-09-22:
Looking at file provided, I noted:
1. Flat tip of the V-bit used is 1.0mm wide and for example the Y height of the BOTTOM of letter T is only 0.7mm, so that cannot be engraved, ie the Tool stays above the surface. The main stem of the T is wider and is engraved as are other wider parts of the letters.  
2. Using both the simulators shows the full letters are engraved in the STOCK surface, as it is 1mm above the model surface due to Job setting to extend model 1.0mm in XY and Z. Reducing the Z+ stock extension to 0, appears to make no difference, but again some parts of text are narrower than tool tip. So the operation seems to behave correctly.
3. I tried version one release either side of OP version above and also v1.1 and always got error below. Mention here to see if anyone else sees this as well....

```
07:42:50  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FC_wklrk73uQ/usr/Mod/CAM/Path/Op/Base.py", line 453, in onDocumentRestored
    self.opOnDocumentRestored(obj)
  File "/tmp/.mount_FC_wklrk73uQ/usr/Mod/CAM/Path/Op/Vcarve.py", line 343, in opOnDocumentRestored
    self.setupAdditionalProperties(obj)
  File "/tmp/.mount_FC_wklrk73uQ/usr/Mod/CAM/Path/Op/Vcarve.py", line 287, in setupAdditionalProperties
    obj.addProperty(
<class 'NameError'>: Property V_Carve_Height_Test#Vcarve.OptimizeMovements already exists
```

```
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git) AppImage
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CamScripts
  * Plot 2022.4.17
```

phaseloop, 2025-03-09:
@sliptonic What do you think about making default vbit tip setting a 0.1 mm?

This issue is being reported over and over again (also first issue mentioned in many youtube tutorials). I think such large tip bits are rarity and if someone uses too large tip for a project it becomes much simpler issue to debug than this.

sliptonic, 2025-03-09:
Makes sense to me.  The default tools were not created with any real consideration for real-world applicability.

phaseloop, 2025-03-09:
ok, I'll try to post a patch soon