FreeCAD/FreeCAD #16702: [Sketcher] Crash when closing a Sketcher tab while its task is open

xtemp09, 2024-09-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create a _Spreadsheet_
2. Create a sketch with _PartDesign_
3. Close _Sketcher_ tab.
4. Close _Sketcher_ task.

@CalligaroV, found when checking tab disappearance in #16378.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Debug
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

### Subproject(s) affected?

Sketcher

### Anything else?

<details><summary>Backtrace</summary>

```cpp
(gdb) bt
#0  0x0000000000000000 in ?? ()
#1  0x00007ffff5f5b27f in Gui::EditableDatumLabel::deactivate (this=this@entry=0x555558840600)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/EditableDatumLabel.cpp:133
#2  0x00007ffff5f5b2b9 in Gui::EditableDatumLabel::~EditableDatumLabel (this=0x555558840600, 
    __in_chrg=<optimized out>)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/EditableDatumLabel.cpp:93
#3  0x00007ffff5f5b2f7 in Gui::EditableDatumLabel::~EditableDatumLabel (this=0x555558840600, 
    __in_chrg=<optimized out>)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/EditableDatumLabel.cpp:97
#4  0x00007fffa65efd26 in std::default_delete<Gui::EditableDatumLabel>::operator() (__ptr=<optimized out>, 
    this=<optimized out>) at /usr/include/c++/13/bits/unique_ptr.h:93
#5  std::unique_ptr<Gui::EditableDatumLabel, std::default_delete<Gui::EditableDatumLabel> >::~unique_ptr (
    this=0x55555884a7b0, __in_chrg=<optimized out>) at /usr/include/c++/13/bits/unique_ptr.h:404
#6  0x00007fffa65efd4e in std::destroy_at<std::unique_ptr<Gui::EditableDatumLabel, std::default_delete<Gui::EditableDatumLabel> > > (__location=0x55555884a7b0) at /usr/include/c++/13/bits/stl_construct.h:80
#7  std::_Destroy<std::unique_ptr<Gui::EditableDatumLabel, std::default_delete<Gui::EditableDatumLabel> > >
    (__pointer=0x55555884a7b0) at /usr/include/c++/13/bits/stl_construct.h:149
#8  std::_Destroy_aux<false>::__destroy<std::unique_ptr<Gui::EditableDatumLabel, std::default_delete<Gui::EditableDatumLabel> >*> (__last=0x55555884a7e0, __first=0x55555884a7b0)
    at /usr/include/c++/13/bits/stl_construct.h:163
#9  std::_Destroy<std::unique_ptr<Gui::EditableDatumLabel, std::default_delete<Gui::EditableDatumLabel> >*>
    (__first=<optimized out>, __last=0x55555884a7e0) at /usr/include/c++/13/bits/stl_construct.h:196
#10 0x00007fffa660b10e in std::_Destroy<std::unique_ptr<Gui::EditableDatumLabel, std::default_delete<Gui::EditableDatumLabel> >*, std::unique_ptr<Gui::EditableDatumLabel, std::default_delete<Gui::EditableDatumLabel> > > (__last=<optimized out>, __first=<optimized out>) at /usr/include/c++/13/bits/alloc_traits.h:945
#11 std::vector<std::unique_ptr<Gui::EditableDatumLabel, std::default_delete<Gui::EditableDatumLabel> >, std::allocator<std::unique_ptr<Gui::EditableDatumLabel, std::default_delete<Gui::EditableDatumLabel> > > >::~vector (this=0x55555a2edd30, __in_chrg=<optimized out>) at /usr/include/c++/13/bits/stl_vector.h:735
#12 0x00007fffa660b1aa in SketcherGui::DrawSketchController<SketcherGui::DrawSketchHandlerRectangle, SketcherGui::StateMachines::FiveSeekEnd, 3, SketcherGui::OnViewParameters<6, 6, 8, 8>, SketcherGui::ConstructionMethods::RectangleConstructionMethod>::~DrawSketchController (this=this@entry=0x55555a2edd20, 
    __in_chrg=<optimized out>)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/Sketcher/Gui/DrawSketchController.h:268
#13 0x00007fffa6612cba in SketcherGui::DrawSketchDefaultWidgetController<SketcherGui::DrawSketchHandlerRectangle, SketcherGui::StateMachines::FiveSeekEnd, 3, SketcherGui::OnViewParameters<6, 6, 8, 8>, SketcherGui::WidgetParameters<0, 0, 0, 0>, SketcherGui::WidgetCheckboxes<2, 2, 2, 2>, SketcherGui::WidgetComboboxes<1, 1, 1, 1>, SketcherGui::ConstructionMethods::RectangleConstructionMethod, true>::~DrawSketchDefaultWidgetController (this=0x55555a2edd20, __in_chrg=<optimized out>)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/Sketcher/Gui/DrawSketchDefaultWidgetController.h:135
#14 0x00007fffa6612d00 in SketcherGui::DrawSketchControllableHandler<SketcherGui::DrawSketchDefaultWidgetController<SketcherGui::DrawSketchHandlerRectangle, SketcherGui::StateMachines::FiveSeekEnd, 3, SketcherGui::OnViewParameters<6, 6, 8, 8>, SketcherGui::WidgetParameters<0, 0, 0, 0>, SketcherGui::WidgetCheckboxes<2, 2, 2, 2>, SketcherGui::WidgetComboboxes<1, 1, 1, 1>, SketcherGui::ConstructionMethods::RectangleConstructionMethod, true> >::~DrawSketchControllableHandler (this=this@entry=0x55555a2edc50, __in_chrg=<optimized out>)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/Sketcher/Gui/DrawSketchControllableHandler.h:78
#15 0x00007fffa6613107 in SketcherGui::DrawSketchHandlerRectangle::~DrawSketchHandlerRectangle (
    this=0x55555a2edc50, __in_chrg=<optimized out>)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/Sketcher/Gui/DrawSketchHandlerRectangle.h:109
#16 SketcherGui::DrawSketchHandlerRectangle::~DrawSketchHandlerRectangle (this=0x55555a2edc50, 
[... 119 lines cut ...]
    at ../../../glib/gmain.c:4282
#75 0x00007ffff2b35279 in QEventDispatcherGlib::processEvents (this=0x5555557ce580, flags=...)
    at kernel/qeventdispatcher_glib.cpp:423
#76 0x00007ffff2ad6a7b in QEventLoop::exec (this=this@entry=0x7fffffffcf10, flags=..., flags@entry=...)
    at ../../include/QtCore/../../src/corelib/global/qflags.h:69
#77 0x00007ffff2adf3e8 in QCoreApplication::exec ()
    at ../../include/QtCore/../../src/corelib/global/qflags.h:121
#78 0x00007ffff2f38ecd in QGuiApplication::exec () at kernel/qguiapplication.cpp:1863
#79 0x00007ffff376bcb9 in QApplication::exec () at kernel/qapplication.cpp:2832
#80 0x00007ffff5ea15db in (anonymous namespace)::tryRunEventLoop (mainApp=...)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/Application.cpp:2207
#81 0x00007ffff5ea18a4 in (anonymous namespace)::runEventLoop (mainApp=...)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/Application.cpp:2227
#82 0x00007ffff5ea5730 in Gui::Application::runApplication ()
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/Application.cpp:2304
#83 0x000055555556a282 in main (argc=<optimized out>, argv=<optimized out>)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Main/MainGui.cpp:293
(gdb) 
```

</details>

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-09-21:
Windows reports

```
22:37:54  ViewProviderSketch: Camera is nullptr!
22:37:59  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
```

```ruby
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

CalligaroV, 2024-09-21:
@xtemp09, I can confirm this.

Trying with
```ruby
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Debug
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * Assembly4 0.50.15
  * freecad.gears 1.2.0
  * Beltrami 1.1.0
  * sheetmetal 0.4.24
  * Curves 0.6.44
  * fasteners 0.5.26
  * kicadStepUpMod 11.2.7
  * OpenTheme 2024.8.10
  * Render 2024.8.4
  * Ondsel-Lens 2024.7.5.02
``` 
I get two different crashes if I close the sketch tab while is active.
<details><summary>Backtrace if I'm quick enough to move the mouse to the MDI view</summary>
<p>
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f32c79d3520]
#1  0x7f32c7618bc8 in SoRenderManager::getCamera() const from /lib/x86_64-linux-gnu/libCoin.so.80c+0x8
#2  0x7f324cd626c2 in SketcherGui::ViewProviderSketch::camSensCB(void*, SoSensor*) from /home/calligarov/Documenti/Progetti/FreeCad/freecad-build/Mod/Sketcher/SketcherGui.so+0x52
#3  0x7f32c764cb6d in SoDataSensor::trigger() from /lib/x86_64-linux-gnu/libCoin.so.80c+0xd
#4  0x7f32c7653053 in SoSensorManager::processDelayQueue(int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x3b3
#5  0x7f32c7653634 in SoSensorManager::processTimerQueue() from /lib/x86_64-linux-gnu/libCoin.so.80c+0xd4
#6  0x7f32ccb25413 in SIM::Coin3D::Quarter::SensorManager::timerQueueTimeout() from /home/calligarov/Documenti/Progetti/FreeCad/freecad-build/lib/libFreeCADGui.so+0x1d
#7  0x7f32ccb2586f in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (SIM::Coin3D::Quarter::SensorManager::*)()>::call(void (SIM::Coin3D::Quarter::SensorManager::*)(), SIM::Coin3D::Quarter::SensorManager*, void**) from /home/calligarov/Documenti/Progetti/FreeCad/freecad-build/lib/libFreeCADGui.so+0x7e
#8  0x7f32ccb257ee in void QtPrivate::FunctionPointer<void (SIM::Coin3D::Quarter::SensorManager::*)()>::call<QtPrivate::List<>, void>(void (SIM::Coin3D::Quarter::SensorManager::*)(), SIM::Coin3D::Quarter::SensorManager*, void**) from /home/calligarov/Documenti/Progetti/FreeCad/freecad-build/lib/libFreeCADGui.so+0x46
#9  0x7f32ccb25755 in QtPrivate::QSlotObject<void (SIM::Coin3D::Quarter::SensorManager::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) from /home/calligarov/Documenti/Progetti/FreeCad/freecad-build/lib/libFreeCADGui.so+0x84
#10  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2f1793) [0x7f32c81de793]
#11  0x7f32c81e27fe in QTimer::timeout(QTimer::QPrivateSignal) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x3e
#12  0x7f32c81d433f in QObject::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x1cf
#13  0x7f32c8c98713 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#14  0x7f32cc5f46d8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/calligarov/Documenti/Progetti/FreeCad/freecad-build/lib/libFreeCADGui.so+0x114
#15  0x7f32c81a6e3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#16  0x7f32c81ff3eb in QTimerInfoList::activateTimers() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x3db
#17  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x312cec) [0x7f32c81ffcec]
#18  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b) [0x7f32c5ddad3b]
#19  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab2b8) [0x7f32c5e302b8]
#20  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7f32c5dd83e3]
#21  0x7f32c82000b8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x68
#22  0x7f32c81a575b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#23  0x7f32c81adcf4 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#24  /home/calligarov/Documenti/Progetti/FreeCad/freecad-build/lib/libFreeCADGui.so(+0x1369f86) [0x7f32cc488f86]
#25  /home/calligarov/Documenti/Progetti/FreeCad/freecad-build/lib/libFreeCADGui.so(+0x136a2b7) [0x7f32cc4892b7]
#26  0x7f32cc48961a in Gui::Application::runApplication() from /home/calligarov/Documenti/Progetti/FreeCad/freecad-build/lib/libFreeCADGui.so+0x244
#27  bin/FreeCAD(+0x2de8e) [0x564642ef5e8e]
#28  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f32c79bad90]
#29  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f32c79bae40]
#30  bin/FreeCAD(+0x2cfe5) [0x564642ef4fe5]
</p>
</details> 

<details><summary>Backtrace if I don't move the mouse</summary>
<p>
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f1c086b1520]
#1  0x7f1c0d2abb12 in SoField::evaluate() const from /home/calligarov/Documenti/Progetti/FreeCad/freecad-build/lib/libFreeCADGui.so+0x14
#2  0x7f1c0d2abcfe in SoSFRotation::getValue() const from /home/calligarov/Documenti/Progetti/FreeCad/freecad-build/lib/libFreeCADGui.so+0x1c
#3  0x7f1b8da1b85c in SketcherGui::ViewProviderSketch::onCameraChanged(SoCamera*) from /home/calligarov/Documenti/Progetti/FreeCad/freecad-build/Mod/Sketcher/SketcherGui.so+0x6e
[... 17 lines cut ...]
#21  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab2b8) [0x7f1c06b002b8]
[... cut ...]
19:33:54  TempoVis.restore: failed to restore detail ('SDCamera', 'Unnamed', 'the_cam'): list index out of range
``` 

For the latter case I would say that the cause is in the Python commands run during `ViewProviderSketch::unsetEdit()`, while for the former we may need some kind of check in `SketcherGui::ViewProviderSketch::camSensCB()`

Looking at your Backtrace it looks that yet another modification may be needed as for you the crash happens in `Gui::EditableDatumLabel::deactivate()`.
Anyway, for the records, I can reproduce also your crash with the same Backtrace

zbynekwinkler, 2024-10-07:
I don't see a crash. But I get this message in report view after closing the tab with the sketcher:

```
11:38:50  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: list index out of range
Exceptions must be caught before they go through Qt. Ignoring this will cause crashes on some systems.
```

So according to #16905 it would be a crash on a Mac.

```
OS: Debian GNU/Linux 12 (bookworm)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git)
Build type: Debug
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

chennes, 2024-10-10:
Setting aside the crash itself, is the expected behavior here that when you close the tab with the 3D view in it, the task panel that had been associated with that view is also closed?

xtemp09, 2024-10-10:
> Setting aside the crash itself, is the expected behavior here that when you close the tab with the 3D view in it, the task panel that had been associated with that view is also closed?

Tasks created by Pythonic workbenches always stay open even if the active document is closed. I've just checked _BIM_ and _Draft_.

PaddleStroke, 2024-10-18:
> Setting aside the crash itself, is the expected behavior here that when you close the tab with the 3D view in it, the task panel that had been associated with that view is also closed?

Yes I think this is what is expected. The task must ask for it actually
See : https://github.com/FreeCAD/FreeCAD/pull/15441
The issue : https://github.com/FreeCAD/FreeCAD/issues/9297

The task needs to use `autoClosedOnDeletedDocument(true)`

PaddleStroke, 2024-10-18:
The reproduction steps are not very clear. What do you mean by Close the sketcher tab?

PaddleStroke, 2024-10-18:
If I do the following : 
- New doc
- Create Spreasheet
- Open spreadsheet
- Click PartDesign New sketch
- select XY plane and click ok (the view goes to the 3d)
- close the unnamed document.
- click on close the sketcher task

I don't get a crash but I get : 

> 11:18:01  ViewProviderSketch: Camera is nullptr!
> 11:18:05  TempoVis.restore: failed to restore detail ('SDClipPlane', 'Unnamed', ''): list index out of range
> 11:18:05  TempoVis.restore: failed to restore detail ('SDCamera', 'Unnamed', 'the_cam'): list index out of range

PaddleStroke, 2024-10-18:
If I do : 
- new doc
- new sketch
- close document
- > the task panel is closed automatically (even though `TaskDlgEditSketch `is not doing `setAutoCloseOnDeletedDocument(true)`. And no crash and no warning.

PaddleStroke, 2024-10-18:
Ah now I can reproduce. Apparantly drawing a rectangle is necessary to reproduce the crash.

PaddleStroke, 2024-10-18:
Erratum: Crash only if there is an active handler. if you quite the rectangle tool before closing then it's OK

PaddleStroke, 2024-10-18:
The crash actually happen in EditableDatumLabel which is the tool on-view-parameters.

Though I would say that the way to fix that is to make sure the sketch edit is closed on tab closure, even if the document stays partially opened due to a spreadsheet or techdraw page.

chennes, 2024-10-18:
> Though I would say that the way to fix that is to make sure the sketch edit is closed on tab closure, even if the document stays partially opened due to a spreadsheet or techdraw page.

Yeah, that was my thinking as well -- there are probably other lurking errors that would occur if we don't do that. IMO if the 3D view closes, any task that is related to it should also be closed.

CalligaroV, 2024-10-18:
@PaddleStroke @chennes, I'm pretty sure you already did that but if not please take into account also my https://github.com/FreeCAD/FreeCAD/issues/16702#issuecomment-2365274479

It confirm what you said but it also points out that the crash isn't related only `Gui::EditableDatumLabel` (`Gui::EditableDatumLabel::deactivate()` AFAICS).

ATM I can't test if all the crashes I've been able to reproduce still happens on today's main (I'm on another branch) but if so I'd say that there are multiple causes

PaddleStroke, 2024-10-20:
Do you think closing the 3d view should close the spreadsheet tab as well? Currently it does not. It keeps the document partially loaded. Which is what is generating this issue.

So to fix this it seems we have 2 solutions : 
1 - On 3dview closing we close the document and so other tabs like spreadsheets and techdraw pages.
2 - On 3dview closing, we just don't close the document (so we leave as it is) and we just close the active dialog.
Thoughts?

xtemp09, 2024-10-20:
@PaddleStroke, mimicking behavior of counterparts of _FreeCAD_ is the best choice, I think.

PaddleStroke, 2024-10-20:
According to documentation (https://wiki.freecad.org/Std_CloseActiveWindow). And the code in `MDIView::closeEvent` / `MDIView::canClose `. It appears that it is intended to have the document being closed only if all views are closed. It is actually not related to partial loading.

So we probably need to do is : when a view is closed, associated task dialogs are closed as well.

Currently we have one mechanism in place: taskview is connected to the document closure event by `TaskView::slotClosedView`
For this to work the taskDialog has to register for it in its ctor like so : 

```
    setDocumentName(doc->getName());
    setAutoCloseOnDeletedDocument(true);
```

Which btw sketcher task dialog doesn't do, yet if the document close the dialog is removed. So it must be implementing this same behavior by itself.

So I think what should be done is that : 
- We replace existing `TaskView::slotDeletedDocument` by a `TaskView::slotClosedView`. 
- We somehow connect this to `MDIView::closeEvent`.

Then the question is, how do we associate a specific view to task dialogs? Do we store a pointer to the correct view in the task dialog the same way we are currently storing the doc name?

@wwmayer do you think this is the correct solution?

xtemp09, 2024-10-20:
@PaddleStroke, I would expect while the dialog is opened, the corresponding MDI window cannot be closed.

wwmayer, 2024-10-20:
> So we probably need to do is : when a view is closed, associated task dialogs are closed as well.

Such a relation doesn't exist in FreeCAD. It's even possible to have multiple 3D views open of the same document. Furthermore, your suggestion only heals the symptoms but not the underlying bug because a crash is always an indication that something in the code is not properly handled.

PaddleStroke, 2024-10-21:
> Such a relation doesn't exist in FreeCAD.

No but we can make it the same way you added `isAutoCloseOnDeletedDocument`. In ctor task dialogs can save a pointer to appropriate view.

> Furthermore, your suggestion only heals the symptoms but not the underlying bug because a crash is always an indication that something in the code is not properly handled.

About the crash sure it would be better if it does not happen. However I wouldn't say that the suggestion only heals symptoms. Because the fact that the sketcher dialog stays open after 3dview is closed is from a user perspective felt as a bug. And fixing the crash wouldn't solve that.

PaddleStroke, 2024-10-21:
As for the `EditableDatumLabel `crash it appears the crash comes from `viewer->getSceneGraph()`: 

```
void EditableDatumLabel::deactivate()
{
    stopEdit();

    if (cameraSensor) {
        auto data = static_cast<NodeData*>(cameraSensor->getData());
        delete data;
        cameraSensor->detach();
        delete cameraSensor;
        cameraSensor = nullptr;
    }

    if (viewer) {
        static_cast<SoSeparator*>(viewer->getSceneGraph())->removeChild(root); // NOLINT
    }
}
```

`viewer` is dangling pointer as the 3dview has been closed. What is the recommended way to handle this in FC?

Besides from @CalligaroV's report there're 2 other crashes that happen with this situation.

WandererFan, 2024-10-21:
You can't close the whole document just because a 3d window closes -  a) 3d tabs close unexpectedly sometimes if multiple documents are open, b) there can be multiple 3d tabs open for a given document,  c) spreadsheet/techdraw/dependency graph can all still perform useful work without an open 3d tab.

Note point b adds a complication in that closing one of the open 3d tabs may or may not have relevance to the open task.

MDIView::canClose could test for an open task dialog (Gui::Control().activeDialog()) but I don't know of a way to tell that a given task dialog depends on having an open 3d tab (maybe #17373 adds this?).

Note that the three modules mentioned above use derived versions of MDIView which may also complicate things.

PaddleStroke, 2024-10-21:
@WandererFan my current PR does not close document. It is closing task associated to the view instead.

wwmayer, 2024-10-21:
> Should viewer be a smart pointer?

You can't use a smart pointer here. When using smart pointers you must use them everywhere and you cannot mix smart pointers with raw pointers.

What you can use is a guarded pointer, i.e. Qt's `QPointer`

PaddleStroke, 2024-10-22:
The crash due to EditableDatumLabel is now fixed by the use of QPointer.
Though without the association of view that https://github.com/FreeCAD/FreeCAD/pull/17373 does, there are still warnings/errors : 

> 10:46:05  ViewProviderSketch: Camera is nullptr!
> 10:46:11  TempoVis.restore: failed to restore detail ('SDClipPlane', 'Unnamed', ''): list index out of range
> 10:46:11  TempoVis.restore: failed to restore detail ('SDCamera', 'Unnamed', 'the_cam'): list index out of range

wwmayer, 2024-10-22:
> there are still warnings/errors

Here the Show module needs some fixes.

closed 2024-10-29 by chennes

chennes, 2024-11-06:
Noting that @PaddleStroke is eligible to claim the bounty from the FPA. To claim, please send an email to fpa@freecad.org referencing this issue, and your preferred method of payment.