FreeCAD/FreeCAD #16704: FEM material setting problem

Roberto8303, 2024-09-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Unable to load material. Clicking on material does not happen, the material tab does not open.

[image]

[image]

[image]

### Full version info

```shell
OS: Windows 10 build 19042
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.67
  * Assembly3 0.12.2
  * Assembly4 0.50.15
  * btl 0.9.9
  * CurvedShapes 1.0.10
  * Curves 0.6.47
  * dxf-library
  * ExplodedAssembly
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * PieMenu 1.8.0
  * POV-Ray-Rendering
  * QuickMeasure 2022.10.28
  * Render 2024.8.4
  * SearchBar 1.0.2
  * sheetmetal 0.4.26
  * TitleBlock 0.5.2.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-09-21:
Can you record your screen while trying to add "Material for solid" ? Or, if you still have 0.21 installed, try with a fresh configuration (copy the configuration files from C:\Users\username\AppData\Roaming\FreeCAD and C:\Users\username\AppData\Local\FreeCAD, delete them temporarily and start FreeCAD).

Roberto8303, 2024-09-21:
nothing happens no material card appears

https://github.com/user-attachments/assets/15ecda70-5737-4e75-b04c-351f58ef24ba

FEA-eng, 2024-09-21:
What if you double-click another object (like SolverCcxTools) in the same way? Does the task panel open?

Roberto8303, 2024-09-21:
yes it opens correctly

FEA-eng, 2024-09-21:
Well, I have the same version and OS and it's working normally. Try with a clean config, it may help.

Roberto8303, 2024-09-21:
I restored the default configuration but I get this error
[image]

FEA-eng, 2024-09-21:
It would be better to disable the notification area so that errors appear only in the report view and can be copied and pasted here.

Can you share the .FCStd file ? I would check it on my side.

Roberto8303, 2024-09-21:
How do you share?

FEA-eng, 2024-09-21:
> How do you share?

Change the extension to .zip (or add this extension after .FCStd). Then it can be uploaded here.

Roberto8303, 2024-09-21:
however I noticed that the material window like yours doesn't even appear in partdesign for me

FEA-eng, 2024-09-21:
> however I noticed that the material window like yours doesn't even appear in partdesign for me

That one is specific to FEM. Part Design has different material assignment - like in your last screenshot attached to the initial bug report.

Roberto8303, 2024-09-21:
[PIASTRA fem.zip.zip](https://github.com/user-attachments/files/17085383/PIASTRA.fem.zip.zip)

FEA-eng, 2024-09-21:
For me, it works normally in your file:

[image]

But you should have only one material object (and of course, also mesh object) before submitting an analysis.

Roberto8303, 2024-09-21:
Weird!!! What can depend perhaps some folder of material references is compromised?

FEA-eng, 2024-09-21:
It can be related to #16516 but I have that one while I can't reproduce this one.

Roberto8303, 2024-09-21:
[image]
how did you set it up here?

Roberto8303, 2024-09-21:
__title__ = "FreeCAD FEM material task panel for the document object"
__author__ = "Juergen Riegel, Bernd Hahnebach, Qingfeng Xia"
__url__ = "https://www.freecad.org"

## @package task_material_common
#  \ingroup FEM
#  \brief FreeCAD FEM _ViewProviderFemMaterial
#  \brief task panel for common material object

from PySide import QtCore
from PySide import QtGui

import FreeCAD
import FreeCADGui
from FreeCAD import Units

from femguiutils import selection_widgets

unicode = str

class _TaskPanel:
    """
    The editmode TaskPanel for FemMaterial objects
    """

    def __init__(self, obj):

        FreeCAD.Console.PrintMessage("\n")  # empty line on start task panel
        self.obj = obj
        self.material = self.obj.Material  # FreeCAD material dictionary of current material
        self.card_path = ""
        self.materials = {}  # { card_path : FreeCAD material dict, ... }
        self.cards = {}  # { card_path : card_names, ... }
        self.icons = {}  # { card_path : icon_path, ... }
        # mat_card is the FCMat file
        # card_name is the file name of the mat_card
        # card_path is the whole file path of the mat_card
        # material_name is the value of the key name in FreeCAD material dictionary
        # they might not match because of special letters in the material_name
        # which are changed in the card_name to english standard characters
        self.has_transient_mat = False

        # parameter widget
        self.parameterWidget = FreeCADGui.PySideUic.loadUi(
            FreeCAD.getHomePath() + "Mod/Fem/Resources/ui/Material.ui"
        )
        # globals
        QtCore.QObject.connect(
            self.parameterWidget.cb_materials,
            QtCore.SIGNAL("activated(int)"),
            self.choose_material
        )
        QtCore.QObject.connect(
            self.parameterWidget.chbu_allow_edit,
            QtCore.SIGNAL("clicked()"),
            self.toggleInputFieldsReadOnly
        )
        QtCore.QObject.connect(
            self.parameterWidget.pushButton_editMat,
            QtCore.SIGNAL("clicked()"),
            self.edit_material
        )
        # basic properties must be provided
        QtCore.QObject.connect(
            self.parameterWidget.input_fd_density,
            QtCore.SIGNAL("editingFinished()"),
            self.density_changed
        )
        # mechanical properties
        QtCore.QObject.connect(
[... 646 lines cut ...]
    def add_cards_to_combo_box(self):
        # fill combobox, in combo box the card name is used not the material name
        self.parameterWidget.cb_materials.clear()

        mat_prefs = FreeCAD.ParamGet("User parameter:BaseApp/Preferences/Mod/Material/Cards")
        sort_by_resources = mat_prefs.GetBool("SortByResources", False)

        card_name_list = []  # [ [card_name, card_path, icon_path], ... ]

        if sort_by_resources is True:
            for a_path in sorted(self.materials.keys()):
                card_name_list.append([self.cards[a_path], a_path, self.icons[a_path]])
        else:
            card_names_tmp = {}
            for path, name in self.cards.items():
                card_names_tmp[name] = path
            for a_name in sorted(card_names_tmp.keys()):
                a_path = card_names_tmp[a_name]
                card_name_list.append([a_name, a_path, self.icons[a_path]])

        for mat in card_name_list:
            self.parameterWidget.cb_materials.addItem(QtGui.QIcon(mat[2]), mat[0], mat[1])
            # the whole card path is added to the combo box to make it unique
            # see def choose_material:
            # for assignment of self.card_path the path form the parameterWidget is used

Roberto8303, 2024-09-21:
it gives me errors on these lines in code 58 93 187 312 433

FEA-eng, 2024-09-22:
@Roberto8303 Could you edit [this comment](https://github.com/FreeCAD/FreeCAD/issues/16704#issuecomment-2365287547) to delete the whole file content pasted there and replace it with just the name of the file (task_material_common)? Will be much easier to scroll through this issue and we don't really need the file content here since it's already in the repository.

@davesrocketshop Can you have a look at the errors reported here? 
- https://github.com/FreeCAD/FreeCAD/issues/16704#issuecomment-2365272423
- https://github.com/FreeCAD/FreeCAD/issues/16704#issuecomment-2365288160 (file task_material_common)

Is it related to #16516 in any way?

FEA-eng, 2024-09-23:
>  how did you set it up here?

I have default settings there, same as you. But it shouldn't matter in the case of such issues. Something seems to be wrong with the installation of the Material system and we have a similar issue currently under investigation. Hopefully, it's the same problem.

FEA-eng, 2024-09-25:
@Roberto8303 Can you retry with the newest weekly build (38830 or newer)? There's a chance it's fixed now.

Roberto8303, 2024-09-25:
[image]
Not work 38794

FEA-eng, 2024-09-25:
> Not work 38794

The PR that could fix it is 38815 so you have to use build higher than that. There's a new Windows weekly 38830 from 10 hours ago so please try there.

https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

davesrocketshop, 2024-09-25:
Looking at the error message, it appears to be a problem with the format statement. My guess is that you have multiple items selected which is causing the index to go out of range.

This is not likely materials related.

In future, I would recommend cutting and pasting the error rather than submitting pictures. I'm unable to read anything useful from many of the pictures posted.

Roberto8303, 2024-09-25:
09:51:54  pyException: Traceback (most recent call last):
  File "C:\Users\privato\Downloads\FreeCAD_weekly-builds-38794-conda-Windows-x86_64-py311\Mod\Fem\femviewprovider\view_material_common.py", line 58, in setEdit
    super().setEdit(vobj, mode, task_material_common._TaskPanel)
  File "C:\Users\privato\Downloads\FreeCAD_weekly-builds-38794-conda-Windows-x86_64-py311\Mod\Fem\femviewprovider\view_base_femobject.py", line 89, in setEdit
    task = TaskPanel(vobj.Object)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\privato\Downloads\FreeCAD_weekly-builds-38794-conda-Windows-x86_64-py311\Mod\Fem\femtaskpanels\task_material_common.py", line 182, in __init__
    self.selectionWidget = selection_widgets.GeometryElementsSelection(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\privato\Downloads\FreeCAD_weekly-builds-38794-conda-Windows-x86_64-py311\Mod\Fem\femguiutils\selection_widgets.py", line 235, in __init__
    self.initUI()
  File "C:\Users\privato\Downloads\FreeCAD_weekly-builds-38794-conda-Windows-x86_64-py311\Mod\Fem\femguiutils\selection_widgets.py", line 275, in initUI
    ).format("<br>", "<b>", self.sel_elem_text, "</b>")
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'IndexError'>: Replacement index 4 out of range for positional args tuple

Roberto8303, 2024-09-25:
11:55:22  pyException: Traceback (most recent call last):
  File "C:\Users\privato\Downloads\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\Mod\Fem\femviewprovider\view_material_common.py", line 58, in setEdit
    super().setEdit(vobj, mode, task_material_common._TaskPanel)
  File "C:\Users\privato\Downloads\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\Mod\Fem\femviewprovider\view_base_femobject.py", line 89, in setEdit
    task = TaskPanel(vobj.Object)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\privato\Downloads\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\Mod\Fem\femtaskpanels\task_material_common.py", line 182, in __init__
    self.selectionWidget = selection_widgets.GeometryElementsSelection(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\privato\Downloads\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\Mod\Fem\femguiutils\selection_widgets.py", line 235, in __init__
    self.initUI()
  File "C:\Users\privato\Downloads\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\Mod\Fem\femguiutils\selection_widgets.py", line 275, in initUI
    ).format("<br>", "<b>", self.sel_elem_text, "</b>")
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'IndexError'>: Replacement index 4 out of range for positional args tuple
11:55:25  pyException: Traceback (most recent call last):
  File "C:\Users\privato\Downloads\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\Mod\Fem\femviewprovider\view_material_common.py", line 58, in setEdit
    super().setEdit(vobj, mode, task_material_common._TaskPanel)
  File "C:\Users\privato\Downloads\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\Mod\Fem\femviewprovider\view_base_femobject.py", line 89, in setEdit
    task = TaskPanel(vobj.Object)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\privato\Downloads\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\Mod\Fem\femtaskpanels\task_material_common.py", line 182, in __init__
    self.selectionWidget = selection_widgets.GeometryElementsSelection(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\privato\Downloads\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\Mod\Fem\femguiutils\selection_widgets.py", line 235, in __init__
    self.initUI()
  File "C:\Users\privato\Downloads\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\Mod\Fem\femguiutils\selection_widgets.py", line 275, in initUI
    ).format("<br>", "<b>", self.sel_elem_text, "</b>")
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'IndexError'>: Replacement index 4 out of range for positional args tuple

Roberto8303, 2024-09-25:
**It doesn't even work with the latest revision**

luzpaz, 2024-09-25:
@Roberto8303 JFYI, you can put code or output in block format by encapsulating it with 3 backticks. This maintains the spacing and format. I've modified your previous 2 comments this way. 

```
This
is
an
Example
```

Roberto8303, 2024-09-25:
"19:41:50  Material card chosen:
    C:/Users/rob/Downloads/FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311/FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311/data/Mod/Material/Resources/Materials/Standard/Metal/Aluminum/AlMg3F24.FCMat
19:41:50  
19:41:50  pyException: Traceback (most recent call last):
  File "C:\Users\rob\Downloads\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\Mod\Fem\femviewprovider\view_material_common.py", line 58, in setEdit
    super().setEdit(vobj, mode, task_material_common._TaskPanel)
  File "C:\Users\rob\Downloads\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\Mod\Fem\femviewprovider\view_base_femobject.py", line 89, in setEdit
    task = TaskPanel(vobj.Object)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\rob\Downloads\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\Mod\Fem\femtaskpanels\task_material_common.py", line 182, in __init__
    self.selectionWidget = selection_widgets.GeometryElementsSelection(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\rob\Downloads\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\Mod\Fem\femguiutils\selection_widgets.py", line 235, in __init__
    self.initUI()
  File "C:\Users\rob\Downloads\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\Mod\Fem\femguiutils\selection_widgets.py", line 275, in initUI
    ).format("<br>", "<b>", self.sel_elem_text, "</b>")
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'IndexError'>: Replacement index 4 out of range for positional args tuple
"

Roberto8303, 2024-09-25:
`19:41:50  Material card chosen:
    C:/Users/rob/Downloads/FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311/FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311/data/Mod/Material/Resources/Materials/Standard/Metal/Aluminum/AlMg3F24.FCMat
19:41:50  
19:41:50  pyException: Traceback (most recent call last):
  File "C:\Users\rob\Downloads\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\Mod\Fem\femviewprovider\view_material_common.py", line 58, in setEdit
    super().setEdit(vobj, mode, task_material_common._TaskPanel)
  File "C:\Users\rob\Downloads\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\Mod\Fem\femviewprovider\view_base_femobject.py", line 89, in setEdit
    task = TaskPanel(vobj.Object)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\rob\Downloads\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\Mod\Fem\femtaskpanels\task_material_common.py", line 182, in __init__
    self.selectionWidget = selection_widgets.GeometryElementsSelection(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\rob\Downloads\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\Mod\Fem\femguiutils\selection_widgets.py", line 235, in __init__
    self.initUI()
  File "C:\Users\rob\Downloads\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\Mod\Fem\femguiutils\selection_widgets.py", line 275, in initUI
    ).format("<br>", "<b>", self.sel_elem_text, "</b>")
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'IndexError'>: Replacement index 4 out of range for positional args tuple
 `

Roberto8303, 2024-10-02:
Is there any news regarding this problem?

luzpaz, 2024-10-02:
@davesrocketshop errors have been pasted above. If you have a moment to diagnose ?

davesrocketshop, 2024-10-02:
As I mentioned earlier, there's a problem in the FEM UI code. This does not appear to be related to the materials subsystem.

Roberto8303, 2024-10-02:
@davesrocketshop I'm sorry, I don't understand, is this a problem with my PC or something I set up wrong?

davesrocketshop, 2024-10-02:
> @davesrocketshop I'm sorry, I don't understand, is this a problem with my PC or something I set up wrong?

It does appear that there is a multi-selection happening. I don't think the UI allows for this so I'm not sure how it's happening. If you build yourself, a clean operation before rebuilding might clear it up. If you've downloaded a pre-release then I can't explain it. Either way, it looks like the file you're working with may have become corrupted somehow.

As for the error itself, if the selected object contains multiple items then the format statement will throw an error. This can be caught, but the error would probably manifest somewhere else. Seeing the contents of `sel_elem_text` might provide a clue. Try adding

 `print(self.sel_elem_text)`

before the line containing the format statement

Roberto8303, 2024-10-02:
`print(self.sel_elem_text)

prima della riga contenente l'istruzione di formato`

where is it located?

davesrocketshop, 2024-10-02:
C:\Users\rob\Downloads\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\Mod\Fem\femguiutils\selection_widgets.py", line 275, in initUI

Change (starting on line 270):

```
        self._helpTextLbl = QtGui.QLabel()
        self._helpTextLbl.setWordWrap(True)
        helpTextPart1 = self.tr(
            'Click on "Add" and select geometric elements to add them to the list.{}'
            "The following geometry elements can be selected: {}{}{}"
        ).format("<br>", "<b>", self.sel_elem_text, "</b>")

```
to

```
        self._helpTextLbl = QtGui.QLabel()
        self._helpTextLbl.setWordWrap(True)
        print(self.sel_elem_text)
        helpTextPart1 = self.tr(
            'Click on "Add" and select geometric elements to add them to the list.{}'
            "The following geometry elements can be selected: {}{}{}"
        ).format("<br>", "<b>", self.sel_elem_text, "</b>")

```

davesrocketshop, 2024-10-02:
> `print(self.sel_elem_text)
> 
> prima della riga contenente l'istruzione di formato`
> 
> where is it located?

Isn't everyone an expert in Python? :)

Roberto8303, 2024-10-02:
"15:47:26  Material card chosen:
    C:/Users/privato/Desktop/FreeCAD_weekly-builds-38871-conda-Windows-x86_64-py311/data/Mod/Material/Resources/Materials/Standard/Metal/Aluminum/AlMg3F24.FCMat
15:47:26  
15:47:26  pyException: Traceback (most recent call last):
  File "C:\Users\privato\Desktop\FreeCAD_weekly-builds-38871-conda-Windows-x86_64-py311\Mod\Fem\femviewprovider\view_material_common.py", line 58, in setEdit
    super().setEdit(vobj, mode, task_material_common._TaskPanel)
  File "C:\Users\privato\Desktop\FreeCAD_weekly-builds-38871-conda-Windows-x86_64-py311\Mod\Fem\femviewprovider\view_base_femobject.py", line 89, in setEdit
    task = TaskPanel(vobj.Object)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\privato\Desktop\FreeCAD_weekly-builds-38871-conda-Windows-x86_64-py311\Mod\Fem\femtaskpanels\task_material_common.py", line 182, in __init__
    self.selectionWidget = selection_widgets.GeometryElementsSelection(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\privato\Desktop\FreeCAD_weekly-builds-38871-conda-Windows-x86_64-py311\Mod\Fem\femguiutils\selection_widgets.py", line 235, in __init__
    self.initUI()
  File "C:\Users\privato\Desktop\FreeCAD_weekly-builds-38871-conda-Windows-x86_64-py311\Mod\Fem\femguiutils\selection_widgets.py", line 275, in initUI
    ).format("<br>", "<b>", self.sel_elem_text, "</b>")
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'IndexError'>: Replacement index 4 out of range for positional args tuple
"
Done but I don't see any difference

Roberto8303, 2024-10-02:
I tried with another PC where I didn't have freecad and I got the same error

davesrocketshop, 2024-10-02:
> I tried with another PC where I didn't have freecad and I got the same error

That's good to know. Consistent errors are easier to fix

Roberto8303, 2024-10-02:
`16:14:59  Material card chosen:
    C:/Users/privato/Desktop/FreeCAD_weekly-builds-38871-conda-Windows-x86_64-py311/data/Mod/Material/Resources/Materials/Standard/Metal/Aluminum/AlMg3F24.FCMat
16:14:59  
16:14:59  pyException: Traceback (most recent call last):
  File "C:\Users\privato\Desktop\FreeCAD_weekly-builds-38871-conda-Windows-x86_64-py311\Mod\Fem\femviewprovider\view_material_common.py", line 58, in setEdit
    super().setEdit(vobj, mode, task_material_common._TaskPanel)
  File "C:\Users\privato\Desktop\FreeCAD_weekly-builds-38871-conda-Windows-x86_64-py311\Mod\Fem\femviewprovider\view_base_femobject.py", line 89, in setEdit
    task = TaskPanel(vobj.Object)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\privato\Desktop\FreeCAD_weekly-builds-38871-conda-Windows-x86_64-py311\Mod\Fem\femtaskpanels\task_material_common.py", line 182, in __init__
    self.selectionWidget = selection_widgets.GeometryElementsSelection(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\privato\Desktop\FreeCAD_weekly-builds-38871-conda-Windows-x86_64-py311\Mod\Fem\femguiutils\selection_widgets.py", line 235, in __init__
    self.initUI()
  File "C:\Users\privato\Desktop\FreeCAD_weekly-builds-38871-conda-Windows-x86_64-py311\Mod\Fem\femguiutils\selection_widgets.py", line 275, in initUI
    ).format("<br>", "<b>", self.sel_elem_text, "</b>")
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'IndexError'>: Replacement index 4 out of range for positional args tuple
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QDragManager::drag in possibly invalid state
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QDragManager::drag in possibly invalid state
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QDragManager::drag in possibly invalid state
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QDragManager::drag in possibly invalid state
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QDragManager::drag in possibly invalid state
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QDragManager::drag in possibly invalid state
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QDragManager::drag in possibly invalid state
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QPixmap::scaled: Pixmap is a null pixmap
16:15:09  QDragManager::drag in possibly invalid state
16:15:10  QPixmap::scaled: Pixmap is a null pixmap
16:15:10  QDragManager::drag in possibly invalid state
16:15:10  QPixmap::scaled: Pixmap is a null pixmap
16:15:10  QDragManager::drag in possibly invalid state
16:15:10  QPixmap::scaled: Pixmap is a null pixmap
16:15:10  QDragManager::drag in possibly invalid state
16:15:10  QPixmap::scaled: Pixmap is a null pixmap
16:15:10  QPixmap::scaled: Pixmap is a null pixmap
16:15:10  QDragManager::drag in possibly invalid state
16:15:10  QPixmap::scaled: Pixmap is a null pixmap
16:15:10  QDragManager::drag in possibly invalid state
16:15:10  QPixmap::scaled: Pixmap is a null pixmap
16:15:10  QDragManager::drag in possibly invalid state
16:15:10  QPixmap::scaled: Pixmap is a null pixmap
16:15:10  QPixmap::scaled: Pixmap is a null pixmap
16:15:10  QPixmap::scaled: Pixmap is a null pixmap
16:15:10  QPixmap::scaled: Pixmap is a null pixmap
16:15:10  QPixmap::scaled: Pixmap is a null pixmap
`

Roberto8303, 2024-10-02:
I don't understand if I copy and replace the fem and material folder from freecad 0.22 where it works, and I copy it to the new freecad 1.0 version I still get this error

davesrocketshop, 2024-10-02:
I found the problem, and it isn't the code. It's the Italian translation.

The original string contains four sets of braces `{}` meaning it expects four arguments. The Italian translation added an extra set of braces, meaning it expected five arguments. The error message seemed to indicate there were too many arguments when in fact there were too few.

Fixing this is both easy and complicated. Editing the translation string is an easy fix, but since the translations are handled on another system that needs to be coordinated so that the fix isn't overwritten. I'll see if that can get updated today.

Roberto8303, 2024-10-02:
so you're saying that since I have the Italian language set on all my PCs it doesn't work?

davesrocketshop, 2024-10-02:
> so you're saying that since I have the Italian language set on all my PCs it doesn't work?

Yes. There is an error in the translation string. If you switch to any other language it should work fine

Roberto8303, 2024-10-02:
so can it be solved and made to work in Italian too?

davesrocketshop, 2024-10-02:
> so can it be solved and made to work in Italian too?

I've already updated the translation in CrowdIn. The next time translations are pulled it should be correct.

Roberto8303, 2024-10-02:
so the next weekly build?

davesrocketshop, 2024-10-02:
> so the next weekly build?

It should be. I've submitted a PR that fixes it.

FEA-eng, 2024-10-02:
> I found the problem, and it isn't the code. It's the Italian translation.

Wow, good find, thanks. I haven't thought about that. But it explains why we couldn't reproduce it.

PR #16969 (2024-10-03): FEM: Translation error causing a crash
An error in the Italian translation added an extra '{}' to the translation string resulting in a dump when formatting.

This has also been updated in CrowdIn but as it's causing a crash is also beiung changed via this PR. It should updated correctly when doing the next translation pull.

Fixes #16704
commit eef681f0 (2024-10-03): FEM: Translation error causing a crash

An error in the Italian translation added an extra '{}' to the
translation string resulting in a dump when formatting.

This has also been updated in CrowdIn but as it's causing a crash is
also beiung changed via this PR. It should updated correctly when
doing the next translation pull.

Fixes #16704
commit dbf849a5 (2024-10-03): FEM: Translation error causing a crash

An error in the Italian translation added an extra '{}' to the
translation string resulting in a dump when formatting.

This has also been updated in CrowdIn but as it's causing a crash is
also beiung changed via this PR. It should updated correctly when
doing the next translation pull.

Fixes #16704
closed 2024-10-03 by chennes