FreeCAD/FreeCAD #16706: new show delta measurement update not working as it should

Roberto8303, 2024-09-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

selecting the circles the measure tool with show delta active gives an incorrect indication on the axes. The measurement must be taken in line with the Y X axes not by chance along the circumference of the hole.
[image]
[image]

### Full version info

```shell
OS: Windows 10 build 19042
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.67
  * Assembly3 0.12.2
  * Assembly4 0.50.15
  * btl 0.9.9
  * CurvedShapes 1.0.10
  * Curves 0.6.47
  * dxf-library
  * ExplodedAssembly
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * PieMenu 1.8.0
  * POV-Ray-Rendering
  * QuickMeasure 2022.10.28
  * Render 2024.8.4
  * SearchBar 1.0.2
  * sheetmetal 0.4.26
  * TitleBlock 0.5.2.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-09-21:
That's the current limitation of this tool itself, not just the projections - it can't snap to the center of the hole or any specific point. It just takes the closest points. But there's a request to improve it in the future: #13708.

Roberto8303, 2024-09-21:
ok fine thank you so much for this current opportunity

luzpaz, 2024-09-23:
Closing as duplicate.

closed 2024-09-23 by luzpaz