FreeCAD/FreeCAD #16707: CAM: New Simulator's task panel cuts translations for Job

FEA-eng, 2024-09-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've noticed that the translation for `Job:` is cut in the task panel of the New Simulator:

[image]

[image]

It should be `Zadanie:`. The same problem occurs with other languages.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-22:
@kaktusus have you seen this before ?

kaktusus, 2024-09-22:
It wasn't until Kuba showed it to me.

[image]

On my machine I do not see this error :no_good:
I checked on releases from 38794 to 38467.

kaktusus, 2024-09-22:
The problem is with the new CAM Simulation.

[image]

luzpaz, 2024-09-22:
@shaise ping

shaise, 2024-09-22:
Sorry, did not see that I'll take a look.

shaise, 2024-09-22:
OK. Added a fix to my latest PR #16665.

[image]

FEA-eng, 2024-09-22:
> OK. Added a fix to my latest PR #16665.

Great, thanks.

kaktusus, 2024-09-23:
And is it possible to add another fix?
So that the untranslated "High" gets a translation :question:

shaise, 2024-09-23:
No need to fix. High, Medium and Low are all translateable

FEA-eng, 2024-10-12:
Fixed by #16665.

closed 2024-10-12 by FEA-eng