FreeCAD/FreeCAD #16708: Memory Leak when adding alias to spreadsheet field

BrettLHolmes, 2024-09-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When adding any data or an  Alias to a Spreadsheet field the memory used often increases by 1-2,5GB and is only released when closing the file in use 
Very quickly the system runs out of memory with MacOS asking to shut down applications
When Opening the same file the memory utilisation is back to normal usage
I have a screen recording of this happening showing memory consumption (if it helps) and can share it

The same issue occurs on both Intel and M1 Mac with Sequoia on Both Platforms
I have created a New File and attempted the same operations (adding values or adding Alias) and it does not seem to occur so I have included the file (its really simple) that I was using)

** Additionally I noticed that Freecad RC1 while it multithreaded on Mac OS it appears Bound to only one CPU ** 
** apologies if this is a duplicate - I Did search but could not see any spreadsheet memory related issues **

### Full version info

```shell
OS: macOS 15.0
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)

OS: macOS 15.0
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * Gridfinity 0.5.0
  * A2plus 0.4.67
```

### Subproject(s) affected?

Spreadsheet

### Anything else?

[usb_rs232Case-v3.FCStd.zip](https://github.com/user-attachments/files/17085060/usb_rs232Case-v3.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-09-22:
A spreadsheet in _FreeCAD_ contains 26 * 26 * 16 384 of cells. When you do not affect a cell, the latter does not occupy memory, having default parameters. When you change something with it, it has non-default parameters and occupies memory. You changed the foreground color of all the cells, thus making the spreadsheet occupy 26 * 26 * 16 384 * 700 bytes of memory (≈7 GiB). Of course, _FreeCAD_ eats RAM and slows down when going through this memory bulk. Compression usually deflates recurring patterns incredibly good, therefore, when you save the file, all these 7 GiB of memory are condensed into 30 MiB.

All hail _ZIP_!

BrettLHolmes, 2024-09-22:
Summary: changing all the cells foreground in the spreadsheet unnecessarily increased the amount of memory Freecad needed for each cell not a bug

Thank you for this I will close this issue as it is not a bug but PEBKAC

closed 2024-09-22 by BrettLHolmes