FreeCAD/FreeCAD #16709: Crash when plotting

facorazza, 2024-09-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Whenever creating a plot FreeCAD crashes.

How to reproduce?

Execute the example code mentioned [here](https://wiki.freecad.org/Plot_Basic_tutorial)

```python
try:
    from FreeCAD.Plot import Plot
except ImportError:
    from freecad.plot import Plot

Plot.figure("TrigonometricTest")
```

### Full version info

```shell
OS: KDE Flatpak runtime (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.5.28
  * Render 2024.8.4
  * Cfd
  * Ship 2022.4.11
  * CfdOF 1.27.7
  * CurvedShapes 1.0.9
  * Plot 2022.4.17
  * woodworking 0.21.2.33771
```

and here's the log:

```
Computing transversal areas...
Some Inventor representation errors can be shown, please ignore them.
1 / 28
2 / 28
3 / 28
4 / 28
5 / 28
6 / 28
7 / 28
8 / 28
9 / 28
10 / 28
11 / 28
12 / 28
13 / 28
14 / 28
15 / 28
16 / 28
17 / 28
18 / 28
19 / 28
20 / 28
21 / 28
22 / 28
[... 71 lines cut ...]
#63  0x75a4c1ee32a1 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x191
#64  /usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x339e93) [0x75a4c1f39e93]
#65  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x239) [0x75a4c0117e59]
#66  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xbe2b8) [0x75a4c01752b8]
#67  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x75a4c0115513]
#68  0x75a4c1f39586 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x66
#69  0x75a4c1ede73b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#70  0x75a4c1ee6bd0 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x90
#71  0x75a4c4981b25 in Gui::Application::runApplication() from /app/freecad/lib/libFreeCADGui.so+0x1e75
#72  FreeCAD(+0x9387) [0x63b2414e6387]
#73  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x2808a) [0x75a4c163b08a]
#74  /usr/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x75a4c163b14b]
#75  FreeCAD(+0x96c5) [0x63b2414e66c5]
```

Linked issue https://github.com/FreeCAD/freecad.ship/issues/60
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-09-21:
I can't reproduce

> OS: Debian GNU/Linux trixie/sid (XFCE/xfce)
> Word size of FreeCAD: 64-bit
> Version: 1.1.0dev.38781 (Git)
> Build type: Debug
> Branch: testPad
> Hash: f82bfd5ddcddfa96359789093190c639ab088e24
> Python 3.12.6, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.2.dev
> Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
> Stylesheet/Theme/QtStyle: unset/Classic/Qt default

facorazza, 2024-09-21:
> I can't reproduce
> 
> > OS: Debian GNU/Linux trixie/sid (XFCE/xfce)
> > Word size of FreeCAD: 64-bit
> > Version: 1.1.0dev.38781 (Git)
> > Build type: Debug
> > Branch: testPad
> > Hash: [f82bfd5](https://github.com/FreeCAD/FreeCAD/commit/f82bfd5ddcddfa96359789093190c639ab088e24)
> > Python 3.12.6, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.2.dev
> > Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
> > Stylesheet/Theme/QtStyle: unset/Classic/Qt default

Are you using the flatpak release?

luzpaz, 2024-09-22:
You're using `0.21.2.33771`  
Try reproducing on the latest 1.0RC1

luzpaz, 2024-09-30:
@facorazza flatpak doesn't support running latest Release Candidate. Can you install Snap or AppImage and test ?

* https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2
* Or run the Snap 'edge' (https://wiki.freecad.org/Ubuntu_Snap)

facorazza, 2024-10-02:
> @facorazza flatpak doesn't support running latest Release Candidate. Can you install Snap or AppImage and test ?
> 
>     * https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2
> 
>     * Or run the Snap 'edge' (https://wiki.freecad.org/Ubuntu_Snap)

The problem might be with the flatpak release. I'm using ondsel and plotting works fine

luzpaz, 2024-10-11:
Can you test RC2 to be certain ? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2)

closed 2024-10-11 by luzpaz