FreeCAD/FreeCAD #16710: FreeCAD inconsistent and unintuitive about adding objects to Std_Part

theosib, 2024-09-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I like to use Std_Part to group different sets of objects into collections that can be positioned relative to each other. For example: I might make a container in one Std_Part and the lid in another Std_Part, and then I can move and rotate the whole lid on top of the base to check the fit. I find this grouping ability of Std_Part as well as the coordinate axes and planes to be very useful.

Now to the bug: I find it frustrating to work with Std_Part because of how FreeCAD chooses to parent new objects. Examples:

- I select the XY (or any other) coordinate plane of the Std_Part and create a sketch. The sketch ends up at the top of the hierarchy instead of inside the Std_Part. This is especially frustrating when I want to reference external geometry that's inside an Std_Part. I have to create the sketch, exit Sketcher, drag the sketch into the Std_Part, and then go back into the sketch. Every time. Very tedious.
- I have multiple solids inside of an Std_Part and perform a boolean. Once again, the new boolean ends up at the top of the hierarchy instead of inside the Std_Part, requiring that I drag it in manually.

But entirely inconsistent with the above:

- Let's say I have a boolean in at the top of the hierarchy, but there is a single child object somewhere inside its tree that belongs to a Std_Part. A new Draft array of that object will end up automatically inside the Std_Part.

Due to the unintuitive and inconsistent behavior, I think perhaps this could benefit from some attention. Thanks.

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 (Git)
Build type: Release
Branch: main
Hash: 758674d40be45e365f4148fb2a5ea6bd6f93bc74
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * Curves 0.6.44
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-09-21:
I cannot verify the bug on my setup, the following steps works for me:

1. Open freecad
2. New document
3. Select Part Design Workbench
4. Create Part (Std_Part)
5. Expand Origin element in tree
6. Select XY-Plane
7. Still in Part Design Workbench, Click Create Sketch
8. On Dialog "Active Body Required", preselected option is "Create new body", Click OK

That results in the following tree (the sketch is hidden inside the body):
[image]

theosib, 2024-09-24:
I'm not using Part Design. I'm using Part Workbench. Perhaps if a Part Design Body is inside a Std_Part, it works differently.

hyarion, 2024-09-24:
I can now verify the bug on my setup, these are the following steps:

1. Open freecad
2. New document
3. Select _Part_ Workbench
4. Create Part (Std_Part)
5. Expand Origin element in tree
6. Select XY-Plane
7. Still in Part Workbench, Click Create Sketch
8. On Dialog "Sketch attachment", select preselected option is "Plane face", Click OK
9. Draw a line
10. Exit sketch

Confirmed!

That results in the following tree (the sketch is _not_ inside the Std_Part):
[image]

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38826 +1 (Git)
Build type: Release
Branch: spreadsheet-tree-icon
Hash: 72103382b05bb05372e6bccf5fcb01651fca4ffa
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CurvedShapes 1.0.10
  * sheetmetal 0.2.61
  * Curves 0.6.6
```

The same thing happens in the Sketcher workbench too.

The reason it works in PartDesign is that the PartDesign sketcher button calls another, PartDesign-specific command.

This is where the sketch is created when performing these steps: https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Sketcher/Gui/Command.cpp#L233

One solution could be to find which Part is the active and add the sketch there instead of on `App.activeDocument()`, but this is not just limited to sketches, the same applies to other Std_Parts as well (and probably other objects too).

Though this being an annoyance, I wouldn't concider this a bug, rather a nice feature that is missing 🙂. 

I don't know what would be the best way forward as it would require changing how all objects that can be placed in a Std_Part would be created.

Any suggestions?

theosib, 2024-11-04:
One weird part of this is that if the object I create is a Draft Array, then it ends up inside the Std_Part. Even if the object I'm making an array of is outside of the Std_Part. Now, usually, I start my sketches on a plane of the Std_Part, so a Std_Part is usually in the ancestry in situations like this. I haven't tested what happens if there's an Std_Part, and I make an array, but none of the ancestry is in the Std_Part.

theosib, 2024-11-08:
Here's an example of a case where the handling of has a result that is completely nuts. Please see the attached design file. Got to the Std_Part called "base" and select "divider slots". Then go to the Draft workbench and make an array out of it. You can just use the default values for this test case. When I do this, what I see is that basically all of the contents of "base" seem to entirely disappear. But what really happens is that they all get moved to the other Std_Part called "lid" and I have to move them back.

I think this is a bona fide bug. There's no good reason why creating a draft array should be just randomly picking a Std_Part to put its results in.

[tea storage.FCStd.zip](https://github.com/user-attachments/files/17681960/tea.storage.FCStd.zip)

recursiveGecko, 2025-07-05:
> I don't know what would be the best way forward as it would require changing how all objects that can be placed in a Std_Part would be created.

> Any suggestions?

Intuitively I would expect that only one Part container can be the "active object", and that the active Part container is the recipient of all compatible operations/results. I can't think of any exceptions.

FEA-eng, 2026-01-12:
Still relevant [these steps](https://github.com/FreeCAD/FreeCAD/issues/16710#issuecomment-2372544980)

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260106 (Git shallow)
Build date: 2026/01/06 15:36:19
Build type: Release
Branch: (HEAD detached at 9b64da8)
Hash: 9b64da827a112d88a025be26316e3d023ff491dc
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```