FreeCAD/FreeCAD #16711: TechDraw_DraftView: view is rotated 90 degrees

furgo16, 2024-09-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

See the attached file.

When inserting a set of objects with [TechDraw_DraftView](https://wiki.freecad.org/TechDraw_DraftView) into a TechDraw page, they appear with 90° rotation from the 3D view. This is unexpected, since the documentation mentions: 

> The camera direction in the 3D view determines the initial value of the DataDirection property of the View 

[image]

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git) Snap 1144
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

semhustej, 2024-09-22:
I can reproduce this behavior. For each camera direction there seems to be default rotation of the view, which the user can not change by rotating the origin 3D view.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

maxwxyz, 2024-09-22:
@Roy-043 @Rexbas FYI

Roy-043, 2025-01-09:
Fixing this for the standard orientations is relatively easy. But a DraftView only has a Direction property. The XDirection property is missing. Any rotation around the view direction is therefore ignored. There is also backward compatibility that has to be considered. If the algorithm that determines the XDirection is changed, some DraftViews in older files will change unexpectedly. Only top view DraftViews would remain the same.

The best solution would be to add the XDirection property. For older views this could be set to 0,0,0 in which case the old algorithm would be used. This would also require some TechDraw mods.

@WandererFan What are your thoughts on this?

WandererFan, 2025-10-17:
I might be missing something,  but Symbols (including DraftView) have a Rotation property that will rotate the view around the projection direction.

[image]

Roy-043, 2025-10-21:
Related/duplicate: #16978.