FreeCAD/FreeCAD #16712: [UI] Additional lines in translation affect a Preference visualitzation

MiqCG, 2024-09-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Catalan language, in Preferences > Import - Export > STEP > Export invisible objects, there is a misalignment between the check box and its text. See attached images.

I have searched in Crowdin and the translation of this string had 2 additional lines at the end. I have corrected the translation and leaved a message to _kaktus_
I don't know if there is something else that I can do to update the translation to be available for 1.0 release. The problem is present in 1.0.0RC1 and before.

In Catalan

[image]

In English for comparison
[image]

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git) AppImage
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-22:
Thanks. If you've made the change in crowdin, it should be in the next translation update. @yorikvanhavre will there be another translation update for RC1 ?

kaktusus, 2024-09-22:
The problem relates directly to string translation for Catalan.
https://crowdin.com/editor/freecad/563/en-ca?view=comfortable#6590971

MiqCG did the revision of the translation. :wink:

luzpaz, 2024-09-22:
Closing as it should be fixed in crowdin.Please re-open if it persists.

closed 2024-09-22 by luzpaz