FreeCAD/FreeCAD #16715: Spreadsheet: Merge Cells Bug

audiochris, 2024-09-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Selecting two cells and using the merge cells icon in the toolbar or the right click menu does not merge the cells unless you already three or more cells merged. 

Troubleshooting and Steps to reproduce:

1. Create a new spreadsheet
2. Select two adjacent cells in the same row.
3. Click the "Merge Cells" icon in the toolbar ; Result: Cells not Merged
4. Select "Merge cells" from the right click menu; Result: Cells not merged
5. Select three or more adjacent cells in the same row.
6. Click "Merge Cells" from the toolbar or right click menu; Result: Cells Merged
7. Split merged cells
8. Select two adjacent cells in the same row.
9. Click "Merge Cells" from the toolbar or right click menu; Result: Cells Not Merged
10. Select three or more adjacent cells in the same row.
11. Click "Merge Cells" from the toolbar or right click menu; Result: Cells Merged
12. Select two adjacent cells in the same row.
13. Click "Merge Cells" from the toolbar or right click menu; Result: Cells Merged
14. Split the three cells that were merged together in step 6
15. Select two adjacent cells in the same row.
16. Click "Merge Cells" from the toolbar or right click menu; Result: Cells Merged

### Full version info

```shell
OS: macOS 15.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
```

### Subproject(s) affected?

Spreadsheet

### Anything else?

Nothing more to add.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-09-21:
I don't seem to be able to reproduce the bug.

I tried step 1, 2, 3/4 and got two merged cells both using the context menu and from the toolbar
It works for me in both RC1 and in my debug build.

@audiochris Here's one of my test sessions. Am I missing something?

https://github.com/user-attachments/assets/88b61238-d306-458a-bb21-44ecfb4d78ca

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * Manipulator 1.5.7
  * CurvedShapes 1.0.10
  * sheetmetal 0.2.61
  * Curves 0.6.6
```

Dev build, in a fresh state (as far as I know)
```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 +3 (Git)
Build type: Release
Branch: safe-mode
Hash: 9ec91fe4493ef86e1141530ee4d66a07de752a2f
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

audiochris, 2024-09-21:
I have deleted all preferences bringing me to default FreeCAD settings. I can reproduce this every time on the MacOS ARM version. Haven't tried x86 with Rosetta yet. FreeCAD file attached.

[spreadsheet_test.FCStd.zip](https://github.com/user-attachments/files/17085809/spreadsheet_test.FCStd.zip)

hyarion, 2024-09-21:
I use macbook pro with m1 pro and arm version too.

The attached file looks like an new empty spreadsheet. What do you want to show with the file?

audiochris, 2024-09-21:
The file is a newly created file with an empty spreadsheet just after deleting preference files in ~/Library/Preferences and ~/Library/Application Support. The issue was still occurring and posting the file was an attempt to demonstrate that issue. However, if you're not running into this issue, perhaps is something that was introduced in the newest OS. I can't say when this issue started but it was defiantly present for me in build 38728.

xtemp09, 2024-09-22:
@audiochris, can you find out the build, with which the issue began, to help the developers to narrow down the commit introducing the bug?

closed 2025-08-09 by github-actions[bot]