FreeCAD/FreeCAD #16717: Placement of a shape is ignoring the input

MarcAndroidmarket, 2024-09-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It is not possible to input placement value of a sketch:
- not via "placement..." dialog (accepts the input, but does not use it in the properties of the sketch after pressing ok)
- not directly in the properties of the sketch (input cursor does not accept any keyboard input)

Steps to recreate:
1. Start with a new model
2. Create a new body
3. Create a new sketch (x-y-plane) within this body (e.g. a rectangle)
4. Use placement dialog and try to change the z-axis value. After pressing ok, no change in the properties of the sketch. Alternatively try changing the z-axis value of placement->position directly (no input accepted from keyboard)
[image]

### Full version info

```shell
OS: EndeavourOS (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: makepkg
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Installed mods: 
  * boltsfc 2022.11.5
  * fasteners 0.5.29
  * kicadStepUpMod 11.3.2
  * A2plus 0.4.67
  * LCInterlocking
  * sheetmetal 0.4.25
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-09-21:
This is by design, you can only apply Placement to the Body, if you wish to move/rotate the Sketch inside a Body then you have to use the Attachment Offset because it's been given support of the Plane, in this case X-Y. See https://wiki.freecad.org/Placement

MarcAndroidmarket, 2024-09-22:
Thanks, my mistake.

closed 2024-09-22 by MarcAndroidmarket