FreeCAD/FreeCAD #16718: Program Hang when Deselecting Hept and Octagon Tool

CDRXavier, 2024-09-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD hangs on right-click with Heptagon and Octagon tool in sketcher.

It won't crash if you use the default hexagon tool, or the triangle, or the rectangle, or the pentagon. Or using the polygon tool with 9 sides.

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-22:
@CDRXavier can you reproduce on 1.0RC1 ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc1

closed 2024-09-22 by luzpaz

CDRXavier, 2024-10-04:
> > Version: 0.20.2.29603 (Git)
> 
> Ancient version. @CDRXavier can you reproduce on 1.0RC1 ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc1

I tried RC2. Seemingly fixed with the new sketcher.