FreeCAD/FreeCAD #16719: BIM: Arch Schedule does not find elements inside an array

furgo16, 2024-09-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Open the attached test file
2. Double-click on the Arch schedule in the document tree. Notice it should pick up objects with IFC type: beam
3. Click ok
4. Open spreadsheet in the document tree
5. **Expected**: beams from the model are listed. **Actual**: they are not. Note: if a beam is subsequently added as a single object not part of an array, it is then picked up.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git) Snap 1144
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

semhustej, 2024-09-22:
I can reproduce this behavior.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

maxwxyz, 2024-09-22:
@yorikvanhavre FYI

yorikvanhavre, 2024-09-23:
Indeed Draft arrays are one object and therefore not handled by the schedule tool yet. Other forms of App::Links are also not supported yet. This should be implemented

furgo16, 2024-12-30:
This is a major drawback of the schedule tool. If e.g. beams are to be counted, they need to be added individually, which is not effective once you have a few tenths of them.

@yorikvanhavre what would roughly be required to implement this? I might have a go at it if I manage . Thanks!

furgo16, 2025-01-21:
@yorikvanhavre friendly reminder: would you have some quick guidance on how do you think this could be implemented? Thanks!

PR #19219 (2025-02-05): BIM: add support for array elements in BIM Schedule
This PR adds functionality to the BIM Schedule command to detect and expand array elements, so that they can be further processed. Before, array elements were ignored.

:information_source: Notes:

1. It covers the array type in the test case file provided in #16719. Element(s) belonging to a single ortho Link array, with `expanded=False`.
1. It has also been successfully tested with:
   - Regular arrays
   - Link arrays with `expanded=True`
1. There might be other cases to cover, but as long as this PR does not break them, they can be addres [... cut]
closed 2025-02-05 by yorikvanhavre