FreeCAD/FreeCAD #16721: BIM: Classification Manager's "Only visible objects" option has no effect

furgo16, 2024-09-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Open the uncompressed [attached test file](https://github.com/user-attachments/files/17088574/BeamTest.zip) in FreeCAD
2. Switch to the BIM Workbench
3. Click on the [Classification Manager](https://wiki.freecad.org/BIM_Classification) command
4. Toggle the "Only visible objects" checkbox. In my case, it's enabled by default, so I unchecked it
5. Click Ok
6. Reopen the Classification Manager as in step 3
7. **Expected**: "Only visible objects" retains the previous state (unchecked).
8. **Actual**: "Only visible objects" state is always checked when the Classification Manager dialog is open 

In addition, when the dialog is open, toggling the state of "Only visible objects" does not have any effect on the listed objects within it. The list is updated only when the option is toggled AND the "Sort by" combo box is updated by selecting a sorting method.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git) Snap 1144
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
```

### Subproject(s) affected?

BIM

### Anything else?

At some point I managed to get a Python traceback (list index out of range) when testing this. Unfortunately, I can no longer reproduce it.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

semhustej, 2024-09-22:
I can reproduce this behavior.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

yorikvanhavre, 2024-09-23:
Indeed this checkbox is not tied to any signal and does not perform any action when checking/unchecking. And since it's reset at each launch, there is actually no way to make it work :)

I'll have a look!

PR #19201 (2025-02-04): BIM: Connect "only visible" checkbox signal to corresponding slot
- Connect "only visible" checkbox signal to slot
- Create corresponding "onVisible" slot
- Remember the status of checkbox between dialog launches (store status as a parameter)

Since they were small, this PR also introduces some minor changes, mostly visual:
- Change default ui dialog `Dialog` name to `bimDialogClassification` to make debugging easier
- Add icons to classification dropdown box. Before, some items were indented to accomodate an (inexistent) icon, some others were not

| Before | After |
| --- | --- |
| [image] | [... cut]
closed 2025-02-04 by yorikvanhavre