FreeCAD/FreeCAD #16724: CLI: Sketch breaks when opening file in CLI mode: Geometry "LineSegment" within a PropertyGeometryList was subject to a partial restore.

AIRCAP, 2024-09-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When opening a freecad file in CLI mode, some sketches with certain (yet to be identified) properties don't load correctly. 
- When opening the original file in GUI mode the sketches load and render correctly and can be edited and saved without problem.
- If the file is opened in CLI mode (flag -c) error messages appear on load. If the file is then saved in CLI mode, the affected sketches are broken and can no longer be edited even in GUI mode (invalid constraints!)

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Sketcher

### Anything else?

When opening in GUI mode, no error messages appear in the console and everything works as intended.

In CLI mode the following errors appear when the document is loaded:
```
Geometry "LineSegment" within a PropertyGeometryList was subject to a partial restore.
Geometry "LineSegment" within a PropertyGeometryList was subject to a partial restore.
Geometry "LineSegment" within a PropertyGeometryList was subject to a partial restore.
Geometry "LineSegment" within a PropertyGeometryList was subject to a partial restore.
Geometry "LineSegment" within a PropertyGeometryList was subject to a partial restore.
Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension
Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension
Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension
Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension
Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension
BSpline curve: Knots interval values too close
Property InternalShape of type Part::PropertyPartShape was subject to a partial restore.
Object "Sketch012" was subject to a partial restore. As a result geometry may have changed or be incomplete.
Geometry "LineSegment" within a PropertyGeometryList was subject to a partial restore.
Geometry "LineSegment" within a PropertyGeometryList was subject to a partial restore.
Geometry "LineSegment" within a PropertyGeometryList was subject to a partial restore.
Geometry "LineSegment" within a PropertyGeometryList was subject to a partial restore.
Geometry "LineSegment" within a PropertyGeometryList was subject to a partial restore.
Geometry "LineSegment" within a PropertyGeometryList was subject to a partial restore.
Geometry "LineSegment" within a PropertyGeometryList was subject to a partial restore.
Geometry "LineSegment" within a PropertyGeometryList was subject to a partial restore.
Geometry "LineSegment" within a PropertyGeometryList was subject to a partial restore.
Geometry "LineSegment" within a PropertyGeometryList was subject to a partial restore.
Geometry "LineSegment" within a PropertyGeometryList was subject to a partial restore.
Geometry "LineSegment" within a PropertyGeometryList was subject to a partial restore.
Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension
Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension
Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension
Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension
Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension
BSpline curve: Knots interval values too close
Property InternalShape of type Part::PropertyPartShape was subject to a partial restore.
Object "Sketch013" was subject to a partial restore. As a result geometry may have changed or be incomplete.
Importing project files......
There were errors while loading the file. Some data might have been modified or not recovered at all. Look above for more specific information about the objects involved.
Postprocessing......				(100 %)		
```

When saving the file with App.ActiveDocument.save() or similar, the resulting file has invalid sketches ("invalid constraints" according to sketch validation tool) 

Attached:  reproduction.zip
-- FullyParametricWing.FCStd <-- original file with intact sketches
-- broken.FCStd <-- file with broken sketches after being saved in CLI mode

[reproduction.zip](https://github.com/user-attachments/files/17088915/reproduction.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AIRCAP, 2024-09-22:
Note: the sketch was originally created with FreeCAD version

[code]
OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

AIRCAP, 2024-09-22:
Additional info: the "Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension" appears as soon as the sketch includes any bspline.

The simplest file I managed to create with a sketch and a bspline that - when saved in CLI mode - creates a broken un-editable sketch is this one:
simple.zip contains simple.FCStd with a single bspline and some helper lines to geometrically constrain it.

however the actual error message is a bit different, it says only:

```
Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension
Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension
Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension
Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension
Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension
BSpline curve: Knots interval values too close
```

the file has been created using rc1 only
[simple.zip](https://github.com/user-attachments/files/17089463/simple.zip)

AIRCAP, 2024-09-22:
**To reproduce:**

1. Open affected file from simple.zip with Freecad in CLI mode,
`freecad simple.FCStd -c`
2. Save the file from python
App.ActiveDocument.saveAs("broken.FCStd")
3. Open broken.FCStd in freecad normally

**expected outcome:**
1. sketch in broken.FCStd is just identical to simple.FCStd and editable with a single BSpline and some helper geometry lines and constraints between them

**actual outcome:**
1. sketch cannot be opened - "Invalid Sketch" error
2. sketch validation finds "invalid constraints" - clicking on fix and then editing the sketch yields an "Empty sketch" (no elements and no constraints)

AIRCAP, 2024-09-22:
For reference, the text "**BSpline curve: Knots interval values too close**" is the error text in an exception thrown by an internal function in OpenCascade - `CheckCurveData()` :
https://github.com/Open-Cascade-SAS/OCCT/blob/495a6a642dbfcebd866a6d6f93b4d9c4efa5b415/src/Geom2d/Geom2d_BSplineCurve.cxx#L69 
Since no backtrace is printed this isn't really too helpful. My educated guess is that the bspline is incorrectly loaded/initialized, thus triggering this exception, which in turn might cause other code not to be executed as the exception escalates - in turn breaking the entire sketch.
I wonder to what degree the ViewProvider geometry extension warning is related to that, if at all.

luzpaz, 2024-09-22:
@AjinkyaDahale any chance you can weigh-in regarding b-splines ?

AjinkyaDahale, 2024-09-22:
@luzpaz, I don't have any b-spline specific opinion as of now without looking into the code.

My guess is something that should have been kept in App was instead in Gui.

My first question would be: is this a regression? Can we follow the same steps with 0.21 and get a valid sketch?

AIRCAP, 2024-09-22:
when using the simple.FCStd file as starting point with

[code]
OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

the exact same thing happens. it opens OK in freecad GUI - when opened in CLI the same error messages appear and the file becomes broken - however there is one additional error message:

```
Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension
Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension
Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension
Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension
Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension
BSpline curve: Knots interval values too close
Importing project files......
Postprocessing......				(200 %)		
SketchObject::onChanged(): Unmanaged change of Constraint Property results in invalid constraint indices
```

if its a regression, it would have to be prior to 21

AIRCAP, 2024-09-22:
The regression was introduced with 0.21 - in 0.20 the problem does not occur.

[code]
OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

caveat: the file needs to be opened in 0.20 in GUI mode once and then saved - then it can also be opened and saved in CLI mode in 0.20 -- the resulting file can then also be opened in later versions without issues.

but if the file is opened in 0.20 in CLI mode directly 0.20 segfaults with another error.

opening in 0.20 in GUI mode strips the ViewProviderGeometryExtensions since those were introduced later. the file when saved in 0.20 GUI does not have them - this file then can be opened in CLI with any version - and also in GUI - and loads fine

if the bspline sketch as been saved in GUI with 0.21 or 1.0rc1 - the file has these extensions and it doesn't load correctly in any CLI version

here is the error messages shown when opening the file created with 1.0 rc1 in GUI in 0.20:
```
No extension found of type 'Gui::ViewProviderFaceTexture'
No extension found of type 'PartGui::ViewProviderGridExtension'
```
aside from these error messages the sketch looks fine (it reports some erroneous redundant constraints but that is a different issue) - it can be saved in 0.20 and yields a file that can be loaded in CLI

when opening the original file in 0.20 CLI (without first re-saving it in GUI) I get the following error
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x7f5e37fc8090]
#1  0x7f5e32c76827 in Part::GeometryPersistenceExtension::Restore(Base::XMLReader&) from /tmp/.mount_FreeCARIFEP4/usr/lib/./Part.so+0x27
#2  0x7f5e32c8e1b4 in Part::Geometry::Restore(Base::XMLReader&) from /tmp/.mount_FreeCARIFEP4/usr/lib/./Part.so+0xc4
#3  0x7f5e32c8efc0 in Part::GeomCircle::Restore(Base::XMLReader&) from /tmp/.mount_FreeCARIFEP4/usr/lib/./Part.so+0x30
#4  0x7f5e32b2139b in Part::PropertyGeometryList::Restore(Base::XMLReader&) from /tmp/.mount_FreeCARIFEP4/usr/lib/./Part.so+0x20b
#5  0x7f5e396e32c0 in App::PropertyContainer::Restore(Base::XMLReader&) from /tmp/.mount_FreeCARIFEP4/usr/bin/../lib/libFreeCADApp.so+0x1110
#6  0x7f5e3959fda9 in App::Document::readObjects(Base::XMLReader&) from /tmp/.mount_FreeCARIFEP4/usr/bin/../lib/libFreeCADApp.so+0xbe9
#7  0x7f5e395b738f in App::Document::Restore(Base::XMLReader&) from /tmp/.mount_FreeCARIFEP4/usr/bin/../lib/libFreeCADApp.so+0x23f
#8  0x7f5e395c3763 in App::Document::restore(char const*, bool, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) from /tmp/.mount_FreeCARIFEP4/usr/bin/../lib/libFreeCADApp.so+0x333
#9  0x7f5e3978dd18 in App::Application::openDocumentPrivate(char const*, char const*, char const*, bool, bool, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&&) from /tmp/.mount_FreeCARIFEP4/usr/bin/../lib/libFreeCADApp.so+0x3d8
#10  0x7f5e3978eab5 in App::Application::openDocuments(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >*, bool) from /tmp/.mount_FreeCARIFEP4/usr/bin/../lib/libFreeCADApp.so+0x815
#11  0x7f5e397908a8 in App::Application::openDocument(char const*, bool) from /tmp/.mount_FreeCARIFEP4/usr/bin/../lib/libFreeCADApp.so+0xc8
#12  0x7f5e39790a5f in App::Application::processFiles(std::__cxx11::list<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) from /tmp/.mount_FreeCARIFEP4/usr/bin/../lib/libFreeCADApp.so+0x13f
#13  0x7f5e39791033 in App::Application::processCmdLineFiles() from /tmp/.mount_FreeCARIFEP4/usr/bin/../lib/libFreeCADApp.so+0x43
#14  0x7f5e39791539 in App::Application::runApplication() from /tmp/.mount_FreeCARIFEP4/usr/bin/../lib/libFreeCADApp.so+0x29
#15  /tmp/.mount_FreeCARIFEP4/usr/bin/freecad(+0x687e) [0x55bd0bd8d87e]
#16  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f5e37fa9083]
#17  /tmp/.mount_FreeCARIFEP4/usr/bin/freecad(+0x6c29) [0x55bd0bd8dc29]

```

AIRCAP, 2024-09-22:
Addendum. in the previous comment I incorrectly reported that the file saved in 0.20 can also be opened in 0.21 and 1.0 rc1 in CLI mode - that is not the case, I doublechecked - the error

**BSpline curve: Knots interval values too close**

still occurs when opening in either 0.21 or 1.0 rc1 in CLI mode -- however the warning message **_Cannot restore geometry extension of type: SketcherGui::ViewProviderSketchGeometryExtension_** does NOT appear with that file. for convenience i am attaching this file:
[simple_20.zip](https://github.com/user-attachments/files/17091253/simple_20.zip)

it has been created by opening the simple.FCStd in freecad 0.20 GUI and saving it again in that version.

this file loads (and saves) correctly in CLI mode in 0.20  but doing the same in 0.21 or 1.0 rc1 corrupts the file

this means

1. the issue does not seem to be directly related to the introduction of the ViewProviderSketchGeometryExtension - since it also occurs if the file does not include any reference to it
2. the issue is a regression introduced in 0.21

AIRCAP, 2024-09-22:
more info. i checked 

[code]
OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33669 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21rc1a)
Hash: e5a00e153d36b0d53ebd568f4f47e95fbb5d0fcd
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

this version is also fine and opens without error and can save in CLI without corruption. that means the regression happened between 0.21rc1 and 0.21.2

doing more checks

AIRCAP, 2024-09-22:
note: 0.21rc1 also segfaults when opening simple.FCStd - same as 0.20

AIRCAP, 2024-09-22:
this version is also fine (0.21.0)
[code]
OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

AIRCAP, 2024-09-22:
0.21.1 is affected. 
[code]
OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

This version still segfaults when opening simple.FCStd in CLI mode onPart::GeometryPersistenceExtension::Restore - it does not know the GeometryExtension - but when opening simple_20.FCStd in CLI mode it says
**BSpline curve: Knots interval values too close**  -- and when saving the file in CLI the sketch gets corrupted.

This really narrows it down - the bug got first introduced with 0.21.1 - and seems to have nothing to do with the GeometryExtension change.

AIRCAP, 2024-09-22:
I have to say I'm slightly confused. I used the AppImages from https://github.com/FreeCAD/FreeCAD-Bundle/releases to do above tests. With 0.21.1 the problem happens, with 0.21.0 it doesn't -- there's however only 19 commits in between these tags in the freecad git repo and none of them looks like they have the potential to cause this -- are there possibly additional changes between these AppImages not reported by the freecad about dialog?

luzpaz, 2024-09-23:
@AIRCAP could it possibly be related to a 3rd party dependency ?

AjinkyaDahale, 2024-09-23:
> could it possibly be related to a 3rd party dependency ?

@luzpaz while this is being figured out, could you mark this issue as a blocker?

AIRCAP, 2024-09-23:
> @AIRCAP could it possibly be related to a 3rd party dependency ?
@luzpaz I don't have a dev environment for freecad yet where I could compile my own version under controlled conditions. As I understand it, the AppImage from above link bundles all 3rd party dependencies, and the versions it reports for python, OCC, coin, etc.. are identical. I tested all on the same system, so the system version of any additional deps, glibc, kernel or X11/OpenGL should not have an impact.
That's why I asked for "additional changes" between these releases - for example in the AppImage build system, switch of compiler or similar. While that is relatively farfetched, a change in gcc or compiler optimization could trigger a bug that was previously hidden/masked.
I can try to compile Freecad by hand today with debug  symbols turned on. then I can see if I can get the same behaviour from the AppImages also with the self compiled version, which should allow narrowing down which commit caused it.

AIRCAP, 2024-09-23:
Playing devils advocate. Imagine a hypothetical use-after-free bug somewhere in freecad (or otherwise stale pointer issue pointing where it shouldn't) with the potential to corrupt the data of the bspline depending on the exact operation order on document load. Completely unrelated side effects could effect if that happens or not - for example loading all the view providers in GUI mode might hide the bug. so could a change in compiler optimization - or even code changes completely unrelated to the sketcher component or bsplines.  This would also mean the actual bug might have been in freecad for a very long time, or it could even be hidden in a 3rd party library - just waiting for the right conditions to strike at that poor unsuspecting Bspline - and there's no obvious relationship at all between the commit that triggers the bugs occurrence and the bug itself (could be as harmless as a change in length of a translation string somewhere, and suddenly BANG the memory regions by 2 pointers suddenly overlap)

I really hope its not that, because that would be very annoying to hunt down.

AjinkyaDahale, 2024-09-23:
Cannot reproduce https://github.com/FreeCAD/FreeCAD/issues/16724#issuecomment-2366814318 with
```
OS: Ubuntu Core 22 (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38803 (Git) Snap 1146
Build type: Release
Branch: main
Hash: 22fbfeddc2e7ff3f44bc80147b875ccc43503af1
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

AjinkyaDahale, 2024-09-23:
> Cannot reproduce [#16724 (comment)](https://github.com/FreeCAD/FreeCAD/issues/16724#issuecomment-2366814318) with
> 
> ```
> OS: Ubuntu Core 22 (GNOME/gnome)
> Word size of FreeCAD: 64-bit
> Version: 1.1.0dev.38803 (Git) Snap 1146
> Build type: Release
> Branch: main
> Hash: 22fbfeddc2e7ff3f44bc80147b875ccc43503af1
> Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
> Locale: English/United States (en_US)
> Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
> ```

@AIRCAP can you attach `broken.FCStd` as you get after those steps?

AIRCAP, 2024-09-23:
> @AIRCAP can you attach `broken.FCStd` as you get after those steps?

[broken.zip](https://github.com/user-attachments/files/17093777/broken.zip)

AIRCAP, 2024-09-23:
I managed to build freecad from source now. I have the issue reproduced in
```
OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38797 (Git)
Build type: Debug
Branch: main
Hash: 86fa65ba489f0168145a1604423999b08090f857
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default

```

AIRCAP, 2024-09-23:
I can reproduce the issue in debug builds from
```
OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38785 (Git)
Build type: Debug
Branch: releases/FreeCAD-1-0
Hash: b2b0af95d85f601471e57283c19d08ec8e3ee5dc
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```
all the way down to
```
[code]
OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Debug
Branch: reproduce-0-21-1
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```
but 0.21.0 works fine, even in a self built debug build

AIRCAP, 2024-09-23:
Will do.
Just to rule out an issue with my computer I tried it on my laptop with
```
OS: Ubuntu 18.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

and another desktop with

'''
'''

I managed to reproduce the issue on both of these. In each cases the issue only occurs in CLI mode (starting freecad with the "-c" flag) not in GUI mode.

AIRCAP, 2024-09-23:
Did a systematic git bisect - this is the result:
```
git bisect good
4fc6b3d830c54b304abafe77a8eee03f3f701d75 is the first bad commit
commit 4fc6b3d830c54b304abafe77a8eee03f3f701d75
Author: Syres916 <46537884+Syres916@users.noreply.github.com>
Date:   Tue Aug 29 18:21:27 2023 +0100

    [0.21] [Gui] Py311 Initialization of locales... (#3)
    
    * [0.21] [MainGui] Py311add initialization of locales ...
    
    ... with native environment setting
    
    * [0.21][Gui] use '%Y-%m-%d %H:%M:%S' as format string for date

 src/Gui/View3DPy.cpp | 2 +-
 src/Main/MainGui.cpp | 1 +
 2 files changed, 2 insertions(+), 1 deletion(-)
```

luzpaz, 2024-09-23:
@Syres916 can you take a look if this is accurate ?

AIRCAP, 2024-09-23:
silly question could the system locale play a role? particularly if a constraint uses comma versus dot as a decimal ? I noticed this in the GUI that on systems with german locale some FreeCAD version insist on using the comma, while others use the dot. if this is incorrectly initialized in the CLI version because it only happens in GUI... just speculating...
but a locale related issue would explain why @AjinkyaDahale can't reproduce

AIRCAP, 2024-09-23:
its definitely locale. I get:

```
~/src/FreeCAD/test/bin/FreeCAD ../simple_20.FCStd -c
BSpline curve: Knots interval values too close
Importing project files......
Postprocessing......				(100 %)		
SketchObject::onChanged(): Unmanaged change of Constraint Property results in invalid constraint indices
[FreeCAD Console mode <Use Ctrl-D (i.e. EOF) to exit.>]	
```

vs

```
LC_ALL=C ~/src/FreeCAD/test/bin/FreeCAD ../simple_20.FCStd -c
Importing project files......
Postprocessing......				(100 %)		
[FreeCAD Console mode <Use Ctrl-D (i.e. EOF) to exit.>]		
>>> 

with the same version
```

AIRCAP, 2024-09-23:
this is my systems locale settings for reproduction:
```
env |grep -E "LC|LANG"
LC_ADDRESS=de_DE.UTF-8
LC_NAME=de_DE.UTF-8
LC_MONETARY=de_DE.UTF-8
LC_PAPER=de_DE.UTF-8
LANG=en_US.UTF-8
LC_IDENTIFICATION=de_DE.UTF-8
LC_TELEPHONE=de_DE.UTF-8
LC_MEASUREMENT=de_DE.UTF-8
LC_TIME=de_DE.UTF-8
LC_NUMERIC=de_DE.UTF-8

```

AIRCAP, 2024-09-23:
LC_ALL=C also fixes it for the 1.0 rc1 release and the original sketch in reproduction.zip posted in the beginning.

luzpaz, 2024-09-23:
Bravo, awesome sleuthing @AIRCAP!

bgbsww, 2024-09-23:
This part of the Document.xml inside of simple.FCStd:

                        <Geometry type="Part::GeomBSplineCurve"
 id="6" migrated="1">
                            <GeoExtensions count="1">
                                <GeoExtension type="Sketcher::SketchGeometryExtension" id="6" internalGeometryType="0" geometryModeFlags="00000000000000000000000000000000" geometryLayer="0"/>
                            </GeoExtensions>
                            <BSplineCurve PolesCount="5" KnotsCount="3" Degree="3" IsPeriodic="0">
                                <Pole X="-20.0000000000000000" Y="0.0000000000000000" Z="0.0000000000000000" Weight="1.0000000000000000"/>
                                <Pole X="0.0000000000000000" Y="10.0000000000000000" Z="0.0000000000000000" Weight="1.0000000000000000"/>
                                <Pole X="20.0000000000000000" Y="0.0000000000000000" Z="0.0000000000000000" Weight="1.0000000000000000"/>
                                <Pole X="0.0000000000000000" Y="-10.0000000000000000" Z="0.0000000000000000" Weight="1.0000000000000000"/>
                                <Pole X="-20.0000000000000000" Y="0.0000000000000000" Z="0.0000000000000000" Weight="1.0000000000000000"/>
                                <Knot Value="0.0000000000000000" Mult="4"/>
                                <Knot Value="0.5000000000000000" Mult="1"/>
                                <Knot Value="1.0000000000000000" Mult="4"/>
                            </BSplineCurve>
                            <Construction value="0"/>
                        </Geometry>

bgbsww, 2024-09-23:
I also note that the original bug indicates use of the '-c' flag, so presumably the freecad binary was used and not the freecad.cmd binary.

This likely matters because the line https://github.com/bgbsww/FreeCAD/blob/af9097ce875d1378fd6487bc5ffedb35bd148274/src/Main/MainGui.cpp#L100-L101 that was changed in the bisect located PR would only be relevant there - this predicts that the freecad.cmd would not show the issue.   Also Windows and MacOSX are off the hook - this is Linux / BSD only.

So, overall this is showing that the Document.XML file reader is sensitive to a locale setting that is not being overridden by https://github.com/bgbsww/FreeCAD/blob/af9097ce875d1378fd6487bc5ffedb35bd148274/src/Main/MainGui.cpp#L108

Which setting that is, and why that particular file snippet is locale sensitive in .de I'm not sure  - is this a "," vs "." in numbers thing?  If so, why doesn't the following putenv fix it?

bgbsww, 2024-09-23:
Likely because after the QLocate::system() has run the environment variables no longer matter.  So, a couple of untested proposed fixes:

    (void)QLocale::system();
    std::locale::global(std::locale::classic());

or

bgbsww, 2024-09-23:
<thinks> That same set of conditions ( freecad -c ) might also make it possible to write locale specific Document.xml inside of FCStd files, and that would be bad.  Worth a test, and if so, more changes.

AIRCAP, 2024-09-23:
My guess is the following (still needs actual confirmation)

> I also note that the original bug indicates use of the '-c' flag, so presumably the freecad binary was used and not the freecad.cmd binary.
> 

This is correct. The main reason to use the "-c" flag was because I predominantly tested with the precompiled AppImages (see link above) and those don't provide FreeCADCmd - so the only way to get to CLI mode is to use freecad -c

> This likely matters because the line https://github.com/bgbsww/FreeCAD/blob/af9097ce875d1378fd6487bc5ffedb35bd148274/src/Main/MainGui.cpp#L100-L101 that was changed in the bisect located PR would [...]

Confirmed. When I tested with the FreeCADCmd from branch main (commit ID a715a4d... ) the error does not get triggered, only with freecad -c

AIRCAP, 2024-09-23:
> (void)QLocale::system();
>     std::locale::global(std::locale::classic());

confirmed -- doing this change fixes the issue -- now checking the other variant

AIRCAP, 2024-09-23:
> in src/Base/Reader.cpp:
> 
> in std::istream& Base::XMLReader::beginCharStream
> 
> ```
>     CharStream = std::make_unique<boost::iostreams::filtering_istream>();
> #ifdef _MSC_VER
>     CharStream->imbue(std::locale::empty());
> #else
>     CharStream->imbue(std::locale::classic());
> #endif
> ```

making those changes does NOT fix the issue - constraint is still broken in -c CLI unless LC_ALL is set to C before invoking freecad

AIRCAP, 2024-09-23:
just confirming setting env variable LC_NUMERIC=C  before running freecad -c also fixes it. - no need to set LC_ALL

bgbsww, 2024-09-23:
Well done!  I assume that the first variant does not affect the UI if you actually run the GUI - you still get whatever localization effects you're looking for?

Are you interested in progressing your first PR?

AIRCAP, 2024-09-23:
when I open the sketch with freecad -c with LC_NUMERIC=de_DE.UTF-8 (my system default) and print App.ActiveDocument.Sketch.Contents -- a lot of numeric constants in the Contents have their decimals cut.  i.e. "-0.351"  turns to "-0"  and "3.14159" turns to "3" --  also "1e-6" turns to "0"

AIRCAP, 2024-09-23:
```
*** content1.xml	2024-09-23 23:48:23.089488804 +0200
--- content2.xml	2024-09-23 23:48:33.793853599 +0200
***************
*** 6,8 ****
          <Property name="ArcFitTolerance" type="App::PropertyPrecision">
!             <Float value="1e-06"/>
          </Property>
--- 6,8 ----
          <Property name="ArcFitTolerance" type="App::PropertyPrecision">
!             <Float value="0"/>
          </Property>
***************
*** 29,31 ****
              <ConstraintList count="33">
!                 <Constrain Name="" Type="19" Value="1" First="0" FirstPos="0" Second="-2000" SecondPos="0" Third="-2000" ThirdPos="0" LabelDistance="3.19757" LabelPosition="-3.14068" IsDriving="1" IsInVirtualSpace="0" IsActive="1" />
                  <Constrain Name="" Type="12" Value="0" First="0" FirstPos="0" Second="1" SecondPos="0" Third="-2000" ThirdPos="0" LabelDistance="10" LabelPosition="0" IsDriving="1" IsInVirtualSpace="0" IsActive="1" />
--- 29,31 ----
              <ConstraintList count="33">
!                 <Constrain Name="" Type="19" Value="1" First="0" FirstPos="0" Second="-2000" SecondPos="0" Third="-2000" ThirdPos="0" LabelDistance="3" LabelPosition="-3" IsDriving="1" IsInVirtualSpace="0" IsActive="1" />
                  <Constrain Name="" Type="12" Value="0" First="0" FirstPos="0" Second="1" SecondPos="0" Third="-2000" ThirdPos="0" LabelDistance="10" LabelPosition="0" IsDriving="1" IsInVirtualSpace="0" IsActive="1" />
***************
*** 61,63 ****
                  <Constrain Name="" Type="1" Value="0" First="-1" FirstPos="1" Second="9" SecondPos="1" Third="-2000" ThirdPos="0" LabelDistance="10" LabelPosition="0" IsDriving="1" IsInVirtualSpace="0" IsActive="1" />
!                 <Constrain Name="" Type="7" Value="10" First="11" FirstPos="1" Second="11" SecondPos="2" Third="-2000" ThirdPos="0" LabelDistance="9.18814" LabelPosition="-0.356825" IsDriving="1" IsInVirtualSpace="0" IsActive="1" />
              </ConstraintList>
--- 61,63 ----
                  <Constrain Name="" Type="1" Value="0" First="-1" FirstPos="1" Second="9" SecondPos="1" Third="-2000" ThirdPos="0" LabelDistance="10" LabelPosition="0" IsDriving="1" IsInVirtualSpace="0" IsActive="1" />
!                 <Constrain Name="" Type="7" Value="10" First="11" FirstPos="1" Second="11" SecondPos="2" Third="-2000" ThirdPos="0" LabelDistance="9" LabelPosition="-0" IsDriving="1" IsInVirtualSpace="0" IsActive="1" />
              </ConstraintList>

```

bgbsww, 2024-09-23:
Ah, okay, so we have some work left to do here, but we're in the right area.  So MainCmd does this:

AIRCAP, 2024-09-23:
> Well done! I assume that the first variant does not affect the UI if you actually run the GUI - you still get whatever localization effects you're looking for?
> 
> Are you interested in progressing your first PR?

will test that, can do a PR if the test goes well

bgbsww, 2024-09-23:
> will test that, can do a PR if the test goes well

Excellent, just let me know if any help is needed either with the code or the process.

AIRCAP, 2024-09-23:
@bgbsww GUI seems unaffected by the LC_NUMERIC override - when set to use the system number format for display in preferences, it does so everywhere including sketches (in my case using comma , as the decimal separator) - assuming that there shouldn't be any other side effects and this also fixes the issue, I started a PR. It's currently running its automated build checks.

AIRCAP, 2024-09-23:
https://github.com/FreeCAD/FreeCAD/pull/16763 <-- all checks successful - anything else I need to do?

bgbsww, 2024-09-23:
I'd just edit the initial comment ( description ) to include "Fix #16724"
which will make Github actually link the fix to the issue.

The maintainers have a Monday meeting so if this doesn't get picked up
before then, I'm sure they'll discuss it.

luzpaz, 2024-09-23:
Bravo!

AIRCAP, 2024-09-24:
Shall we close the isdue or dig a bit deeper why the incorrect numeric constants caused the geometry to fail?

My assumption was this:
1. Wrong locale causes numeric distance in constraint to be zero
2. solving the sketch with zero reference size causes the bspline to throw an exception in OCC ( the one printed)
3. the exception is not handled properly and escalates too far up, causing not just the bspline but the entire sketch geometry to abort/ fail initialisation
4. when thisbis then saved, the sketch is invalid, since it has constraints referencing nonexistent geometry.

The glaw in this thinking is thatvthe only numeric constraint, 10mm in the simple.FCStd had no decimal and was actually read in correct, so it has to be slightly more complicated.

AIRCAP, 2024-09-24:
I created a FCStd in which I deliberately replaced all numeric constants with integers (cutting off the decimals) - mimicking the effect of the locale bug. This allows debugging and reproduction of the fault chain even in GUI - respectively with the fixed version to see why the geometry vanishes.

[deliberatelyBroken.zip](https://github.com/user-attachments/files/17111540/deliberatelyBroken.zip)

on loading, it creates the following error messages:

```
BSpline curve: Knots interval values too close
No extension found of type 'Gui::ViewProviderFaceTexture'
SketchObject::onChanged(): Unmanaged change of Constraint Property results in invalid constraint indices
```

bgbsww, 2024-09-24:
Exactly.  So I don't think there's anything to chase here.  This should be closed, which would have happened automatically if that "Fix #16724" had made it into the PR description before chennes got to merging it.

There are a lot of places where invalid geo will cause non-crash warnings, and defending all of them would be difficult, non-performant, and possibly position us to mask underlying engine bug fixes.

If you're interested in new fun things to chase, there are a couple of them [here](https://github.com/FreeCAD/FreeCAD/milestones/1.0)

AIRCAP, 2024-09-24:
good points. 

just for completeness sake, here is a backtrace of the throw in latest "main" branch

```
(gdb) catch throw Standard_ConstructionError
Catchpoint 10 (throw)
(gdb) cont
Continuing.

Thread 1 "FreeCAD" hit Catchpoint 10 (exception thrown), 0x00007ffff13b8662 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) bt
#0  0x00007ffff13b8662 in __cxa_throw () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007fffc575d194 in CheckCurveData(TColgp_Array1OfPnt const&, TColStd_Array1OfReal const&, TColStd_Array1OfInteger const&, Standard_Integer, Standard_Boolean) (CPoles=..., CKnots=..., CMults=..., Degree=3, Periodic=<optimized out>) at ./src/Standard/Standard_DomainError.hxx:35
#2  0x00007fffc5797202 in Geom_BSplineCurve::Geom_BSplineCurve(NCollection_Array1<gp_Pnt> const&, NCollection_Array1<double> const&, NCollection_Array1<double> const&, NCollection_Array1<int> const&, int, bool, bool) (this=0x55555a6d87f0, Poles=..., Weights=..., Knots=..., Mults=..., Degree=<optimized out>, Periodic=false, CheckRational=false) at ./src/Geom/Geom_BSplineCurve.cxx:180
#3  0x00007fffc6c3cd4d in Part::GeomBSplineCurve::Restore(Base::XMLReader&) (this=0x5555565f8360, reader=...) at ~/src/FreeCAD/src/Mod/Part/App/Geometry.cpp:2030
#4  0x00007fffc6a00366 in Part::PropertyGeometryList::tryRestoreGeometry(Part::Geometry*, Base::XMLReader&) (this=0x555557f3af48, geom=0x5555565f8360, reader=...) at ~/src/FreeCAD/src/Mod/Part/App/PropertyGeometryList.cpp:219
#5  0x00007fffc6a0098b in Part::PropertyGeometryList::Restore(Base::XMLReader&) (this=0x555557f3af48, reader=...) at ~/src/FreeCAD/src/Mod/Part/App/PropertyGeometryList.cpp:263
#6  0x00007ffff4061a42 in App::PropertyContainer::Restore(Base::XMLReader&) (this=0x555557f3a2b0, reader=...) at ~/src/FreeCAD/src/App/PropertyContainer.cpp:379
#7  0x00007ffff3ea8b54 in App::ExtensionContainer::Restore(Base::XMLReader&) (this=0x555557f3a2b0, reader=...) at ~/src/FreeCAD/src/App/ExtensionContainer.cpp:306
#8  0x00007fffc69c64f3 in Part::Part2DObject::Restore(Base::XMLReader&) (this=0x555557f3a2b0, reader=...) at ~/src/FreeCAD/src/Mod/Part/App/Part2DObject.cpp:254
#9  0x00007fffc027fd77 in Sketcher::SketchObject::Restore(Base::XMLReader&) (this=0x555557f3a2b0, reader=...) at ~/src/FreeCAD/src/Mod/Sketcher/App/SketchObject.cpp:9974
#10 0x00007ffff3d9b217 in App::Document::readObjects(Base::XMLReader&) (this=0x555557e75a40, reader=...) at ~/src/FreeCAD/src/App/Document.cpp:1442
#11 0x00007ffff3d982f2 in App::Document::Restore(Base::XMLReader&) (this=0x555557e75a40, reader=...) at ~/src/FreeCAD/src/App/Document.cpp:1065
#12 0x00007ffff3d9dc8e in App::Document::restore(char const*, bool, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)
    (this=0x555557e75a40, filename=0x555556aa2a10 "~/src/FlyingWing/reproduce/repro2/test/bla/deliberatelyBroken.FCStd", delaySignal=true, objNames=std::vector of length 0, capacity 0) at ~/src/FreeCAD/src/App/Document.cpp:2102
#13 0x00007ffff4147143 in App::Application::openDocumentPrivate(char const*, char const*, char const*, bool, bool, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&&)
    (this=0x555555735eb0, FileName=0x5555581c15e0 "~/src/FlyingWing/reproduce/repro2/test/bla/deliberatelyBroken.FCStd", propFileName=0x5555581c15e0 "~/src/FlyingWing/reproduce/repro2/test/bla/deliberatelyBroken.FCStd", label=0x555558434a20 "deliberatelyBroken", isMainDoc=true, createView=true, objNames=...) at ~/src/FreeCAD/src/App/Application.cpp:1023
#14 0x00007ffff4145623 in App::Application::openDocuments(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >*, bool) (this=0x555555735eb0, filenames=std::vector of length 1, capacity 1 = {...}, paths=0x0, labels=0x0, errs=0x0, createView=true) at ~/src/FreeCAD/src/App/Application.cpp:807
#15 0x00007ffff4144639 in App::Application::openDocument(char const*, bool) (this=0x555555735eb0, FileName=0x55555846d650 "~/src/FlyingWing/reproduce/repro2/test/bla/deliberatelyBroken.FCStd", createView=true) at ~/src/FreeCAD/src/App/Application.cpp:698
#16 0x00007ffff418be07 in App::Application::sOpenDocument(_object*, _object*, _object*) (args=0x7fffeb5e9fa0, kwd=0x0) at ~/src/FreeCAD/src/App/ApplicationPy.cpp:263
#17 0x00007ffff2969748 in  () at /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0
#18 0x00007ffff2969b2b in _PyObject_MakeTpCall () at /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0
#19 0x00007ffff2735df3 in  () at /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0
[... cut ...]
#70 0x00007ffff58be6ac in (anonymous namespace)::tryRunEventLoop(Gui::GUISingleApplication&) (mainApp=...) at ~/src/FreeCAD/src/Gui/Application.cpp:2207
#71 0x00007ffff58be9db in (anonymous namespace)::runEventLoop(Gui::GUISingleApplication&) (mainApp=...) at ~/src/FreeCAD/src/Gui/Application.cpp:2227
#72 0x00007ffff58bed18 in Gui::Application::runApplication() () at ~/src/FreeCAD/src/Gui/Application.cpp:2304
#73 0x0000555555581ddc in main(int, char**) (argc=1, argv=0x7fffffffddd8) at ~/src/FreeCAD/src/Main/MainGui.cpp:295

```

closed 2024-09-24 by AIRCAP