FreeCAD/FreeCAD #16725: Draft: When moving Draft_Mirrored elements to layer, only the copied/mirrored elements take over layer properties

semhustej, 2024-09-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When moving Draft_Mirrored elements to layer, only the copied/mirrored elements take over layer properties.

Steps to reproduce:

1. Open test file [FC-mirrored-object-layer.zip](https://github.com/user-attachments/files/17088935/FC-mirrored-object-layer.zip)
2. Drag and drop the Rectangle (mirrored) element to 'Test layer' in the project tree
3. Only 1/2 of the object takes over the layer properties (line color, line type, fill color, etc.). The base object for the mirror remains with the original properties

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-09-23:
This is indeed the case, you need to move the base object to the layer too.

I'm not sure this is a bug? Isn't this the most "expected" behaviour?

semhustej, 2024-09-24:
> This is indeed the case, you need to move the base object to the layer too.
> 
> I'm not sure this is a bug? Isn't this the most "expected" behaviour?

I see - you have to first move base object to the layer and then assign the mirror operation to it. 

This is the behavior as I understand it:
1. Once the mirror operation is assigned to the base object, there is no way to move the base object to the layer. The user has to Draft_Downgrade the mirror and than he/she can move both objects to the layer.
2. When mirrored object is moved to the layer, the base object is move together with the mirror object. It is not on any other place in the model tree, but does not take over the properties of the layer. This seems unintuitive to me, but I am just getting used to how model tree works.
[image]

If this behavior is intended, the issue can be closed.

semhustej, 2024-09-26:
There is one more case, where mirrored object behavior does not make sense to me: when I am placing it on a TechDraw page as part of a group or a layer:

https://github.com/user-attachments/assets/21c544c4-4a33-48b4-9b39-c0c19e072fdf

I think I will just have to recalibrate how to handle mirrored objects, based on FreeCAD's behavior.

maxwxyz, 2025-03-09:
Adding from #18309 
Changing Draft Layer shape appearance, e.g. shape appearance color or transparency does not change the appearance of children / elements inside that layer.

Steps to reproduce:

1. Create a layer
2. Change shape appearance of the layer
3. Move objects in the layer

[image]

luzpaz, 2025-12-11:
Anyone with a proposed solution ?