FreeCAD/FreeCAD #16727: TechDraw: Access violation when activating the surface finish symbols tool with wrong tree selection

FEA-eng, 2024-09-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Select something else than a View in the model tree (e.g. Page or Template).
2. Activate the Surface finish symbols tool.

I know it's not how this tool is supposed to be used but I guess that messages like `Exception: Access violation` should be avoided if possible. Also, other tools show proper warnings about incorrect selection.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-09-22:
Confirmed:

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x7ce26fe45320]
#1  0x7ce222bf2ff2 in CmdTechDrawSurfaceFinishSymbols::activated(int) from /home/john/freecad-daily-build/Mod/TechDraw/TechDrawGui.so+0xb2
#2  0x7ce27390d0c6 in Gui::Command::_invoke(int, bool) from /home/john/freecad-daily-build/lib/libFreeCADGui.so+0x226
#3  0x7ce27390d4a6 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /home/john/freecad-daily-build/lib/libFreeCADGui.so+0x136
#4  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x312e16) [0x7ce270912e16]
#5  0x7ce271564f94 in QAction::triggered(bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x44
#6  0x7ce271567eab in QAction::activate(QAction::ActionEvent) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xbb
#7  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2fc512) [0x7ce2716fc512]
#8  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x304702) [0x7ce271704702]
#9  0x7ce2715b0df8 in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x278
#10  0x7ce27156bd45 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x85
#11  0x7ce2715746b0 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1290
#12  0x7ce2738aa1e8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/john/freecad-daily-build/lib/libFreeCADGui.so+0x98
#13  0x7ce2708d8118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x128
#14  0x7ce271572874 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1d4
#15  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cb5d2) [0x7ce2715cb5d2]
#16  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cdfbf) [0x7ce2715cdfbf]
#17  0x7ce27156bd45 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x85
#18  0x7ce2738aa1e8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/john/freecad-daily-build/lib/libFreeCADGui.so+0x98
#19  0x7ce2708d8118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x128
#20  0x7ce270d45a3b in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x80b
#21  0x7ce270d17bfc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#22  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x75d06) [0x7ce26a67bd06]
#23  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x5d5b5) [0x7ce26eb145b5]
#24  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xbc717) [0x7ce26eb73717]
#25  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7ce26eb13a53]
#26  0x7ce270935279 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x69
#27  0x7ce2708d6a7b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13b
#28  0x7ce2708df3e8 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x98
#29  0x7ce2737cd956 in Gui::Application::runApplication() from /home/john/freecad-daily-build/lib/libFreeCADGui.so+0x9f6
#30  /home/john/freecad-daily-build/bin/FreeCAD(+0x82e4) [0x6432b16e92e4]
#31  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x7ce26fe2a1ca]
#32  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7ce26fe2a28b]
#33  /home/john/freecad-daily-build/bin/FreeCAD(+0x86c5) [0x6432b16e96c5]
```

At a glace the `IsActive` looks correct, though it obviously isn't because I shouldn't have been able to click the icon:

```
bool CmdTechDrawSurfaceFinishSymbols::isActive()
{
    bool havePage = DrawGuiUtil::needPage(this);
    bool haveView = DrawGuiUtil::needView(this);
    return havePage && haveView;
}
```

Something tells me this is going to be a bigger change as there's more than just this command coded as such.

Syres916, 2024-09-22:
@WandererFan should the `IsActive` be something like:

```
bool CmdTechDrawSurfaceFinishSymbols::isActive()
{
    bool havePage = DrawGuiUtil::needPage(this);
    bool haveView = DrawGuiUtil::needView(this);
    if (havePage && haveView) {
        std::vector<Gui::SelectionObject> selection = this->getSelection().getSelectionEx();
        if (!selection.empty()) {
            std::vector<App::DocumentObject*> selProjGroupItem =
                this->getSelection().getObjectsOfType(
                    TechDraw::DrawProjGroupItem::getClassTypeId());
            std::vector<App::DocumentObject*> selView =
                this->getSelection().getObjectsOfType(TechDraw::DrawView::getClassTypeId());
            if (!selView.empty() || !selProjGroupItem.empty()) {
                return true;
            }
        }
    }
    return false;
}
```

And if so, there's going to be a few more commands that need changing?

Syres916, 2024-09-22:
@FEA-eng you've opened a can of worms now :) , I've just selected a `VarSet` in the tree and gone though the Techdraw menus looking for what `isActive` and I'd say 90% of the icons that are incorrectly active.

FEA-eng, 2024-09-22:
@Syres916 Thanks for investigating it so quickly. Other tools I checked just showed a warning message about incorrect selection but maybe there are also other cases like this one.

WandererFan, 2024-09-23:
Checking if the right things have been selected generally happens in the activated method.   Might be better placed in isActive so the buttons aren't shown unless your selection is appropriate, but, as you say, that means changing a lot of commands.

Syres916, 2024-09-24:
I'm completely OK with this being closed by #16766 but just for info I've started on improving the icons Enabled/Disabled status at https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:TechDraw_Restrict_Icons_Where_Reqd_dev

WandererFan, 2024-09-24:
That's a nice change.  Looking forward to it.

commit 0ff7c3a3 (2024-09-24): [TD]fix crash on bad selection (fix #16727)
commit d272722d (2024-09-24): [TD]fix crash on bad selection (fix #16727)
closed 2024-09-24 by chennes