FreeCAD/FreeCAD #16729: crash when creating b-spline in sketcher

sslupsky, 2024-09-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

B-Splines are crashing in 1.1.0dev build 38794:

Might be a regression of the problem reported here #14945 ?

To reproduce:
create a sketch, 
click first point in b-spline (anywhere, does not need to be origin)
click second point (anywhere),
right click

FreeCAD crashes at this point.

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * kicadStepUpMod 11.3.2
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sslupsky, 2024-09-22:
I went back to 1.0.0RC1 and the same problem exists.

sslupsky, 2024-09-22:
Trying to find a work around, I clicked several points and then clicked  the "close" button.  The sketch closed and FreeCAD did not crash but there was no b-spline rendered in the sketch.  The control points were in the sketch when I reopened it but there was no b-spline.

sslupsky, 2024-09-23:
Build 0.22.0dev.38495 crashes.
Build 0.21.2dev.33711 works.

sslupsky, 2024-09-23:
I disabled auto constraints and the crash still occurs so I believe the problem is not related to auto constraints per PR #14954 

The original fix in PR #14951 appears to only exempt the special case when there is only one control point.  If you only have one control point there is no crash.  When there are two or more control points, the crash occurs.

sslupsky, 2024-09-23:
It appears this issue appears to be related to the machine.  I attempted creating a BSpline on the same OS (macOS 14.6.1) but different architecture (arm64 M2 Max MBP) and creating the BSpline does not crash FreeCAD.  So, this issue appears to be specific to the x86 version.

maxwxyz, 2024-09-23:
@AjinkyaDahale @chennes FYI

luzpaz, 2024-09-23:
@johnocfii could you reproduce on macOS ? TIA!

sslupsky, 2024-09-23:
Here is a screen recording.
[image]

JohnOCFII, 2024-09-24:
No crash with my M1/arm Mac.  As @sslupsky mentioned, this might be an architecture-related crash and only impacting the Intel Macs.

```
OS: macOS 15.0
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.47
```

[image]

chennes, 2024-09-24:
No crash on my Intel Mac:
```
OS: macOS 13.7
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38758 (Git)
Build type: Release
Branch: releases/FreeCAD-1-0
Hash: e19f583e7724c535c6e035d6173f2016ce601f6d
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

sslupsky, 2024-09-24:
My Mac is a MP 6,1.  Could it be compiler related?  Compiler settings maybe using AVX2 instructions?

chennes, 2024-09-24:
Mine is MacBookPro16,4, so definitely different. Can you try self-compiling? If you've never done it before, I'd install [Miniforge](https://conda-forge.org/download/) then follow the instructions [here](https://freecad.github.io/DevelopersHandbook/gettingstarted/#conda) (but I know this is a big ask, getting things set up and compiling can be a headache, so if you don't have time I understand).

sslupsky, 2024-09-24:
I tried it on two other Intel machines (MBP 10,1 and MacMini 6,1) running 14.6.1 and OCLP and it works fine.  So it appears to be something to do with the MP 6,1.  Seems like it could be a QT related issue:
````
Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libQt5Gui.5.15.13.dylib       	       0x10b182e7c QCursor::QCursor(QCursor const&) + 30
1   SketcherGui.so                	       0x4297a78f8 SketcherGui::DrawSketchDefaultHandler<SketcherGui::DrawSketchHandlerBSpline, SketcherGui::StateMachines::TwoSeekEnd, 2, SketcherGui::ConstructionMethods::BSplineConstructionMethod>::finish() + 24
2   SketcherGui.so                	       0x4298debcb SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) + 1339
3   libFreeCADGui.dylib           	       0x10bd1a622 Gui::ViewProvider::eventCallback(void*, SoEventCallback*) + 674
```

I'll look into compiling it on the MP and see what that does.  But, this appears to be a very specific machine issue.

sslupsky, 2024-09-24:
@chennes  I am running into a problem with installation of one of the dependencies (netgen) when I follow the instructions you linked to.  Is there some way to work around this dependency?  I get an error when cmake attempts to link Mod/Fem/Fem.so: 

```
Undefined symbols for architecture x86_64:
  "fmt::v11::report_error(char const*)", referenced from:
      void fmt::v11::detail::vprintf<char, fmt::v11::basic_printf_context<char>>(fmt::v11::detail::buffer<char>&, fmt::v11::basic_string_view<char>, fmt::v11::basic_format_args<fmt::v11::basic_printf_context<char>>) in FemMeshShapeNetgenObject.cpp.o
```

chennes, 2024-09-24:
Honestly for now I'd just disable building FEM (e.g. -DBUILD_FEM=Off if you are using cMake on the command line).

sslupsky, 2024-09-24:
Yeah, I turned off BUILD_FEM_NETGEN in the CMakePresets.json file.  I was able to compile and there is **no crash.**

```
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38834 (Git)
Build type: Release
Branch: main
Hash: 490d6c5abc41bb61fa5bf47398df142a8ac2519e
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk , OCC 7.6.3
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * kicadStepUpMod 11.3.2
```

I noticed a new build (38827) was posted to the bundle repo a few minutes ago so I downloaded that build and it still crashes.

sslupsky, 2024-10-18:
Crash is still occurring in built 38946.

I found a work around of sorts by drawing a line or arc in the sketch and then converting the arc to a bspline.  I can then edit and manipulate the bspline without crashing FreeCAD.

depthoffocus, 2024-10-21:
No crash for me -- a 13" 2018 Intel MacBook Pro with 16gb RAM, Sonoma.

```
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git)
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ThreadProfile 1.92.0
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * Manipulator 1.5.7
  * fasteners 0.5.29
  * lattice2 1.0.0
  * CurvedShapes 1.0.12
  * Curves 0.6.50
```

https://github.com/user-attachments/assets/91df8ecb-b16e-419c-89ed-0aebcd757611

sslupsky, 2024-10-21:
I downloaded build 38991 (same build as @depthoffocus) and it is still crashing for me.

depthoffocus, 2024-10-21:
@sslupsky you're using OCLP, right? 

So this could be something to do with that. AVX2 maybe (not sure if I understand how that would be). Or legacy Metal?

Have you tried "use Software OpenGL" to see if that solves it?

The only other thing I wonder is if it crashes without KiCad StepUp installed. That's the only add-on you list, and none of us above appear to be using it.

sslupsky, 2024-10-21:
@depthoffocus Yes, I am running OCLP.  I tried it on a couple other machines running OCLP and didn't have a problem.  So uncertain if it is OCLP related or something specific to the MP 6,1.  I have D500 GPU's.

I removed KiCad StepUp but it did not make a difference.

I also tried the "use Software OpenGL" preference but that didn't make a difference either.

When I compiled FreeCAD from source on the MP 6,1, it did not crash.  It might have something to do with a compiler flag and AVX2.  But, I think I would have seen the same problem on the Mini 6,1 and MBP 10,1 since those CPU's do not support AVX2 either?

sslupsky, 2024-10-21:
Are the weekly builds archived somewhere?  If I could download older builds, I might be able to zero in on the build where the problem started.

closed 2025-08-09 by github-actions[bot]