FreeCAD/FreeCAD #16730: "New File" tiles/buttons background doesn't change for dark theme

j8spencer, 2024-09-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I started up the 1.0rc1 AppImage Mint 22 (~=Ubuntu Noble), and selected Dark theme at the first screen. The next screen allows me to chose a file to start on, and I see this:
[image]

The "New File" options do not have a suitable background for dark theme.

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
  * A2plus 0.4.60n
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-23:
Can you reproduce with the weekly ? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

j8spencer, 2024-09-23:
When I open up the version you linked to, the start screen does not show the problem:

[image]

I then switched theme to FreeCad classic, and back to dark. On returning to Dark theme, the "Recent Files" Chips/Buttons kept a light background, but the "New Files" changed back to dark correctly. I also got a "Preferences dialog is not responding" dialog, but I waited and it resolved itself.

I then switched to FreeCad Light and back to dark. I got a "FreeCad 1.1.0.dev is not responding." I then saw the preferences dialog close, but a second later, the app closed/crashed and I got

`/tmp/.mount_FreeCAmceIFj/AppRun: line 41: 56484 Killed                  "${MAIN}" "$@"`

in the console window. On restarting, Dark mode was applied correctly.

Switching to FreeCad light worked fine, but slow. Switching back to Dark gave me three "FreeCad 1.1.0 dev is not responding," and I clicked wait in all of them. It seemed to be ok, and both the New File and Recent Files chip/buttons were correctly in dark mode.

I'd say the problem seems better in this version, but there are still some issues, and it may be non-deterministic getting the background correct.

My version is:

OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git) AppImage
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Help 1.0.3
  * A2plus 0.4.60n

Syres916, 2024-09-23:
@j8spencer you're "is not responding" has been reported a while ago but it's in the packaging repo https://github.com/FreeCAD/FreeCAD-Bundle/issues/302

maxwxyz, 2024-11-01:
Fixed in latest weekly

luzpaz, 2024-11-01:
@maxwxyz do you mind adding the revision it was fixed in (for posterity) ?

maxwxyz, 2024-11-01:
Fixed by https://github.com/FreeCAD/FreeCAD/pull/17376