FreeCAD/FreeCAD #16732: Crash - Hover over DraftWB Polar Pattern

tilation88, 2024-09-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[00.zip](https://github.com/user-attachments/files/17090879/00.zip)
Open File Attached, hover over the Model, FC crashes. 

### Full version info

```shell
OS: LMDE 6 (faye) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * fcgear
  * CurvedShapes 1.0.9
  * OpenDark 2024.3.13
  * CADExchanger
  * freecad.gears 1.2.0
  * ProDarkThemePreferencePack 1.0.0
  * OpticsWorkbench 1.0.17
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-23:
Do you mind testing the weekly and reporting back? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

yorikvanhavre, 2024-09-23:
> Open File Attached, hover over the Model, FC crashes.

I cannot reproduce. No crash observed (main branch, debian testing)

Roy-043, 2024-09-23:
Can't reproduce either.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

luzpaz, 2024-09-23:
@tilation88 can you backup config files and try to reproduce again ?

tilation88, 2024-09-23:
[2024-09-23 16-44-52.zip](https://github.com/user-attachments/files/17099361/2024-09-23.16-44-52.zip)

ich deleted this folder:
/home/tino/.config/FreeCAD/

maybe its the GPU: gtx760 with the old v470 driver?  Update not possible, no support

luzpaz, 2024-09-23:
Can you start freecad from the console/terminal  ? (in order to see what the failure/error is when the crash occurs)

tilation88, 2024-09-23:
tino@tino-pc:~/Pakete_lokal/freecad$ ./freecad.AppImage 
FreeCAD 1.0.0, Libs: 1.0.0RC1R38642 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

During initialization the error "[Errno 13] Keine Berechtigung: ''" occurred in /home/tino/.local/share/FreeCAD/Mod/CADExchanger/Init.py
Please look into the log file for further information
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3c050) [0x7f473b45b050]
#1  0x7f473e9bfed3 in Gui::LinkView::linkGetElementPicked(SoPickedPoint const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) const from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libFreeCADGui.so+0x543
#2  0x7f473e9c0423 in Gui::ViewProviderLink::getElementPicked(SoPickedPoint const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) const from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libFreeCADGui.so+0x113
#3  0x7f473e83a9d2 in Gui::SoFCUnifiedSelection::getPickedList(SoHandleEventAction*, bool) const from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libFreeCADGui.so+0x572
#4  0x7f473e83f797 in Gui::SoFCUnifiedSelection::handleEvent(SoHandleEventAction*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libFreeCADGui.so+0x1e7
#5  0x7f473acd54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/./libCoin.so.80+0x68
#6  0x7f473ae8cb9a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/./libCoin.so.80+0xea
#7  0x7f473aedc348 in SoGroup::doAction(SoAction*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/./libCoin.so.80+0x58
#8  0x7f473aef713a in SoSeparator::doAction(SoAction*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/./libCoin.so.80+0x2a
#9  0x7f473acd54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/./libCoin.so.80+0x68
#10  0x7f473acdf26c in SoHandleEventAction::beginTraversal(SoNode*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/./libCoin.so.80+0xac
#11  0x7f473acd61ba in SoAction::apply(SoNode*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/./libCoin.so.80+0x2ea
#12  0x7f473aea8ad2 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/./libCoin.so.80+0x72
#13  0x7f473e8edfee in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libFreeCADGui.so+0x1e
#14  0x7f473e8f9588 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libFreeCADGui.so+0x58
#15  0x7f473e906230 in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libFreeCADGui.so+0x40
#16  0x7f473e908f2b in Gui::BlenderNavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libFreeCADGui.so+0x2db
#17  0x7f473e9064c5 in Gui::NavigationStyle::processEvent(SoEvent const*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libFreeCADGui.so+0x65
#18  0x7f473e92e6bb in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libFreeCADGui.so+0x6b
#19  0x7f473e8e491d in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libFreeCADGui.so+0x1fd
#20  0x7f473b8b68bb in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libQt5Core.so.5+0x7b
#21  0x7f473c77039e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libQt5Widgets.so.5+0x6e
#22  0x7f473c7778f9 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libQt5Widgets.so.5+0xa39
#23  0x7f473e53e499 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libFreeCADGui.so+0x99
#24  0x7f473b8b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libQt5Core.so.5+0x108
#25  0x7f473c7763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#26  /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x7f473c7c6e6f]
#27  /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x7f473c7ca0c9]
#28  0x7f473c7703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#29  0x7f473e53e499 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libFreeCADGui.so+0x99
#30  0x7f473b8b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libQt5Core.so.5+0x108
[... cut ...]
#39  0x7f473b8bd212 in QCoreApplication::exec() from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libQt5Core.so.5+0x82
#40  0x7f473e49f943 in Gui::Application::runApplication() from /tmp/.mount_freecaxO7J3s/usr/bin/../lib/libFreeCADGui.so+0x983
#41  /tmp/.mount_freecaxO7J3s/usr/bin/freecad(+0x7640) [0x5627cf392640]
#42  /lib/x86_64-linux-gnu/libc.so.6(+0x2724a) [0x7f473b44624a]
#43  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7f473b446305]
#44  /tmp/.mount_freecaxO7J3s/usr/bin/freecad(+0x79a9) [0x5627cf3929a9]

Syres916, 2024-09-24:
@tilation88 please upgrade to rc2 https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2 this segfault has been fixed after rc1 was released.

maxwxyz, 2024-10-07:
@tilation88 please test on the RC2 or dev version and report back. If there is no feedback I'll close this when not reproducible.

tilation88, 2024-10-11:
Sorry for my late answer, i got corona recently. Now i am back and tested RC2, everything works fine now. Thanks

closed 2024-10-11 by tilation88