FreeCAD/FreeCAD #16733: Chamfer size2 field has strange behavior

j8spencer, 2024-09-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm trying to chamfer the corner shown here to a mitre on this .75" part. When I click in the Size2 field, the text box for it goes blank with a cursor bar about 4 spaces in. When I type, no numbers show up in the text box. But if I type "0.75" (but don't hit enter), the part in the view rotates to vertical.

View on text entry:
[image]

View after typing ".75":
[image]

I can't seem to get the chamfer to range like I'd expect. But I'm a noob, so don't stress too much about that.

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
  * A2plus 0.4.60n
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-23:
Chamfers and fillets cannot fully consume a face. The OCC geometric kernel does not allow it. So you must do what you have discovered, leave a tiny face, or alternative extrude a chamfered sketch.

"Size 2" can only be entered after changing the "Chamfer Type" to "Two distances". If set to "One distance" the "Size 2" field is read-only and user input is interpreted as a short cut. See [Std_ViewFront](https://wiki.freecad.org/Std_ViewFront) etc.

Closing this.

closed 2024-09-23 by Roy-043

j8spencer, 2024-09-23:
I fully grant using the size2 field with the chamfer set to "equal distance" is not meaningful to the tool. But are you arguing that, in that context, it's correct behavior for the size2 field to change the 3D view orientation? I think that seems like a clear bug--for text box entry to be tied to some completely different effect. I get that, in that mode, the text field is probably not accepting input and so instead, it goes directly to the view window, where the input is working as intended. But I would suggest it's a usability bug not to indicate that somehow, and maybe consider having some UI feedback that clicking in that field is not allowed under "equal distance."

luzpaz, 2024-10-17:
@Roy-043 :point_up_2:

Roy-043, 2024-10-18:
@luzpaz If you want you can create a new issue to improve the UI in such cases (this is not just a PD issue).

luzpaz, 2024-10-18:
@j8spencer feel free to open a new issue per roy's suggestion.