FreeCAD/FreeCAD #16736: DRAFT: Crash on import DXF

gianlucafiore, 2024-09-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Good afternoon colleagues, I would like to know if there is a possibility of making Freecad work with complex line models in DRAFT, as Autocad, Librecad or Qcad can do, the minimum comparison I am going to make is, how when importing a DXF in Freecad (as a block first and then ungrouped with the inverted arrow) and in Autocad opened normally. Freecad collapses the RAM (reaching 6 GB), while Autocad remains stable at approximately 400 MB. I really do not understand much how the memory optimization is built in Freecad, but here there is a huge open door to constant crash.

#### In this first image, the memory consumed by freecad began to rise more and more when pressing the button
[image]

#### almost reaching 7gb of ram, i forced close it
[image]

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git)
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Qt default
Installed mods: 
  * Dracula 0.0.5
  * OpenDark 2024.3.13
  * OpenTheme 2024.9.1
  * SearchBar 1.0.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-23:
cc @yorikvanhavre

furgo16, 2024-09-23:
Possibly related PRs: https://github.com/FreeCAD/FreeCAD/pull/16611, https://github.com/FreeCAD/FreeCAD/pull/16596

yorikvanhavre, 2024-09-23:
The issue here is a classical one. FreeCAD. like all parametric modelers, is a very different kind of application than traditional 2D CAD like AutoCAD or QCAD. 

Each object, in FreeCAD (the same is valid for any other kind of parametric modeler like Revit, SoldiWorks, etc.. even Blender suffers from this) is a complex feature that has several connection slots and processes a lot of information, even the simple ones. In traditional CAD apps, on the contrary, individual objects are very simple structure that live in a database and have no parametric capabilities.

The result is this basic rule, parametric modelers can handle way less objects than traditional CAD apps.

When passing from one to the other, you need to change a bit your way of working, to limit the number of individual objects. Try to split complex DXF files into smaller ones, use [Arch References](https://wiki.freecad.org/Arch_Reference), use a lot of compounds, etc. The viewport of FreeCAD is able to handle millions of faces/lines easily, the key is to have few objects.

It is totally possible to work with complex 2D drawings, but you'll need to adapt a bit and find strategies.

Roy-043, 2024-09-23:
Looking at the images I get the idea that the texts have been exploded in the DXF. In other words, each letter consists of multiple lines. If that is indeed the case then that would be a contributing factor.

Roy-043, 2024-09-26:
@gianlucafiore Can you attach the DXF file?

gianlucafiore, 2024-09-28:
yes indeed the file is not the best case scenario for this, as it was a pdf converted to DXF. and therefore the texts were converted into segmented curves in thousands of small lines, for example. It is not relevant to continue addressing this problem.

anyway, as @yorikvanhavre comments, we will not be able to import a super complex DWG or DXF to be edited in freecad by the Draft tool.
Therefore, it represents for FreeCAD to carry out this process.

Perhaps we should address a little more on the documentation side how to avoid problems like this, from long processes that collapse to full Freecad. I put together another issue questioning the UX #16890 .

closed 2024-09-28 by gianlucafiore

yorikvanhavre, 2024-09-30:
> anyway, as @yorikvanhavre comments, we will not be able to import a super complex DWG or DXF to be edited in freecad by the Draft tool.

That's not totally exact. With NativeIFC we have seen that we're now able to import very big files with ease. They open fast and the viewport stays fluid.

However, that works while the whole model is packed in only a few objects. As the user expands the objects into their subcomponents (and therefore creates more objects in the FreeCAD document), one sees a drastic drop in performance. 

We could have the same system with DXF files: Import everything into one object, then allow the user to "expand" that further.

The current DXF importer is not really tailored for that level of user interaction, though, and I'm not sure how interesting it would be to pursue this path, as usually it's simpler to do some cleaning in the DXF file beforehand, and the "group objects by layer" function usually does the job and, typically, you would not "explode" everything to individual lines (and even so it's still possible with Draft Downgrade)