FreeCAD/FreeCAD #16737: Dimensioning doesn't latch to features properly

Arnold2020, 2024-09-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using measuring ruler, especially in assemblies, I cannot pick a line to measure from, rather just have to point and hope, and, even when selecting flat surfaces, not sure its really measuring properly.

### Full version info

```shell
The above explains fully, didn't have issue on older versions.
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-23:
Sounds like a duplicate of #13708

Roy-043, 2024-09-23:
Please post your full 'about' information. You also need to specify which assembly workbench you are using.

I can't reproduce with the built-in Assembly WB and the AssemblyExample.FCStd file.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

Arnold2020, 2024-09-23:
Hello Roy, I have added more detail in Full Version info, above. Here I am repeating it: **Updated to Roy's request**: I am using Version 0.21.2 of FreeCAD. The Assembly function I use is A2plus (the only one I have used). Previously (V19 of FreeCAD I believe), I would click on the Measure_Distance button, then I would select an edge of one part and an edge of another part and I would get an accurate distance measurement. But now, after clicking the Measure_Distance button, it doesn't allow me to select an edge and instead I point the cursor best I can to an edge and try to hit it. I still get a measurement, but often miss the edge which gives me a rough distance, but not so good for fine gaps, as example.

luzpaz, 2024-09-23:
@Arnold2020 Full [About](https://wiki.freecad.org/About) info please. TIA

luzpaz, 2024-09-23:
@Arnold2020 better to use [1.0 Release Candidate](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc1) as we are very close to releasing v1.0 soon. We aren't backporting any  fixes to 0.21.x

maxwxyz, 2024-09-23:
As I read it, it does not snap to the feature (edge/surface). This is solved with the current RC / dev version as mentioned above.
I'll close this, please comment if there still is an issue in the newest version.

closed 2024-09-23 by maxwxyz