FreeCAD/FreeCAD #16738: PartDesign: Chamfer tool not working properly (FreeCAD 1.0.0RC1.38642, Windows 11)

evanmac03, 2024-09-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello, I am having an issue with the chamfer in FreeCAD 1.0.0RC1.38642, I am tying to add an 1 mm chamfer to three pockets.

I am following this youtube tutorial:

https://www.youtube.com/watch?v=nG4Wh6Kk1bA&list=PL_28gc6LBA1ve8Aamf1izQjNlqgtBWoLS&index=7

Here are the steps I am currently taking: 

1. Selecting edge38 (the circle on left if you looking at the bottom of the model)
2. Press the Chamfer button
3. Leave size at 1 mm
4. Click "Select"
5. Click on middle and right circle edges.
6. Click preview 

After clicking on preview, the only hole that gets chamfered is edge38 (left circle). 

[image]

When clicking on the okay button I get a pop up window called 'Input error' with the message 'BRep_API: command not done"

I am not exactly sure if this a bug or not because if I change the size of the chamfer to 0.9 mm, the chamfer operation works. 

[image]

This makes me think that FreeCAD is not performing the chamfer at 1 mm because there is not enough room between the chamfer on the rectangle pocket and the chamfer on the circle pockets. But then again edge38 was able to do a 1 mm chamfer while edge36 and edge 37 couldn't.

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Canada (en_CA)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

evanmac03, 2024-09-23:
Here is the file I am currently using. 
[beginner6-2.FCStd.zip](https://github.com/user-attachments/files/17092258/beginner6-2.FCStd.zip)

Roy-043, 2024-09-23:
This is a limitation of the OCC geometric kernel. A chamfer and a fillet cannot fully consume a face. The wall thickness of the object is 1mm, therefore the chamfer size must be smaller. The fact that it 'accidentally' works for one hole is probably due to small rounding errors.

I am closing this.

closed 2024-09-23 by Roy-043