FreeCAD/FreeCAD #16739: Freecad RC1.1.0dev - 38794 -  Part Design - Expression editor accepts multiply and divide, but not pluss and minus

terjenauf, 2024-09-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Parts editor I have created a few variables under VarSet. When using them on Constraints on a sketch, i can edit the value with "*" or "/" (multiply or divide) but not "+" or "-" (plus or minus). I did not find a bug for this here.
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Norwegian Bokmal/Norway (nb_NO)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.67
  * btl 0.9.9
  * ExplodedAssembly
  * fasteners 0.5.29
  * frame 0.1.1
  * freecad.gears 1.2.0
  * Gridfinity 0.6.0
  * parts_library
  * Render 2024.8.4
  * Reporting
  * sheetmetal 0.4.25
  * symbols_library
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-09-23:
`Base_tykkelse_vegg` is 18mm. The unit here is `mm`. To add 1mm you need to include the unit. Same for subtraction.

The reason multiplication "works" in your example is that you need to have a unitless value for one of the values, otherwise you would get `mm^2` which is an area.

terjenauf, 2024-09-23:
Thanks for your quick reply. I thougth I tried that as well. Mathematically you are off cource correct.
Then this is no bug, and I will close it. Sorry for the inconvenience.

closed 2024-09-23 by terjenauf