FreeCAD/FreeCAD #16743: FC 1.1.0 dev38794 - Display picture

Bernard-19, 2024-09-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When importing a .png or jpg picture, File/Import, the representation does not appear on the screen.
Is this a bug or is there a new way to import an picture.

### Full version info

```shell
https://forum.freecad.org/viewtopic.php?t=90840
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-09-23:
> OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)

Confirmed with
```
OS: Debian GNU/Linux trixie/sid (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 +2 (Git)
Build type: Debug
Branch: dev
Hash: c1cae4a9c5469061dea4eb137bd0e9ef78d55758
Python 3.12.6, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.2.dev
Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
```

luzpaz, 2024-09-23:
is this a blocker ?

FlachyJoe, 2024-09-23:
@luzpaz no possibility to import a picture, yes it is.
It's reported to work in **1.1.0dev.38768** so one of the recent changes with colors is certainly the culprit.

Syres916, 2024-09-23:
@FlachyJoe can you please confirm you're non visible object has the same Shape Appearance info as my screenshot including the Lighting - two side.

FlachyJoe, 2024-09-23:
@Syres916  Updated to last 

> OS: Debian GNU/Linux trixie/sid (XFCE/xfce)
> Word size of FreeCAD: 64-bit
> Version: **1.1.0dev.38807** (Git)
> Build type: Debug
> Branch: main
> Hash: a55e7047588f512f713cbf60b5e4b854b1d7f23f
> Python 3.12.6, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.2.dev
> Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
> Stylesheet/Theme/QtStyle: unset/Classic/Qt default

the problem persists
[image]

FlachyJoe, 2024-09-23:
The problem is with accents in image path!
```
14:02:16  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'OSError'>: {'sclassname': 'N4Base15FileSystemErrorE', 'sErrMsg': 'File /home/user/T/xe9l/xe9chargements/20230415_064924.jpg does not exist.', 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': 'File /home/user/T/xe9l/xe9chargements/20230415_064924.jpg does not exist.', 'btranslatable': False, 'breported': False}
14:02:16  Nouveau: App.getDocument('Unnamed').ActiveObject.ImageFile = '/home/user/T\\xe9l\\xe9chargements/20230415_064924.jpg'
14:02:16  pyException: Traceback (most recent call last):
  File "/XHome/user/FreeCAD/Ext/freecad/module_io.py", line 12, in OpenInsertObject
    except PyExc_FC_AbortIOException:
           ^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'NameError'>: name 'PyExc_FC_AbortIOException' is not defined
```

Syres916, 2024-09-26:
To be clear, this issue doesn't just apply to images, at the very least it applies to Stl, Step etc imports.

Syres916, 2024-09-26:
By creating a folder École on my English based system I can create the error so I've narrowed the regression down to:

Good:
```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git) AppImage
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.2.0
  * Curves 0.6.45
  * A2plus 0.4.67
```

Bad:
```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git) AppImage
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.2.0
  * Curves 0.6.45
  * A2plus 0.4.67

```

Bisect in progress.

Syres916, 2024-09-26:
@supermixed [commit](https://github.com/FreeCAD/FreeCAD/commit/f763425b4f64bd36ce5e184becca62be29fcccab) appears to be the root cause of importing various formats with non-ascii characters such as É causing errors like detailed above, could you take a look please.

bgbsww, 2024-09-26:
The issue here is likely the simple Py::String(unicodepath) that should be a PyUnicode_FromString() or the like.

The python code that was replaced with c++ was explicitly using a u"" string on the value from c++ which did that.

However there's also the question of whether the python code should have been replaced with c++ code at all, rather than being updated to call the new method.  That's because it removes things that used to be in the console and macros.  See my rather late comment on the commit.

luzpaz, 2024-09-27:
Being addressed in #16847 ?

bgbsww, 2024-09-27:
> Being addressed in #16847 ?

I don't think so.  While that fixes the exception name, it doesn't address the cause of the exception.

supermixed, 2024-09-27:
Whoops! I'll have a poke at it now.

Syres916, 2024-09-27:
> Whoops! I'll have a poke at it now.

Thanks for getting back in touch, there has been some work done https://github.com/FreeCAD/FreeCAD/pull/16859 to save you from duplication

mwganson, 2024-09-27:
Let me know if I need to close #16859 if the issue is going to fixed in another PR.

maxwxyz, 2024-10-06:
Can someone test if #16859 fixes the issue?

luzpaz, 2024-10-06:
@mwganson could you test this ticket with your fix ?

mwganson, 2024-10-06:
There could be some hardware or driver issue on that computer that my computer does not have, so a test on my end would still be inconclusive.  I don' t even have the file that was causing the issue to work with.

A simple test to do without building with the PR applied is to rename the image file so it does not contain any non-latin characters.  (_There must also be no non-latin characters in the folder names._)  If without the non-latin characters the image displays correctly, then this PR (https://github.com/FreeCAD/FreeCAD/pull/16859) should resolve the issue once merged.

FEA-eng, 2024-10-08:
Seems to be fixed by #16859.

maxwxyz, 2024-10-08:
Closing this then

closed 2024-10-08 by maxwxyz