FreeCAD/FreeCAD #16744: SEGV Changing pad length with measurement tool open

j8spencer, 2024-09-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I had the measurement tool open, set to length, and measured an edge along a pad. I realized I need that pad longer, so I went to increase the length--selected the pad in the model view, typed in a new length in the parameter text box, hit enter. The application crashed with the following on the console (some of which may have been older):

`<Sketch> SketchObject.cpp(9053): Failed to project external geometry in BaseSideWall#Sketch001: Binder.;#52:3;:G;CHF;:H9af:7,F;:H9b0,F
Selected external reference plane must be normal to sketch plane
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Not able to add external shape element Face1
App.getDocument('BaseSideWall').getObject('Sketch001').addExternal("Binder","Face1")
Failed to add external geometry
<Sketch> SketchObject.cpp(9053): Failed to project external geometry in BaseSideWall#Sketch001: Binder.;#52:3;:G;CHF;:H9af:7,F;:H9b0,F
Selected external reference plane must be normal to sketch plane
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Not able to add external shape element Face1
App.getDocument('BaseSideWall').getObject('Sketch001').addExternal("Binder","Face1")
Failed to add external geometry
Link to Edge5 already exists in this sketch.
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Not able to add external shape element Edge5
App.getDocument('BaseSideWall').getObject('Sketch001').addExternal("Binder","Edge5")
Failed to add external geometry
<Sketch> SketchObject.cpp(9053): Failed to project external geometry in BaseSideWall#Sketch001: Binder.;#52:3;:G;CHF;:H9af:7,F;:H9b0,F
Selected external reference plane must be normal to sketch plane
Link to Edge5 already exists in this sketch.
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Not able to add external shape element Edge5
App.getDocument('BaseSideWall').getObject('Sketch001').addExternal("Binder","Edge5")
Failed to add external geometry
Link to Edge5 already exists in this sketch.
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Not able to add external shape element Edge5
App.getDocument('BaseSideWall').getObject('Sketch001').addExternal("Binder","Edge5")
Failed to add external geometry
Link to Edge5 already exists in this sketch.
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Not able to add external shape element Edge5
App.getDocument('BaseSideWall').getObject('Sketch001').addExternal("Binder","Edge5")
Failed to add external geometry
Link to Edge5 already exists in this sketch.
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Not able to add external shape element Edge5
App.getDocument('BaseSideWall').getObject('Sketch001').addExternal("Binder","Edge5")
Failed to add external geometry
1.25e-07 Selection.cpp(157): Unhandled std::exception caught in selection observer: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAZXcM2y/usr/Mod/Measure/MeasureCOM.py", line 114, in parseSelection
    obj.Element = (o, item["subName"])
    ^^^^^^^^^^^
<class 'ValueError'>: PropertyLinkSub does not support external object
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAZXcM2y/usr/Mod/Measure/MeasureCOM.py", line 112, in parseSelection
    item = selection[0]
           ~~~~~~~~~^^^
<class 'IndexError'>: tuple index out of range
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAZXcM2y/usr/Mod/Measure/MeasureCOM.py", line 114, in parseSelection
    obj.Element = (o, item["subName"])
    ^^^^^^^^^^^
<class 'ValueError'>: PropertyLinkSub does not support external object
[... 127 lines cut ...]
Recovery worked great on re-start. Whew! :D

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
  * A2plus 0.4.60n
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hlorus, 2024-09-25:
How do you even get to the dialog of the pad without closing the measure dialog, don't you get that popup which closes the measure dialog?

[image]

j8spencer, 2024-10-02:
> How do you even get to the dialog of the pad without closing the measure dialog, don't you get that popup which closes the measure dialog?

I do _not_ see that dialog. I can click in the model tree and edit in the properties window even while the measure dialog is open. I've noticed the same thing while a task dialog is open as well, and got myself into trouble a couple of times.

j8spencer, 2024-10-04:
I could not create a crash in RC2 (or again in RC1). I *think* the specifics were that I had an edge created by a chamfer on a pad and I increased the pad length which invalidated the chamfer, but not sure. Anyway, no crash now.

RC2 seems to make sure that if you have a measurement shown and you click inside the properties box, the measurement disappears. The measurement task stays open, but I could not keep the measurement up and edit a property at the same time. No dialog box popped up, though. The measurement just disappeared as soon as I clicked in a text box.

I _was_ able to get some measurements stuck in the 3D view that I have no idea how to turn off except bail and come back.

[image]

hlorus, 2024-10-05:
The remaining issue should be resolved with https://github.com/FreeCAD/FreeCAD/pull/16836 which apparently isn't included in RC2

luzpaz, 2024-10-07:
@j8spencer can you test the weekly and try to reproduce the remaining issue? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

Closing for now. Please re-open if issues persist.

closed 2024-10-07 by luzpaz