FreeCAD/FreeCAD #16745: flatpak FreeCAD 0.21 on AlmaLinux 8.10, Crashed FreeCAD

pcyookies, 2024-09-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

flatpak FreeCAD 0.21.2 on AlmaLinux 8.10.
Allways crashed.
The same problem occurs even if I try it on multiple PCs.
[z2mini v2.zip](https://github.com/user-attachments/files/17096086/z2mini.v2.zip)

### Full version info

```shell
1. open target file.
2. edit Pad001.Sketch001
3. on left side, create rectangle. already exists same two rectangles.
4. Skecher Constrain Distance X -> 5mm
5. task -> close -> Crashed FreeCAD.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-23:
Can you reproduce on 1.0RC1 version ?
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc1

pcyookies, 2024-09-23:
Thank you for your reply, Mr.luzpaz.
I have tried it.
There was nothing wrong with that.
I'm a beginner so I don't really understand it, but
I'd like to try using that version from now on.

luzpaz, 2024-09-23:
@pcyookies we are on the verge of release version 1.0 right now. So it's safe to use the 1.0 Release Candidate until that happens. We're not fixing issues in v0.21.x going forward. Thanks for report. Closing.

closed 2024-09-23 by luzpaz