FreeCAD/FreeCAD #16746: Mirrored object icons do not help to identify the object.

kaiwas, 2024-09-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Regardless of the object we mirrored, all icons in the tree view look the same.

This problem is also inherent in arrays and many other tools.

Perhaps it is worth considering displaying multiple icons. This would allow us to immediately see which object the tool is applied to and at the same time display the tool itself.
In essence, this could be a "micro tree view". But seeing the collapsed tree, we could know what's inside.

[image]

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-09-23:
A question about displaying sketches may arise. But given that a sketch is often the lowest-level object, displaying it in the "microtree" is not necessary.

kaiwas, 2024-09-23:
> 1. If a user is adequately naming features in complex projects is this necessary?

If someone doesn't need this, it would be great if it was an option to turn on in the global settings.

> 2\. How does quickly seeing the underlying feature type help the user?

Icons (if they are done correctly) can make it easier to perceive information. Seeing an icon, you might not have to read the text.
And it can be useful for structuring the scene. For example, you can quickly move objects to the appropriate groups without expanding the entire tree. We see that a collapsed object has walls - we move it to the wall group. (This is a possible example of use).

> 3\. I would also be interesting in possibly modifying the proposal to have the extra icon be grayed out as if it were diabled to make the visual imagery less prominent but still available for reference.

I made a very rough draft to show a possible principle. Of course, it can be designed better so that it is not so cumbersome and the included sub-objects do not draw attention.

I can give an example from blender (probably everyone is tired of it) ))) Looking at collapsed collections, we can see what is inside them without expanding it. 
For example, I see that in the marked collection inside there is an empty/mesh/lamps
[image]

kaiwas, 2024-09-24:
I analyzed the possible application of the original proposal and it seems it will not work.

The main reason is that there can be many, many levels in the assembly. And the main object will not always be the bottom one. As an example, walls with windows. Despite the fact that the window is located below (there can be many of them) - it is not the main one. The wall has a higher priority for display in the micro tree.

I rethought the approach. It seems you can mark the necessary objects in the view settings. And assign any icon as it is done when setting up macros in the control panel.

After such a mark, the object will be displayed in the micro tree when the branch is collapsed.

I tried to design the new concept in the images.

[image]

[image]
[image]

Mannshoch, 2024-09-24:
Related idea. See also:  #16152