FreeCAD/FreeCAD #16747: Sketcher: B-Spline by knots - weird snapping to imported external geometries

stefanpeter218, 2024-09-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Trying to make a B-Spline by knots attached to points imported from an external geometry results in weird snapping to points. Looks to me like something like an indexing error.

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Estimate 0.1.2
  * AirPlaneDesign 0.4.0
  * Curves 0.6.40
  * Manipulator 1.5.7
  * CurvedShapes 1.0.9
  * OpenDark 2024.3.13
```

### Subproject(s) affected?

None

### Anything else?

[Weird_b_Spline_by_knots_in_Sk_referencing.zip](https://github.com/user-attachments/files/17096370/Weird_b_Spline_by_knots_in_Sk_referencing.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2025-10-19:
@stefanpeter218 Can you check if it's still relevant in the newest weekly build available [here](https://github.com/FreeCAD/FreeCAD/releases)?

FEA-eng, 2026-11-06:
Still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 621.14/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```