FreeCAD/FreeCAD #16748: FEM: Quasi-structured Quad Gmsh algorithm doesn't generate 2nd order quads

FEA-eng, 2024-09-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create Part Plane
2. Mesh it with Gmsh keeping the default 2nd order setting.
3. Change the 2D algorithm to Quasi-structured Quad and remesh
4. Check the contents of the mesh (using FEM Mesh info or written CalculiX input deck). The elements are of the 1st order.

Other methods of generating quad meshes (such as recombination or subdivision) work properly.

@marioalexis84 FYI

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

marioalexis84, 2024-11-19:
This is a bug in Gmsh. The quasi-structured quad algorithm generates biquadratic quadrangles (9 nodes) but the `MQuadrangle9` class https://gitlab.onelab.info/gmsh/gmsh/-/blob/master/src/geo/MQuadrangle.h?ref_type=heads#L326 doesn't override the function `getTypeForUNV()` with the proper element ID.

FEA-eng, 2024-12-04:
> This is a bug in Gmsh. The quasi-structured quad algorithm generates biquadratic quadrangles (9 nodes) but the `MQuadrangle9` class [...]

Then I guess we should report it on their GitLab.

ickby, 2025-05-11:
I run into a similar issue. It is not a gmsh bug, but a file format limitation: UNV simply does not support Quad9, hence it gets downsampled to Quad4 for export. Other formats support it and export nicely. For some gmsh code I currently work on I replaced gmsh export format with vtk and this exports the correct element.
Note: One needs to set "Mesh.SecondOrderIncomplete = 1" for this, as FreeCAD also does not support Quad9, but needs Quad8, which is created with this setting.

Hopefully the PR with the relevant changes will come in the next weeks.

luzpaz, 2025-05-11:
> Hopefully the PR with the relevant changes will come in the next weeks.

@ickby can this then be assigned to you ?

ickby, 2025-05-11:
> > Hopefully the PR with the relevant changes will come in the next weeks.
> 
> [@ickby](https://github.com/ickby) can this then be assigned to you ?

yes, please do.

luzpaz, 2025-11-14:
* Copied over from duplicate #16028 also opened by @FEA-eng

-----

### Problem description

As pointed out in [this issue](https://github.com/FreeCAD/FreeCAD-Bundle/issues/277) (posted in the wrong repo), the newly added Quasi-structured Quad Gmsh algorithm doesn't work with _Create mesh from shape_ (_Mesh_FromPartShape_). The following messages appear:

```
...
Warning : No vector element found containing point (9.76562,10,0) in PostView field (for norm)
...
Warning : Mesh generation error summary
Warning :  1665 warnings
Warning :     0 errors
Warning : Check the full log for details
Warning : --------------------------
```

However, it works with _Refinement_ (_Mesh_RemeshGmsh_).

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git)
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Mesh

ickby, 2025-11-14:
> Hi [@ickby](https://github.com/ickby), just checking if there has been any progress ?

Yes, the PR including the fix for this issue is up: #24792 
However, still early on that PR, there is more work needed to make it merge ready.