FreeCAD/FreeCAD #16749: AutoUpdate Feature - Add Operation Details

LB45440078L, 2024-09-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When doing an automatic update of the app, using an AppImage release, there's no sort of progress or detail of what's happening during it.
On slow computers, an update can take several minutes, and since the app gets stuck and doesn't have a loading bar or any step, one may be easily prone to close it or unable to know if installation steps are proceeding successfully.

### Full version info

```shell
OS: CachyOS (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Render 2024.8.4
```

### Subproject(s) affected?

None

### Anything else?

Tips for implementation: Add some sort of progress bar, or some text that indicates ongoing operations and eventual errors.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-09-23:
Maybe your update Progress Bar is being hidden behind the main window, it should be On Top as mine is:

[image]

LB45440078L, 2024-09-23:
Oh, so there's supposed to be a progress bar already.
I will try doing another update and seeing if it's an issue with the windows not opening.

LB45440078L, 2024-09-23:
Further updates on this issue.
I was on release candidate 1.0.
I pressed the update button, and it downloaded FreeCAD 0.21.1... ?
[image]
[image]

luzpaz, 2024-10-05:
>  I pressed the update button, and it downloaded FreeCAD 0.21.1... ?

Can't reproduce on weekly  
[image]

```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git) AppImage
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Classic/Qt default
```

luzpaz, 2024-10-05:
I also can't reproduce the OP bug report in this ticket since i observe the dialog with a progress bar

```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git) AppImage
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Classic/Qt default
```