FreeCAD/FreeCAD #16750: CAM: Deburr failure, maybe the same as issue #16495

ShamanTcler, 2024-09-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have tried to make it easy to replicate, including file and a movie.  (is this a good or bad thing?)

Deburr op is failing with ....

08:50:30  pyException: Traceback (most recent call last):
  File "C:\FreeCadBeta\Mod\CAM\PathScripts\PathUtils.py", line 70, in new_function
    res = function(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\FreeCadBeta\Mod\CAM\Path\Op\Base.py", line 788, in execute
    result = self.opExecute(obj)
             ^^^^^^^^^^^^^^^^^^^
  File "C:\FreeCadBeta\Mod\CAM\Path\Op\Deburr.py", line 380, in opExecute
    wire = PathOpUtil.offsetWire(w, base.Shape, offset, True, side)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\FreeCadBeta\Mod\CAM\Path\Op\Util.py", line 287, in offsetWire
    owire = orientWire(wire.makeOffset2D(offset), True)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ValueError'>: {'sclassname': 'class Part::NullShapeException', 'sErrMsg': 'makeOffset2D: result of offsetting is null!', 'sfile': 'C:\\bld\\freecad_1726857343022\\work\\src\\Mod\\Part\\App\\TopoShapeExpansion.cpp', 'iline': 2773, 'sfunction': 'class Part::TopoShape &__cdecl Part::TopoShape::makeElementOffset2D(const class Part::TopoShape &,double,enum Part::JoinType,enum Part::FillType,enum Part::OpenResult,bool,const char *)', 'swhat': 'makeOffset2D: result of offsetting is null!', 'btranslatable': False, 'breported': False}
08:50:30  Deburr: {'sclassname': 'class Part::NullShapeException', 'sErrMsg': 'makeOffset2D: result of offsetting is null!', 'sfile': 'C:\\bld\\freecad_1726857343022\\work\\src\\Mod\\Part\\App\\TopoShapeExpansion.cpp', 'iline': 2773, 'sfunction': 'class Part::TopoShape &__cdecl Part::TopoShape::makeElementOffset2D(const class Part::TopoShape &,double,enum Part::JoinType,enum Part::FillType,enum Part::OpenResult,bool,const char *)', 'swhat': 'makeOffset2D: result of offsetting is null!', 'btranslatable': False, 'breported': False}

### Full version info

```shell
OS: Windows 11 build 22635
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.65
  * btl 0.9.9
  * Curves 0.6.42
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Anything else?

https://github.com/user-attachments/assets/8b90167a-173b-41dc-8ac4-f051c06d6cb0

[TrimmedShield.zip](https://github.com/user-attachments/files/17097844/TrimmedShield.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-23:
Is this a dupe of #16495 ?

ShamanTcler, 2024-09-23:
It maybe, but there was not a lot of detail in 16495. I tried to give a lot of detail

luzpaz, 2024-09-23:
@jffmichi can you reproduce this issue?

jffmichi, 2024-09-23:
Yes, I can reproduce the error exactly as in the video.

```
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38742 (Git)
Build type: Release
Branch: main
Hash: b6599c5315ba557371dbdacca0b9cf35f5b7f7f2
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.2.dev
Locale: English/United States (en_US) [ OS: German/Germany (de_DE) ]
Stylesheet/Theme/QtStyle: unset/Classic/breeze
Installed mods: 
  * kicadStepUpMod 11.2.7
  * dodo 1.0.1
  * sheetmetal 0.4.24
  * fcgear 1.0.0
  * fasteners 0.5.27
  * Assembly4 0.50.15
  * Curves 0.6.45
  * lattice2 1.0.0
```

FEA-eng, 2025-10-19:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/14 15:14:51
Build type: Release
Branch: main
Hash: 41ce37233d64c81c2f105de085635d4dcd669943
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```