FreeCAD/FreeCAD #16755: Segfault when reordering objects in the objects tree

leoheck, 2024-09-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

To reproduce this:

Drag the `LCS_Base` after the `Origin` in this sample file.

> Remove the `.zip` from the filename.
[Cables.FCStd.zip](https://github.com/user-attachments/files/17101248/Cables.FCStd.zip)

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x74a297245320]
#1  0x74a2716680b4 in MeasureGui::QuickMeasure::addSelectionToMeasurement() from /tmp/.mount_FreeCABekwiq/usr/lib/MeasureGui.so+0x1e4
#2  0x74a2716685f7 in MeasureGui::QuickMeasure::tryMeasureSelection() from /tmp/.mount_FreeCABekwiq/usr/lib/MeasureGui.so+0x17
#3  0x74a27166862e in MeasureGui::QuickMeasure::processSelection() from /tmp/.mount_FreeCABekwiq/usr/lib/MeasureGui.so+0x1e
#4  /tmp/.mount_FreeCABekwiq/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x74a2978eb6f2]
#5  0x74a2978ef34a in QTimer::timeout(QTimer::QPrivateSignal) from /tmp/.mount_FreeCABekwiq/usr/bin/../lib/libQt5Core.so.5+0x2a
#6  0x74a2978e064d in QObject::event(QEvent*) from /tmp/.mount_FreeCABekwiq/usr/bin/../lib/libQt5Core.so.5+0x7d
#7  0x74a2987703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCABekwiq/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#8  0x74a29a5433a9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCABekwiq/usr/bin/../lib/libFreeCADGui.so+0x99
#9  0x74a2978b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCABekwiq/usr/bin/../lib/libQt5Core.so.5+0x108
#10  0x74a297909f51 in QTimerInfoList::activateTimers() from /tmp/.mount_FreeCABekwiq/usr/bin/../lib/libQt5Core.so.5+0x381
#11  /tmp/.mount_FreeCABekwiq/usr/bin/../lib/libQt5Core.so.5(+0x30a75c) [0x74a29790a75c]
#12  /tmp/.mount_FreeCABekwiq/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x74a295b10ad7]
#13  /tmp/.mount_FreeCABekwiq/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x74a295b13ac7]
#14  /tmp/.mount_FreeCABekwiq/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x74a295b140ce]
#15  0x74a29790aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCABekwiq/usr/bin/../lib/libQt5Core.so.5+0x66
#16  0x74a2978b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCABekwiq/usr/bin/../lib/libQt5Core.so.5+0x11b
#17  0x74a2978bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCABekwiq/usr/bin/../lib/libQt5Core.so.5+0x82
#18  0x74a29a4a4958 in Gui::Application::runApplication() from /tmp/.mount_FreeCABekwiq/usr/bin/../lib/libFreeCADGui.so+0xa28
#19  /tmp/.mount_FreeCABekwiq/usr/bin/freecad(+0x7640) [0x5c85c649e640]
#20  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x74a29722a1ca]
#21  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x74a29722a28b]
#22  /tmp/.mount_FreeCABekwiq/usr/bin/freecad(+0x79a9) [0x5c85c649e9a9]

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git) AppImage
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.26
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.22
  * lattice2 1.0.0
  * kicadStepUpMod 11.3.2
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-25:
@PaddleStroke FYI

PaddleStroke, 2024-10-08:
I think this is fixed by https://github.com/FreeCAD/FreeCAD/pull/16921
@leoheck can you please check on your side?

luzpaz, 2024-10-11:
@leoheck :point_up_2:

depthoffocus, 2024-10-21:
I can't test on Ubuntu but this is fine on:

```
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git)
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ThreadProfile 1.92.0
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * Manipulator 1.5.7
  * fasteners 0.5.29
  * lattice2 1.0.0
  * CurvedShapes 1.0.12
  * Curves 0.6.50
```

https://github.com/user-attachments/assets/a95f7db5-0688-40a6-ac6e-ff7119a70b58

luzpaz, 2024-10-22:
I can't reproduce either on 
```
OS: Manjaro Linux (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git) AppImage
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

luzpaz, 2024-10-22:
CC @leoheck

luzpaz, 2024-10-26:
Unable to reproduce on  
```
OS: Manjaro Linux (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git) AppImage
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

maxwxyz, 2024-10-26:
closing this for no further feedback. Please comment if there is still an issue.

closed 2024-10-26 by maxwxyz