FreeCAD/FreeCAD #16757: Thread with cylinder leads to deformation

Tommy-01, 2024-09-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Followed this [instruction](https://wiki.freecad.org/Thread_for_Screw_Tutorial/de#Arbeitsbereich_PartDesign) to make an thread.
Everything works until I add a cylinder at point 8. As soon as this cylinder is rendered, the structure is defomed.

Here you can see the cylinder with the thread.
[image]

Here you can see the rendered version.
[image]

The construction as a file.
[Thread_with_cylinder.FCStd.zip](https://github.com/user-attachments/files/17102439/Thread_with_cylinder.FCStd.zip)

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * fasteners 0.5.29
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-23:
This should be fixed on 1.0RC1 or in weekly. Can you try reproducing ?  (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc1)

Tommy-01, 2024-09-23:
I can reproduce it.
[image]

OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)

FlachyJoe, 2024-09-23:
It's a known OCC bug, rotate Helix per 1° around Z-Axis in order the seam edge of the cylinder doesn't coincide anymore with the thread start.
[image]

Tommy-01, 2024-09-25:
I'm probably doing something wrong. If I turn the thread, the body is still deformed.

Here you see the Change an the rendered Body.
[image]
[image]

Tommy-01, 2024-09-25:
Ah sry, now i got it. I rotated the wrong thing. As you write, rotate the Helix an now it's all right.
Thank you!
[image]

closed 2024-09-25 by luzpaz