FreeCAD/FreeCAD #16759: make pyngcore optional?

looooo, 2024-09-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I tried to mesh a model with netgen and this seems to be broken due to import of `pyngcore`. The conda-packages use an older netgen version, so I think there is no `pyngcore available for this version and updateing to a newer version of netgen is difficult. So for now is it an option to make the import of `pyngcore` optional?

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

looooo, 2024-09-23:
@marioalexis84

marioalexis84, 2024-09-23:
Ok. I will prepare a PR request to optionally choose the old or new Netgen FemMesh implementation until the package issues are resolved.
In the meantime, is the `ngsolve` pip package a solution for you? (in fact, just the `netgen-mesher` package is enough)

looooo, 2024-09-24:
actually the pip package worked for me. But I am not sure netgen is running in parallel. Is freecad using the pip installed package if it is available. I am not familiar with the new implementation. Maybe you can give me some insights.

marioalexis84, 2024-09-24:
The new implementation uses only the Netgen Python bindings. Unlike the old FemMesh Netgen implementation, it doesn't use the smesh plugin at all, so it is no longer necessary build FreeCAD with Netgen. If you're able to generate the mesh it's because the installed python package is being used.
From a user point of view, in the task panel there are no major differences. But from the properties panel there are more than 40 meshing parameters available (basically, all available in Netgen) and you can cancel the process if necessary by pressing the Cancel button.
Parallelization should work by default and you should see a second freecad process running for most of the meshing process with a load percentage according to the number of threads set in the Thread property of the object.
Here are some more details: https://ngsolve.org/news/new-features/52-netgen-parallelization
Note that old FemMesh Netgen objects are not migrated to the new implementation: If you have a project with one of these objects, they will continue to work with the old implementation.

FEA-eng, 2024-09-25:
> actually the pip package worked for me.

What commands did you use to install it? It still doesn't work for me after using `pip install --upgrade ngsolve` and some other variations.

looooo, 2024-09-25:
> > actually the pip package worked for me.
> 
> What commands did you use to install it? It still doesn't work for me after using `pip install --upgrade ngsolve` and some other variations.

I just installed it with:
pip install netgen-mesher

> The new implementation uses only the Netgen Python bindings. Unlike the old FemMesh Netgen implementation, it doesn't use the smesh plugin at all, so it is no longer necessary build FreeCAD with [...]

ok this is really great work. This will simplify the maintainace. One question that came up, when reading this was: Is it also possible to move the 2d meshing (meshing of sufaces) like it is done in the mesh-workbench to the python interface. If this is possible I guess we can drop the smesh dependency in the future.

FEA-eng, 2024-09-25:
> I just installed it with: pip install netgen-mesher

Globally or from a specific location (by starting the command line e.g. in FreeCAD/bin)?

looooo, 2024-09-25:
> > I just installed it with: pip install netgen-mesher
> 
> Globally or from a specific location (by starting the command line e.g. in FreeCAD/bin)?

from a conda environment where I have installed freecad

marioalexis84, 2024-09-25:
Netgen forum post:  https://forum.ngsolve.org/t/problems-with-netgen-pip-package/3078
>from a conda environment where I have installed freecad

Can you try in a clean conda environment?

jalajc, 2024-10-14:
facing the same issue with FC 1.0 RC2. conda-Windows-x86_64-py311.
```
0.0: Start process...
0.5: To use FemMesh Netgen objects, install the Netgen Python bindingsTraceback (most recent call last): File "", line 4, 
in File "C:\Users\XXX\FreeCAD_1.0.0RC2-conda-Windows-x86_64-py311\Mod\Fem\femmesh\netgentools.py", line 122, 
in run_netgen import pyngcore as ngcore : No module named 'pyngcore' 

```
above discussion seems too confusing. wonder what it has to do with pyngcore, 

for time being, If someone can tell, what to do to mitigate error? like install / uninstall packages?

jalajc, 2024-10-14:
facing the same issue with FC 1.0 RC2. conda-Windows-x86_64-py311.
```
0.0: Start process...
0.5: To use FemMesh Netgen objects, install the Netgen Python bindingsTraceback (most recent call last): File "", line 4, 
in File "C:\Users\XXX\FreeCAD_1.0.0RC2-conda-Windows-x86_64-py311\Mod\Fem\femmesh\netgentools.py", line 122, 
in run_netgen import pyngcore as ngcore : No module named 'pyngcore' 

```
above discussion seems too confusing. wonder what it has to do with pyngcore, 

for time being, If someone can tell, what to do to mitigate error? like install / uninstall packages?
because getting following error after `pip install --upgrade ngsolve` , result in following error.

[image]

FEA-eng, 2024-10-14:
> for time being, If someone can tell, what to do to mitigate error? like install / uninstall packages?

For now, there's a setting (under Edit --> Preferences --> FEM --> Netgen) to use old Netgen implementation in weekly builds. But there's also a plan to include proper files with installers for 1.0, hopefully making it work on all systems.

looooo, 2024-10-14:
can't you just check if netgen is importable. 
```
try:
    import netgen
    new implementation
    print(DeprecationWarning("python bindings for netgen are necessary for future versions of netgen")
except importError:
    old implementation
```

leoheck, 2024-11-06:
> For now, there's a setting (under Edit --> Preferences --> FEM --> Netgen) to use old Netgen implementation in weekly builds. But there's also a plan to include proper files with installers for 1.0, hopefully making it work on all systems.

[image]

I could not find this `Edit --> Preferences --> FEM --> Netgen` on 1.0-rc2. 
Do you guys could configure Netgen to work?

leoheck, 2024-11-06:
By the way, I installed `netgen-mesher` from inside the running Freecad using the python console as suggested here https://forum.freecad.org/viewtopic.php?t=85541

But it is not working... with the same message:

```
0.0: Start process...
0.8: To use FemMesh Netgen objects, install the Netgen Python bindingsTraceback (most recent call last): File "", line 4, in File "/tmp/.mount_FreeCAmOJ8H3/usr/Mod/Fem/femmesh/netgentools.py", line 122, in run_netgen import pyngcore as ngcore : No module named 'pyngcore' 
```

luzpaz, 2024-11-06:
@leoheck can you add the full About info please ?

marioalexis84, 2024-11-07:
Are you sure the package is installed in the correct path?
Try this.
```
import os, subprocess
# get AppImage Python interpreter path
temp_dir = os.path.dirname(os.sys.executable)
py_path = os.path.join(temp_dir, "python")
# use pip from AppImage Python and install package in home user directory
subprocess.run([py_path, "-m", "pip", "install", "--user", "netgen-mesher"])
```

marioalexis84, 2024-11-24:
@maxwxyz this is solved in https://github.com/FreeCAD/FreeCAD/pull/17011

closed 2024-11-24 by maxwxyz