FreeCAD/FreeCAD #16760: Measure Tool crashes app

sthenge, 2024-09-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Measure Tool.  If you click on the Result: field and hit enter, the app crashes.  There is no error message, however, below is the error report resulting from running app from the terminal.

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * freecad.gears 1.3.0
  * Curves 0.6.47
  * fasteners 0.5.29
  * ThreadProfile 1.92.0
```

### Subproject(s) affected?

None

### Anything else?

bill@bill-MINI-S:~/FreeCad$ ./FreeCAD_1.0.0RC1-conda-Linux-x86_64-py311.AppImage 
FreeCAD 1.0.0, Libs: 1.0.0RC1R38642 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

3DfindIT: FreeCAD below version 0.19 is not supported. Please update to a recent version.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x720100a45320]
#1  0x7200d6fd1f08 in Gui::TaskMeasure::saveObject() from /tmp/.mount_FreeCAKakhJA/usr/lib/MeasureGui.so+0x98
#2  0x7200d6fd401a in Gui::TaskMeasure::apply() from /tmp/.mount_FreeCAKakhJA/usr/lib/MeasureGui.so+0xa
#3  0x7200d6fd4166 in Gui::TaskMeasure::eventFilter(QObject*, QEvent*) from /tmp/.mount_FreeCAKakhJA/usr/lib/MeasureGui.so+0xa6
#4  0x7201010b67ca in QCoreApplicationPrivate::sendThroughApplicationEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAKakhJA/usr/bin/../lib/libQt5Core.so.5+0x6a
#5  0x720101f703e8 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAKakhJA/usr/bin/../lib/libQt5Widgets.so.5+0xb8
#6  0x720103d3e499 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAKakhJA/usr/bin/../lib/libFreeCADGui.so+0x99
#7  0x7201010b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAKakhJA/usr/bin/../lib/libQt5Core.so.5+0x108
#8  0x7201017312fb in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /tmp/.mount_FreeCAKakhJA/usr/bin/../lib/libQt5Gui.so.5+0xdb
#9  0x72010170b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAKakhJA/usr/bin/../lib/libQt5Gui.so.5+0xac
#10  /tmp/.mount_FreeCAKakhJA/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x7200fb894bea]
#11  /tmp/.mount_FreeCAKakhJA/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7200ff710ad7]
#12  /tmp/.mount_FreeCAKakhJA/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7200ff713ac7]
#13  /tmp/.mount_FreeCAKakhJA/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7200ff7140ce]
#14  0x72010110aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAKakhJA/usr/bin/../lib/libQt5Core.so.5+0x66
#15  0x7201010b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAKakhJA/usr/bin/../lib/libQt5Core.so.5+0x11b
#16  0x7201010bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAKakhJA/usr/bin/../lib/libQt5Core.so.5+0x82
#17  0x720103c9f943 in Gui::Application::runApplication() from /tmp/.mount_FreeCAKakhJA/usr/bin/../lib/libFreeCADGui.so+0x983
#18  /tmp/.mount_FreeCAKakhJA/usr/bin/freecad(+0x7640) [0x642c862a6640]
#19  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x720100a2a1ca]
#20  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x720100a2a28b]
#21  /tmp/.mount_FreeCAKakhJA/usr/bin/freecad(+0x79a9) [0x642c862a69a9]
bill@bill-MINI-S:~/FreeCad$ 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-09-24:
Please try in the newest weekly build, issues with this tool are fixed frequently.

sthenge, 2024-09-24:
I just retested the latest daily and the problem no long occurs.

OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38827 (Git) AppImage
Build type: Release
Branch: main
Hash: abd5c3ca31539f8ed37ac3dec2b9abd252a5310d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * 3DfindIT 1.2.0
  * freecad.gears 1.3.0
  * Curves 0.6.47
  * fasteners 0.5.29
  * ThreadProfile 1.92.0

closed 2024-09-24 by sthenge