FreeCAD/FreeCAD #16761: Combo View

pcharliep01, 2024-09-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
[image]

### Full version info

```shell
The Combo panel doesn't display properly and makes the program unusable. I am using ver 0.21.2 and all the previous version work fine but the ver1.0RC will not display the Model/Task Combo View panel properly. I have tried to change the setting in the View-Panels but even here the option of for the Combo View is missing.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-24:
Press F4 and restart FreeCAD. If that does not fix it then you need to explain better.

maxwxyz, 2024-09-24:
I cannot reproduce either. Are all panels activated?

Roy-043, 2024-09-24:
See my previous comment.
View -> Dock window overlay -> Toggle overlay for all

pcharliep01, 2024-09-24:
I tried holding down F4 and restarting FreeCAD with no change. With 0.21.2 and older I get the Combo View with Model and Tasks Panels. How can I get them to appear the same way on 1.0RC . In 0.21.0 pull down menu View > Panels I get 4 options Report, Selection, Combo (checked) and Python Console with 1.0RC I get 5 options Selection View, Python Console, Model (checked) Tasks (checked) and Report (checked) . Can I get these panels (Model and Tasks) to appear the same way as they do on 0.21.0 ( side by Side)
Thanks

Roy-043, 2024-09-26:
I did not say 'hold down F4'. I said: 'Press F4'.

You must have accidentally pressed F3 or F4 to get this result in V1.0.
See: View > Dock panel overlay

To get the old layout:
https://wiki.freecad.org/Combo_view#Dock_Task_panel_on_top_of_Combo_view

In the version listed below F4 works better. I do not have to restart FreeCAD.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38830 (Git)
Build type: Release
Branch: main
Hash: b38c42262f58cf0587b627e7a866d3756da34dc4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

pcharliep01, 2024-09-26:
Hi, got it 🙂 I used the link to get the old layout and that worked. I'm not sure if I pressed any of the F-buttons, also I did try deleting the download and re-downloading ver1.0 a few times with the same result every time.
Many thanks for your help

https://wiki.freecad.org/Combo_view#Dock_Task_panel_on_top_of_Combo_view

closed 2024-09-27 by maxwxyz