FreeCAD/FreeCAD #16762: Property view in FreeCAD v1.0RC1

dbyy, 2024-09-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I'm open an existing object and look at the property view in RC1 every second item in the property view is colored black font on very dark gray background. That makes it (almost) impossible to identify the contents. I tried to adjust it in the preference settings "Display section" but I was unable to change it.  I reporting it to bring it to your attention and verification.
Please see attached screenshot.
FreeCAD v1.0RC1 on Linux Mint (AppImage)

Regards.

### Full version info

```shell
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Canada (en_CA)
Installed mods: 
  * Curves 0.6.47
  * woodworking 0.21.2.33771
  * 3D_Printing_Tools
  * Help 1.0.3
  * btl 0.9.9
  * Woodworking 0.21.30486
  * Pyramids-and-Polyhedrons
  * lattice2 1.0.0
  * Xray
  * Manipulator 1.5.7
  * fasteners 0.5.28
  * Assembly4 0.50.15
  * Behave-Dark-Colors 0.1.1
  * A2plus 0.4.67
  * Render 2024.8.4
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-24:
What theme did you use there? Do you have also the stable version on your system and could try with a fresh install (without user.cfg file)?

luzpaz, 2024-09-24:
Can you install weekly to see if you can reproduce ? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

dbyy, 2024-09-24:
hello luzpaz,

it seems to me that the problem has been fixed with vRC1.dev.38827.
[image]

Thanks for the quick reply.

About FreeCAD:

OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38827 (Git) AppImage
Build type: Release
Branch: main
Hash: abd5c3ca31539f8ed37ac3dec2b9abd252a5310d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * Curves 0.6.47
  * woodworking 0.21.2.33771
  * 3D_Printing_Tools
  * Help 1.0.3
  * btl 0.9.9
  * Woodworking 0.21.30486
  * Pyramids-and-Polyhedrons
  * lattice2 1.0.0
  * Xray
  * Manipulator 1.5.7
  * fasteners 0.5.28
  * Assembly4 0.50.15
  * Behave-Dark-Colors 0.1.1
  * A2plus 0.4.67
  * Render 2024.8.4

closed 2024-09-24 by maxwxyz

dbyy, 2024-09-25:
> What theme did you use there? Do you have also the stable version on your system and could try with a fresh install (without user.cfg file)?

Theme is FreeCAD classic. If needed I could install it in a VM (if the result is still of interest for you). :-)

luzpaz, 2024-09-25:
@dbyy thanks! but not necessary anymore. Though feel free to help us test and confirm any other tickets using said VM :smile: