FreeCAD/FreeCAD #16764: Cannot Create Ruled Surface on Arcs

LB45440078L, 2024-09-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I was following Mango Jelly youtube tutorial, and I noticed that the same actions that he used on a previous version, do result in broken surfaces and errors in 1.0.0 release candidate.

I will attach here the URL with the specific time of the operation that fails: [Airpods Tutorial](https://youtu.be/IALkQrtdJcg?t=2511)

The circle is made up by three, fully constrained, circular arcs. First he starts by selecting two, and then creates a face from the edge of the half-circle to the remaining arc. That does end up resulting in a full surface of a circle.
This does not happen in release candidate 1.0.0.

To further confirm that this is an issue related to 1.0.0, and not to my specific file, I have opened the same file on a 0.21 release, and was able to follow the steps correctly.

I have attached the FCStd file so whoever checks out the issue doesn't waste further time recreating it.
[https://www.mediafire.com/file/z3pxm8fqmxgd422/Airpods.FCStd/file](Download AirPods.FCStd)

Images:
Created the first surface with "ruled surface" joining together the two smaller arcs
[image]
Now joining the line with the arc, this will create a broken ruled surface that doesn't render properly and is empty.
[image]

### Full version info

```shell
OS: EndeavourOS (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-09-24:
I was unable to reproduce this issue on 1.0RC1 using a sketch with three connected arcs sharing the center point.

<details><summary>Sketch used when trying to reproduce the bug</summary>
<p>
[image]
</p>
</details> 

I did however notice that auto orientation in ruled surfaces makes mistakes quite often, which can be difficult to see when closing up a shape.

<details><summary>How difficult it can be to spot errors when auto orientation fails (notice the strange shadow in the middle)</summary>
<p>
[image]
</p>
</details> 

Could you please share a file where this bug is more easily reproducible? That would help with further investigation.

LB45440078L, 2024-09-24:
I will close this. I tried replicating this on a simpler scenario and it didn't happen again. I must've used something wrong

closed 2024-09-24 by luzpaz