FreeCAD/FreeCAD #16765: [CAM] ToolBit default shapes require recompute/save AGAIN.

spanner888, 2024-09-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It seems this  #16540, or related work now causes CAM WB to trigger this message in a variety of situations related to creating/edit or even adding ToolBits to a Job, for example:

[image]

Due to these being "system" files in install directories, or within appimage/temp dirs proposed solution above is probably best way to move forward with this.

However while I understand the need for the recompute dialog, I do not understand the actual trigger in this case, when it is only than ONE MONTH since those files were updated and released via fix for issue #15902.

Also wondering if other areas of FreeCAD may be negatively affected.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 (Git) AppImage
Build type: Release
Branch: main
Hash: 758674d40be45e365f4148fb2a5ea6bd6f93bc74
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
  * CamScripts
  * Plot 2022.4.17

**DOES NOT OCCUR IN the very recent:

OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git) AppImage
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
  * CamScripts
  * Plot 2022.4.17
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

JohnOCFII, 2024-09-24:
@bgbsww -- isn't that dialog box and message something you added recently specific to older files that needed a re-compute for TNP?  If so, this seems to be an odd place for this message to appear.  Thoughts?

bgbsww, 2024-09-24:
Yes, you can see the final trigger points in #16647.

This should happen only in Application::open and the StartView screen.

Presumably Application::open is getting used on the .fcstd files referenced by the .fctb files referenced by the library.

I don't know if we need to regenerate those files, or change the logic to somehow detect and exclude them from recomputes, or change the logic to not think they need a recompute.

JohnOCFII, 2024-09-24:
> I don't know if we need to regenerate those files, or change the logic to somehow detect and exclude them from recomputes, or change the logic to not think they need a recompute.

Guessing that @sliptonic might have some thoughts.

bgbsww, 2024-09-24:
I am unable to replicate, however I don't have an existing tool library from an older version right now.

If I create a new tool library I don't see the bug.

Any information on the age of the library in question?

My current thinking is to disable the recompute check if the file extension is lower case ".fcstd" by checking with FileInfo::extension instead of the case insensitive FileInfo::hasExtension as a good filter here - it might exclude a few files that should have the recompute dialog, but that's just an inconvenience.

But I'm not positive that's the fix since I haven't replicated it :(

hyarion, 2024-09-24:
TL;DR This is most likely already fixed, please check with 1.0RC2.

---

I've been going through the code from the where the bit library loads a documents to finished loading and I have also set up a break point in the top of the function that you made modifications to and I haven't found anything that my 24h old head (4d35aa0) can cause the dialog to open.

`Bit.py` etc in CAM Mod calls `App::Application::openDocument`.
The only way the recompute dialog can be shown is if `Gui::Application::open()`, `StartGui::StartView::openExistingFile()`, or `StartGui::StartView::fileCardSelected()` is called.

The issue was reported with quite recent versions of freecad, but not as recent as a [change](https://github.com/FreeCAD/FreeCAD/pull/16647/commits/bd06567d25dfcf2382806f8a31203c5ba6b109e5), which changed where and when the dialog is shown when loading normal files. It used to be connected to a signal that was sent when a document was restored/loaded but now it's called directly after a file has been loaded through the gui.

So since the CAM workbench doesn't load any file that way it should be fine with the current way the dialog is shown.

It would be nice if someone (@spanner888?) could test if the dialog is shown in RC2 (I can't see how it could be, but I could be wrong). 
@bgbsww I think you can wait with the Pull Request you've done, as I don't think it is needed :)

spanner888, 2024-09-25:
So I have tested with 38827 and issue does not occur, including via macros which originally alerted me to the dialog.

FYI: Those shape files are part of FreeCAD. Most users only use the default system shapes, ie those provided "inside" FreeCAD.
This is distinct from the actual ToolLibrary and Tools in each library table. When the Tool Edit dialog is open. the actual path to the ToolBit Shape file is shown, eg for my case with appimage FC: 

```/tmp/.mount_FC_wklhOmH8t/usr/Mod/CAM/Tools/Shape/endmill.fcstd```

The Tool Library edit dialog shown above is one way to trigger CAM Workbench to actually open the shape file. If library dialog is moved to one side, you will see the shape open in a document (after you respond to modal dialog that this issue is about).

In addition, using a decompressed appimage of 38768, I opened, recomputed and saved every one of the system ToolBit shape files, but the dialog still appears after doing this! This occurs when doing either a File Open or when triggering the File open via CAM WB Library manager as originally described above. I repeated this process using 38827 to open/recompute/save the uncompressed 38768 files and then opening again with the older 38768 showed the error still occurs!!

Further to above, more testing with 38768 shows that there is some intermittent aspect to the issue. It does mostly occur over 80% of time, but only sometimes does not open that dialog.  This also occurs when directly opening one of the shape files using FreeCAD File Open.

On some occasions it appears to be first edit of Tool in Library ToolTable does not open the dialog, but this is not consistent.

In summary, some unusual behaviour with 38768, but 38827 seems to have fixed the issue.

luzpaz, 2024-09-25:
How to proceed? Please advise?

bgbsww, 2024-09-25:
There appears to be no bug in the current code.  It was fixed with #16647.  For whatever reason, there is still testing going on with versions that predate that, and those results aren't relevant.  So, this should be closed as already fixed, unless @spanner888 has some issue with the current code.

spanner888, 2024-09-25:
"In summary, some unusual behaviour with 38768, but 38827 seems to have fixed the issue."

closed 2024-09-25 by spanner888