FreeCAD/FreeCAD #16767: Sketcher: Some constraints have disruptive popups..

leoheck, 2024-09-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

These kind of popups break the hole flow of design.
Please, avoid using popups that need with human interaction for things we dont have any opption to choose.
[image]

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git) AppImage
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.26
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.24
  * lattice2 1.0.0
  * kicadStepUpMod 11.3.8
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-24:
should be a warning instead IMO

Roy-043, 2024-09-24:
There already is a preference for this:
"Enable non-intrusive notifications" needs to be checked (which is also the default).

https://wiki.freecad.org/Preferences_Editor#Notification_Area

closed 2024-09-24 by maxwxyz

leoheck, 2024-09-24:
So this is a bug. I dont have notifications enabled.

This silly popup should be removed.
It should log to the Report View at most.

[image]

leoheck, 2024-09-24:
Right now it looks like I have to "enable notifications" to "disable" them.

leoheck, 2024-09-25:
Can you please reopen this ticket? The problem still persists.

Roy-043, 2024-09-26:
You need to check the two checkboxes at the top of your screenshot.

leoheck, 2024-09-26:
How this logic works? I *don't* want notification. So I have to enable notifications? That doesn't make any sense.

luzpaz, 2024-09-27:
It is kind of awkward.

Roy-043, 2024-09-27:
Try unchecking the first checkbox (leaving the 2nd checkbox checked).

leoheck, 2024-09-27:
I see what you are saying @Roy-043 but enabling something to disable it, does not make any sense. No matter if you are an common end user or the smartest developer.

Roy-043, 2024-09-28:
I suppose the text can be clearer. As far as I understand, by enabling "non-intrusive notifications" you avoid "intrusive notifications". I do not know if there is a way to avoid all notifications. But that would probably be a bad idea anyway.

leoheck, 2024-09-28:
Messages on report view are non intrusive.

leoheck, 2024-09-28:
> I suppose the text can be clearer. As far as I understand, by enabling "non-intrusive notifications" you avoid "intrusive

I am using it like this ... (I dont wan to do it, but it was recomended here)
[image]

Still receiving things like this which is still a very intrusive thing.
[image]

Why things like this have to explode on users face?

Roy-043, 2024-09-30:
> Try unchecking the first checkbox (leaving the 2nd checkbox checked).

Have you tried this?

leoheck, 2024-09-30:
@Roy-043 , like on this message here https://github.com/FreeCAD/FreeCAD/issues/16767#issuecomment-2371183624?
It was like that when I opened the issue.

Roy-043, 2024-09-30:
Check both checkboxes, and then uncheck the first.

leoheck, 2024-09-30:
This, right?
[image]

leoheck, 2024-09-30:
@maxwxyz time to open this again, see the picture above.

Roy-043, 2024-10-01:
Are you restarting FreeCAD after changing the preferences?

Here is an overview of all possible combination of the mentioned checkboxes:
```
Test scenario:
Restart after every change to the preferences.
Apply a horizontal constraint to a sketch line that aready has such a constraint.

0=unchecked
1=checked

Enable Notification Area           0
Enable non-intrusive notifications 0
=> 2 warning popups.

Enable Notification Area           1
Enable non-intrusive notifications 0
=> 2 warning popups. Nothing in the notification area.

Enable Notification Area           0
Enable non-intrusive notifications 1
=> 2 messages only appear in the status bar (only the last can be read).

Enable Notification Area           1
Enable non-intrusive notifications 1
=> 2 messages appear in the status bar (only the last can be read). And also in the notification area.
```

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38843 (Git)
Build type: Release
Branch: main
Hash: d519f24df9931e4a98d19ee39cdcd6124ae6a7bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

Roy-043, 2024-10-01:
There are some minor issues here:
* The "Enable non-intrusive notifications" checkbox in the Preferences Editor should not depend on the "Enable Notification Area" checkbox as it does now. "Enable non-intrusive notifications" has a separate behavior.
* Apparently the "Enable non-intrusive notifications" text is unclear.
* Sketcher should issue only a single warning here.