FreeCAD/FreeCAD #16769: Part Sweep: Create a solid checkbox seems inverted

leoheck, 2024-09-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Part sweep, Frenet is the default, Solid not. Why?

What the "Solid" option means? 

If I check it, it creates a hollow structure, if I unchecked it, it makes a solid.

[image]

Also, if this issue is sound, check the Frenet, since I have no clue if it is right or wrong like the Solid option.

UX/UI: Also, Selecting a Sketch as the profile is horrible, I have to pick one from the list there urgh.

I want to pick one from the screen, or the Object tree, since they are where I know they are.

That list is garbage, there are tons of things out of order, and I cannot even locate something without having to dig for a couple of minutes.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git) AppImage
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.26
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.24
  * lattice2 1.0.0
  * kicadStepUpMod 11.3.8
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-24:
> If I check it, it creates a hollow structure, if I unchecked it, it makes a solid.

Can't confirm. You need to attach a file.

leoheck, 2024-09-25:
Sometimes things behave oddly. I am closing this since I cannot reproduce it anymore.

closed 2024-09-25 by leoheck