FreeCAD/FreeCAD #16770: Set Color By Face RESETS THE SELECTION!!!

leoheck, 2024-09-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Seriously WHYYYYYYYYYYYYY??????? 

It is disgusting when functions RESETS/CLEAN the selected ITEMS!! WHY this happens?

Steps to reproduce it:

Select a bunch of things on an object.
Right click on the Object
Set Color Per Face
die of sadness.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git) AppImage
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.26
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.24
  * lattice2 1.0.0
  * kicadStepUpMod 11.3.8
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-09-24:
According to the documentation, you have to select an object in the tree first, activate the tool and only then you can select the faces to set their color. That's how this tool works, preselection of the faces is not supported.

maxwxyz, 2024-09-24:
yes, works as expected.

closed 2024-09-24 by maxwxyz

leoheck, 2024-09-24:
> yes, works as expected.

@maxwxyz I am suggesting that it should not remove the selection.
If it works as expected is because the bug is in the specification.