FreeCAD/FreeCAD #16771: [ASSEMBLY] Poor performance in simple assemblies.

obelisk79, 2024-09-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a simple assembly consisting of fixed joints, I noticed extreme performance hangups where the solver maxes out a single cpu core and freezes the application for seconds to minutes. The performance is most notably impacted when attempting to select elements for new joints/mates.

Attached is the (incomplete) assembly I was working with.
[redesign.zip](https://github.com/user-attachments/files/17107261/redesign.zip)

### Full version info

```shell
OS: EndeavourOS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38797 (Git)
Build type: Release
Branch: main
Hash: 86fa65ba489f0168145a1604423999b08090f857
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Installed mods: 
  * OpenTheme 2024.9.1
  * sheetmetal 0.4.25
  * Curves 0.6.47
```

### Subproject(s) affected?

Assembly

### Anything else?

It seems the solver should not need to be fully invoked while selecting elements for a resulting joint. Could it be paused during this time? Or perhaps the workbench could only send the affected/related components/joints to the solver in a more targeted manner. Rather than resolve/solve the entire assembly anytime a change is made, it can solve only for affected joints/components by checking adjacent relationships.

It would be worth noting that I believe Ondsel should also explore optimizations in the realm of multithreading/concurrency which may or may not be reasonable to do, given the complex nature of such a solver.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-24:
Selecting new elements and creating joints works without delay with this file on my system. Dragging the slider joint laggs about a second and it is not smooth.
@PaddleStroke FYI

PaddleStroke, 2024-09-24:
I have tested the enclosed file and find the same as Max : 
- Making a joint is still possible
- Dragging is very laggy.

Solving once a large assembly with 20 joints or so, is not too bad. The problem is that when you drag it is solving at every mouse move. So quite often.

So my take on this is that : 
- On regular solve, fixed joints should be processed normaly. ie as it is now.
- On dragging : The parts that are fixed together should be considered as a single part by the solver.

So when you click a part to drag it, it will start by a normal solve, to make sure the fixed joints are correctly assembled, then on mouse move it will be solving on a simplified model.

I'm on it as it's quite limiting it should get in 1.0 I think.

obelisk79, 2024-09-24:
It's quite interesting, because I just ran an appimage build and the performance is much more acceptable. Must be some kind of dependency problem on my system. It's an Arch based linux distro which tends to operate on the bleeding edge.

PaddleStroke, 2024-09-24:
So this is very interesting.
I managed to do this bundle thing and it speeds up the solver by a factor of 10 in your sample file ! 
However the dragging is still lagging!

So after a bit with timers I found that :
- Solver takes 0.01s without the bundle.
- Solver takes 0.001s with the bunde (so 10 times faster that's good).

But actually what is taking a lot of time is the redrawJointPlacements with 0.20s
This function is redrawing the joints at the correct position in python. So it's not so fast.

PaddleStroke, 2024-09-24:
So I've improved the situation so that the 'moving joint' is redrawn while the rest is hidden.
Which is good.
But at which point I found out that it was actually getJoints() in redrawJointPlacements(getJoints) that was slow.
So anyway it's fast now :)

Panda631, 2024-09-27:
I can second this behavior.
I did an assembly with 10 independent cylindrical joints which is nearly impossible to move anything.

closed 2024-09-30 by yorikvanhavre

ftdt10, 2025-01-29:
Solutions for "still touched" errors and performance issues
1. Wrapping with [Assembly container](https://wiki.freecad.org/Assembly_CreateAssembly), especially objects imported from step files. 
2. When adding multiple identical parts to an assembly, use the [Link Array](https://wiki.freecad.org/Std_LinkMake#Link_Array) function instead of Copy-Paste or Link.