FreeCAD/FreeCAD #16774: Measurement: Saving does not create a folder anymore

maxwxyz, 2024-09-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Saving a measurement does not create a measurement folder for all the saved measurements. Instead, they are added as objects at root level.
@hlorus FYI
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.47
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.22
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hlorus, 2024-09-25:
This seems to fixed already by af9097ce875d1378fd6487bc5ffedb35bd148274

maxwxyz, 2024-09-25:
confirmed

closed 2024-09-25 by maxwxyz