FreeCAD/FreeCAD #16775: The model crashes when reducing size in the spreadsheet

def-prog, 2024-09-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I created a model in part design using a spreadsheet (three dimensions only and lots of formulas)  with tow bodies. When I try to reduce the size to 10 % of the initial, it shows fools things. Sometimes only one body is downsized, sometime nothing is changed, sometime it shows bodies really different than expected.

### Full version info

```shell
OS: Debian (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38797 (Git) Snap 1145
Build type: Release
Branch: main
Hash: 86fa65ba489f0168145a1604423999b08090f857
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * woodworking 0.21.2.33771
  * LCInterlocking
  * fasteners 0.5.28
  * MOOC 2022.4.21
```

### Subproject(s) affected?

None

### Anything else?

What a great job you do!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-09-24:
@def-prog, how does it change if you reduce the size step-by-step? For example, first set it to 90%, then 70%, then 50%.

def-prog, 2024-09-25:
I tested right now by changing only one dimention, the sketch adapted but the rendering 3D didn't nor the tech draw page
----Message d'origine----

hyarion, 2024-09-25:
I have found out this the hard way too. However it's always how I've constrained my sketches.

Even though sketches are fully constrained there might be multiple solutions to the system.

Take this sketch for example:
[image]
Changing the bottom dimension (10mm) to 30mm gives this update:
[image]

As can be seen here, the triangle at the top is not the "same" anymore, in the first one the left leg is pointing NW and in the second it is pointing NE.

However, as @xtemp09 mentions, if it is gradually changed it usually works. This as I assume the solver won't pick the wrong solution when there are multiple.

There are usually ways to design sketches so this doesn't become a problem, but the way to do it depends on the specific case.

maxwxyz, 2024-09-25:
I'll close this for now if the issue is related to the solver (known issue).

closed 2024-09-25 by maxwxyz

def-prog, 2024-09-25:
Ok Tank's for your job I'l keep testing.
have a good day
def
----Message d'origine----

def-prog, 2024-09-30:
<Exception> Quantity::operator -(): Unit mismatch in minus operation
in property binding 'EscabeauSecuPolyval4_1#Sketch.Constraints'
Sketch: Quantity::operator -(): Unit mismatch in minus operation
in property binding 'EscabeauSecuPolyval4_1#Sketch.Constraints'
<Exception> Quantity::operator -(): Unit mismatch in minus operation
in property binding 'EscabeauSecuPolyval4_1#Sketch.Constraints'
Sketch: Quantity::operator -(): Unit mismatch in minus operation
in property binding 'EscabeauSecuPolyval4_1#Sketch.Constraints'
<Exception> Quantity::operator -(): Unit mismatch in minus operation
in property binding 'EscabeauSecuPolyval4_1#Sketch.Constraints'
Sketch: Quantity::operator -(): Unit mismatch in minus operation
in property binding 'EscabeauSecuPolyval4_1#Sketch.Constraints'
<Exception> Quantity::operator -(): Unit mismatch in minus operation
in property binding 'EscabeauSecuPolyval4_1#Sketch.Constraints'
Sketch: Quantity::operator -(): Unit mismatch in minus operation
in property binding 'EscabeauSecuPolyval4_1#Sketch.Constraints'
Traceback (most recent call last):
  File "/snap/freecad/1152/usr/Mod/Assembly/JointObject.py", line 1198, in camera_callback
    factor = self.get_lock_factor()
  File "/snap/freecad/1152/usr/Mod/Assembly/JointObject.py", line 1214, in get_lock_factor
    return get_camera_height(self.gui_doc) / 300
  File "/snap/freecad/1152/usr/Mod/Assembly/JointObject.py", line 153, in get_camera_height
    camera = activeView.getCameraNode()
AttributeError: getCameraNode
Traceback (most recent call last):
  File "/snap/freecad/1152/usr/Mod/Assembly/JointObject.py", line 1198, in camera_callback
    factor = self.get_lock_factor()
  File "/snap/freecad/1152/usr/Mod/Assembly/JointObject.py", line 1214, in get_lock_factor
    return get_camera_height(self.gui_doc) / 300
  File "/snap/freecad/1152/usr/Mod/Assembly/JointObject.py", line 153, in get_camera_height
    camera = activeView.getCameraNode()
AttributeError: getCameraNode
<Exception> Quantity::operator -(): Unit mismatch in minus operation
in property binding 'EscabeauSecuPolyval4_1#Sketch.Constraints'
Sketch: Quantity::operator -(): Unit mismatch in minus operation
in property binding 'EscabeauSecuPolyval4_1#Sketch.Constraints'
Traceback (most recent call last):
  File "/snap/freecad/1152/usr/Mod/Assembly/JointObject.py", line 1198, in camera_callback
    factor = self.get_lock_factor()
  File "/snap/freecad/1152/usr/Mod/Assembly/JointObject.py", line 1214, in get_lock_factor
    return get_camera_height(self.gui_doc) / 300
  File "/snap/freecad/1152/usr/Mod/Assembly/JointObject.py", line 153, in get_camera_height
    camera = activeView.getCameraNode()
AttributeError: getCameraNode
Traceback (most recent call last):
  File "/snap/freecad/1152/usr/Mod/Assembly/JointObject.py", line 1198, in camera_callback
    factor = self.get_lock_factor()
  File "/snap/freecad/1152/usr/Mod/Assembly/JointObject.py", line 1214, in get_lock_factor
    return get_camera_height(self.gui_doc) / 300
  File "/snap/freecad/1152/usr/Mod/Assembly/JointObject.py", line 153, in get_camera_height
    camera = activeView.getCameraNode()
AttributeError: getCameraNode
<Exception> Quantity::operator -(): Unit mismatch in minus operation
in property binding 'EscabeauSecuPolyval4_1#Sketch.Constraints'
Sketch: Quantity::operator -(): Unit mismatch in minus operation
in property binding 'EscabeauSecuPolyval4_1#Sketch.Constraints'
<Exception> Quantity::operator -(): Unit mismatch in minus operation
in property binding 'EscabeauSecuPolyval4_1#Sketch.Constraints'
Sketch: Quantity::operator -(): Unit mismatch in minus operation
in property binding 'EscabeauSecuPolyval4_1#Sketch.Constraints'
TempoVis.restore: failed to restore detail ('SDClipPlane', 'EscabeauSecuPolyval4_1', ''): list index out of range
TempoVis.restore: failed to restore detail ('SDCamera', 'EscabeauSecuPolyval4_1', 'the_cam'): list index out of range
this is the terminal output after I changed size in my spreadsheet the first two changes worked well and it broke when I try to get back to the original dimension.
I hope that could help.
Have a nice day
def
----Message d'origine----

image.png (view on web)
image.png (view on web)